Ysurac/openmptcprouter #911: Relationship between OpenMPTCProuter and overthebox

gwarf, 2020-03-21:
Not a bug report, but maybe something that may clarified in documentation/FAQ: what's the relationship between OpenMPTCProuter and overthebox?

OVH people seem to repeat it's something base on Overthebox, while I seem to recall from past discussions that it may not exactly be the case/your opinion. Being or not being based on overthebox is not a problem, free software are purposely made to encourage this, and I'm not willing to start a flamewar, but I'm interested in understanding the background.

> J'arrive un peu tard mais il me semble important de confirmer qu'OpenMPTCPRouter est bien un projet à l'origine basé sur OverTheBox. Et nous aussi, notre projet est basé sur des projets opensources comme OpenWRT, MPTCP et beaucoup d'autres.

Cf. https://community.ovh.com/t/inutilisable-depuis-plusieurs-jours-bye-bye-overthebox-welcome-openmptcprouter/11844/19

Thanks for all this great work @Ysurac, I'm working from home in the countryside and while being an early adopter of overthebox very happy at the beginning switching to OpenMPTCProuter changed lots of things !

Ysurac, 2020-03-21:
OpenMPTCProuter is not based on OverTheBox (it's not a fork). But OpenMPTCProuter is also based on OpenWrt and MPTCP with some patches. Original idea to use OpenWrt as base for the router come from OVH (before I configured a Debian and a Raspbian for that: http://mptcp.zugaina.com/).

For the router part, some functions (like the status page base) and init script of glorytun come (with many changes) from OVH, a ```grep -R OVH *``` (or searching OVH on the feed) can give you the result of what is coming from OVH/OverTheBox (not a lot), I always (or try to) credit original authors.

For the VPS part, nothing come from OverTheBox.

I never tried OverTheBox and I don't know if they still use MPTCP in latest release.

gwarf, 2020-03-25:
Thanks @Ysurac that's quite clear.

closed 2020-09-14 by github-actions[bot]

angt, 2020-10-27:
I just discovered this issue :)
Don't be like that, just a glance at the first commit https://github.com/Ysurac/openmptcprouter/commit/d31291ea49f74d6ddb7d5bb26881ff46c90bc806 is enough to see that it's a fork... I recognize my own code with basically just an s/OTB/OMR, and the same for -feeds.
That being said, we have no problem with that and I even had the opportunity to contribute here.
Regards.

Ysurac, 2020-10-27:
I will add a header to say that some part of build.sh come from OverTheBox.

But applications used by OpenMPTCProuter are in the feed and it's not a fork at all as I already said. And all part in the feeds that is coming from OverTheBox is credited in source.

The build script was a copy of OTB script but was widely modified since. This doesn't make OpenMPTCProuter a fork of OverTheBox or based on OverThebox.

I forgot to credit build.sh script and I apologize for that, this will be fixed asap.

Adorfer, 2020-10-27:
BTW: I still own an OTB+, which i can not use any more since OVH-Telecom decided to cancel all contracts "based in Germany" (i suppose due to some regulation issue, and before you asked: we tried a to change the contract or account via several supportcalls in .de and .fr. without legal address in France it was not possible) 
But nevertheless: with OMR we are quite happy (in plus there is no 400MBit/s limit on the "VPS-side").

Support for the physical hardware would be nice (as a target)
it needs's an x64 with EFI-boot (there is no legacy bios).

Ysurac, 2020-10-27:
The x86_64 EFI image doesn't work ?
I don't have any OTB hardware, I can't make any test on them.

Adorfer, 2020-10-27:
@Ysurac looks like i just manage to boot EFI from USB stick, not from the internal microsd-slot. 

now i have to fight the lan config, since there is an internal vlan-switch, the atom-board is tied to it via a single eth connection. 
(i am not sure how OTB does this, if it is simply vlan or if the switch is activly scripted from openwrt in order to configure the ports.)