Ysurac/openmptcprouter #913: iOS / iPadOS no stable internet

rubaz100, 2020-03-21:
## Expected Behavior

  Everything works fine like in my android devices.
I resetted my vps and my router.
It's a common problem from the past. I had the problem in the past but now I want to share it with the heros <3
And I tried to give them a static ip adress. but it wont work. I tried to connect them with the modems. But it didnt work either.

## Actual Behavior

iPadOS got a slow internet connection
iOS has sometime an internet connection.

## Steps to Reproduce the Problem

  1. 
  2.
  3.

## Specifications

  - OpenMPTCProuter version: (0.56.4)
  - OpenMPTCProuter VPS version: (Last is not a version)
  - OpenMPTCProuter platform: (WRT3200ACM)

Ysurac, 2020-03-21:
When you say it's not working with static ip, it's not working at all ? Can you ping the router ?
Can you check DNS server used on these devices ? (this must be OpenMPTCProuter IP)

rubaz100, 2020-03-21:
Yeah of course. I tried it with the fixed DNS Server (192.168.100.1) and it didnt work.

Ysurac, 2020-03-21:
So, can you ping the router ?
If yes, can you ping 8.8.8.8 ?

rubaz100, 2020-03-21:
No I cant ping

Ysurac, 2020-03-21:
So it's more a network problem here. I don't have any iOS devices, so it's difficult to say what it the problem exactly but if IP, gateway and DNS is configured this should work. And at least there is not reason that you can't ping the router IP.

closed 2020-09-14 by github-actions[bot]