Ysurac/openmptcprouter #914: Intel ax200 support?

gitxmax, 2020-03-21:
## Expected Behavior
Would like to get pcie intel ax200 card working

  In English please.

## Actual Behavior
not supported

## Specifications
  - OpenMPTCProuter platform: x86_64

Since openmptcprouter uses an old 4.9 kernel, there's no support for the widespread intel ax200 card, which I believe, is included since kernel 5.3
Openwrt backports seems to have it included.

I would like to use it as an access point, on my machine.

Is there any chance, I can have the driver and firmware compiled or build the openmptcprouter with a modern kernel?
Any guidance please?

Ysurac, 2020-03-21:
OpenMPTCProuter use a 4.19 kernel. Next release should be based on at least 5.4 kernel.
OpenMPTCProuter is based on OpenWrt, and iwlwifi is backported but a firmware is also needed.
I don't really know what is the support status of ax200...

gitxmax, 2020-03-21:
thanks, will wait for the next release then.

closed 2020-03-21 by gitxmax