Ysurac/openmptcprouter #915: Bug redirection de port depuis maj 0.54.6

RH2WTF, 2020-03-21:
OpenMPTCProuter version: 0.54.6 Officiel
    OpenMPTCProuter VPS version: 0.1016 (vps FIRSTHEBERG)
    OpenMPTCProuter platform: x86_64
    Glorytun UDP
    3 WAN 

[image]

Salut @Ysurac,

tous d’abord merci pour le travail que tu fournit sur ce projet.

mon problème est le suivant, depuis la mise à jour du 0.53.3 à la 0.54.6, je rencontre un problème de port forwarding lorsqu'une déconnexion du routeur à lieu, au lieu de remettre mes ouvertures de ports dans le fichier /etc/shorewall/rules sur le VPS, cela le laisse vide et je suis obligé de reboot le VPS pour résoudre mon problème. 

Ci joint, suite à une coupure :

[image]

[image]

suite au reboot du VPS j'ai bien mes 3 ouvertures : 

[image]

et je précise que si je redémarre juste le VPS lorsque les règles d'ouvertures sont en place, cela fonctionne sans problème, elle se remette en place sans soucis dans le fichier /etc/shorewall/rules
Par contre dans le cas d'une coupure brève, cela pose soucis.

Merci d'avance pour ton retour,

Ysurac, 2020-03-21:
Les règles ne sont pas enlevées quand le routeur se déconnecte (vu que le routeur ne peux pas communiquer), elles sont enlevées quand il se reconnecte ?

RH2WTF, 2020-03-21:
oui c'est bien cela

Ysurac, 2020-03-21:
Faut juste que j'arrive à comprendre comment cela est possible

RH2WTF, 2020-03-21:
l'exemple que j'ai rencontré cet après midi c'est : 
- je suis connecté en VPN depuis l’extérieur de mon réseau, sur le site possédant le routeur MPTCP, sur un des 3 ports redirigé ci dessus.

- au bout de 2 heures d'activité, je suis déconnecté, je regarde la supervision, j'ai perdu 3 ping sur le tunnel 10.255.251.1, et je regarde le fichier  /etc/shorewall/rules, il est vide comme ci dessus.

- je redémarre le routeur, tous passe en vert mais je ne peux toujours pas accédé à mon port depuis l'extérieur, je revérifie le fichier /etc/shorewall/rules, mes port n'apparaissent pas.

- j'ai redémarré le VPS et la cela à fonctionné, comme si un service ou autre bloquait sur le VPS.

Ysurac, 2020-03-21:
Oui mais je vois pas comment les ports peuvent être supprimés si ils sont ouverts via OMR.
Ce n'est pas le même soucis, les ports étaient ajoutés et enlevés 1s après.

RH2WTF, 2020-03-21:
si cela peut aider :

uci show openmptcprouter :

openmptcprouter.settings=settings
openmptcprouter.settings.enabled='1'
openmptcprouter.settings.disable_ipv6='1'
openmptcprouter.settings.firstboot='0'
openmptcprouter.settings.shadowsocks_disable='0'
openmptcprouter.settings.external_check='1'
openmptcprouter.settings.scaling_governor='ondemand'
openmptcprouter.settings.scaling_min_freq='800000'
openmptcprouter.settings.check_ipv4_website='http://ip.openmptcprouter.com'
openmptcprouter.settings.check_ipv6_website='http://ipv6.openmptcprouter.com'
openmptcprouter.settings.status_vps_timeout='2'
openmptcprouter.settings.status_getip_timeout='2'
openmptcprouter.settings.ha='0'
openmptcprouter.settings.master='change'
openmptcprouter.settings.scaling_max_freq='800000'
openmptcprouter.settings.version='0.54.6'
openmptcprouter.settings.vpn='glorytun_tcp'
openmptcprouter.vps=server
openmptcprouter.vps.master='1'
openmptcprouter.vps.backup='0'
openmptcprouter.vps.port='65500'
openmptcprouter.vps.username='openmptcprouter'
openmptcprouter.vps.machine='x86_64'
openmptcprouter.vps.pihole='0'
openmptcprouter.vps.lastbackup='1578068485.3253574'
openmptcprouter.vps.redirect_ports='0'
openmptcprouter.vps.password=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.vps.ip=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.vps.ipv6=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.vps.kernel='4.19.104-mptcp'
openmptcprouter.vps.omr_version='0.1015'
openmptcprouter.vps.available_vpn='glorytun_tcp' 'glorytun_udp' 'dsvpn' 'openvpn' 'mlvpn'
openmptcprouter.vps.get_config='0'
openmptcprouter.vps.admin_error='0'
openmptcprouter.vps.token=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.vps.lastchange='XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX'
openmptcprouter.omr=router
openmptcprouter.omr.shadowsocks='up'
openmptcprouter.omr.detected_public_ipv4=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.omr.detected_ss_ipv4=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.wan1=interface
openmptcprouter.wan1.mptcp_status='MPTCP enabled'
openmptcprouter.wan1.latency_max='1052'
openmptcprouter.wan1.mtu='1500'
openmptcprouter.wan1.metric='3'
openmptcprouter.wan1.publicip=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.wan1.multipathvpn='0'
openmptcprouter.wan1.asn='PROXAD'
openmptcprouter.wan1.state='up'
openmptcprouter.wan1.manufacturer='huawei'
openmptcprouter.wan1.lc='1584809015'
openmptcprouter.wan1.multipath='on'
openmptcprouter.wan1.latency_previous='30'
openmptcprouter.wan1.latency='39'
openmptcprouter.wan2=interface
openmptcprouter.wan2.mptcp_status='MPTCP enabled'
openmptcprouter.wan2.latency_max='2199365'
openmptcprouter.wan2.metric='4'
openmptcprouter.wan2.asn='FREEM'
openmptcprouter.wan2.multipathvpn='0'
openmptcprouter.wan2.state='up'
openmptcprouter.wan2.mtu='1432'
openmptcprouter.wan2.multipath='on'
openmptcprouter.wan2.publicip=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.wan2.lc='1584809014'
openmptcprouter.wan2.latency_previous='47'
openmptcprouter.wan2.latency='42'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.state='down'
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='5'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.vps='0.1015'
openmptcprouter.latest_versions.omr='0.54.6'
openmptcprouter.latest_versions.lc='1584810966'
openmptcprouter.WAN3=interface
openmptcprouter.WAN3.mptcp_status='MPTCP enabled'
openmptcprouter.WAN3.latency_max='1262'
openmptcprouter.WAN3.metric='7'
openmptcprouter.WAN3.multipathvpn='0'
openmptcprouter.WAN3.asn='FREEM'
openmptcprouter.WAN3.publicip=XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
openmptcprouter.WAN3.mtu='1432'
openmptcprouter.WAN3.multipath='on'
openmptcprouter.WAN3.state='up'
openmptcprouter.WAN3.lc='1584810966'
openmptcprouter.WAN3.latency_previous='35'
openmptcprouter.WAN3.latency='36'

Ysurac, 2020-03-21:
J'ai réussi à reproduire le soucis, maintenant faut que je trouve pourquoi et comment ;)

RH2WTF, 2020-03-21:
Bonne nouvelle ça !!!

Ysurac, 2020-03-21:
J'ai trouvé le soucis, je vais recompiler une nouvelle image et ça sera bon.

RH2WTF, 2020-03-21:
super, merci @Ysurac, t'es le best, encore merci pour ta dispo, comme toujours.

Et par curiosité le problème venait de quoi ?

Ysurac, 2020-03-22:
Should be fixed in 0.54.7

RH2WTF, 2020-03-22:
0.54.7 installé et fonctionnel, merci encore pour ta réactivité @Ysurac .

Bon courage et bon week end

closed 2020-03-22 by RH2WTF