Ysurac/openmptcprouter #916: VPN down

patimages, 2020-03-21:
Hi, 
I used to have a fully working instal for months but with the latest updates on NUC (VPS and OMR), I get VPN down alerts daily forcing me to reboot... True with glorytun UPD or TCP... Any help ? Thanks !

Ysurac, 2020-03-21:
This would be far better to use the issue template...
I need the log from State->System log when you get VPN down and ```/var/log/daemon.log``` on the VPS.

patimages, 2020-03-21:
I am a lazy bastard ;-D ! Ok, I ll provide the required info next time I get the issue. 
Many thanks !

robotsflo, 2020-03-22:
## Expected Behavior

VPN TUNNEL UP

## Actual Behavior

VPN TUNNEL DOWN after few minutes on DSVPN

## Steps to Reproduce the Problem

  1. Nothing

## Specifications

  - OpenMPTCProuter version: (0.54.7)
  - OpenMPTCProuter VPS version: (0.1015)
  - OpenMPTCProuter platform: (x86_64)

[logOMR.txt](https://github.com/Ysurac/openmptcprouter/files/4365384/logOMR.txt)
[daemon.log](https://github.com/Ysurac/openmptcprouter/files/4365394/daemon.log)

Ysurac, 2020-03-22:
I already had the same problem with DSVPN.

uryupinsk, 2020-03-22:
I have the same problem too since 0.54.6 on DSVPN. It was not the case on 0.53.3 which was exceptionally stable.

patimages, 2020-03-22:
Got the feeling that it is due to connexion issues ... If I have a connexion going down, and then back up, the VPN remains down... Need to restart the VPS (not the router). I tried to stabilise my connexions and this looks ok ...

Ysurac, 2020-03-22:
When connexion going down public IP of the WAN change or not ?

patimages, 2020-03-22:
When I get "no ping for 1s" alert on an interface (for whatever reason)

Ysurac, 2020-03-22:
So, no IP changes only some packets lost ?

patimages, 2020-03-22:
Hard to say ... Did not get the issue today ... I placed a fixed apn : seems to help

patimages, 2020-03-22:
If I reboot a 4G box, nothing wrong ... However, if I reboot the NUC with OMR, I get the error and need to restart the VPS ...

patimages, 2020-03-22:
Same issue if I change OMR-bypass... need to restart the VPS ...

Ysurac, 2020-03-22:
I think you have an issue somewhere else...
Can I get ```/etc/shorewall/rules``` from the VPS and your configuration in System->OpenMPTCProuter and all parameters tab (and paste it here, I don't like to download file).

patimages, 2020-03-22:
# For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#       DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW             vpn
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
#
DNAT            net             vpn:$OMR_ADDR   tcp     1-64999
DNAT            net             vpn:$OMR_ADDR   udp     1-64999
ACCEPT          net             $FW             tcp     65101   # OMR openmptcprouter open shadowsocks port tcp
ACCEPT          net             $FW             udp     65101   # OMR openmptcprouter open shadowsocks port udp
DNAT            net             vpn:$OMR_ADDR   tcp     30250   # OMR openmptcprouter redirect router 30250 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     30250   # OMR openmptcprouter redirect router 30250 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     30251   # OMR openmptcprouter redirect router 30251 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     30251   # OMR openmptcprouter redirect router 30251 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     32400   # OMR openmptcprouter redirect router 32400 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     32400   # OMR openmptcprouter redirect router 32400 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     30258   # OMR openmptcprouter redirect router 30258 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     30258   # OMR openmptcprouter redirect router 30258 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     1001    # OMR openmptcprouter redirect router 1001 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     1001    # OMR openmptcprouter redirect router 1001 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     443     # OMR openmptcprouter redirect router 443 port tcp
DNAT            net             vpn:$OMR_ADDR   tcp     30253   # OMR openmptcprouter redirect router 30253 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     30253   # OMR openmptcprouter redirect router 30253 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     30259   # OMR openmptcprouter redirect router 30259 port tcp
DNAT            net             vpn:$OMR_ADDR   tcp     9001    # OMR openmptcprouter redirect router 9001 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     9001    # OMR openmptcprouter redirect router 9001 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     30255   # OMR openmptcprouter redirect router 30255 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     30255   # OMR openmptcprouter redirect router 30255 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     30254   # OMR openmptcprouter redirect router 30254 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     30254   # OMR openmptcprouter redirect router 30254 port udp
DNAT            net             vpn:$OMR_ADDR   tcp     30256   # OMR openmptcprouter redirect router 30256 port tcp
DNAT            net             vpn:$OMR_ADDR   udp     30256   # OMR openmptcprouter redirect router 30256 port udp

patimages, 2020-03-22:
Did some cleanup on keys, domain, IPS (removed or replaced by DDD: 

collectd.globals=globals
collectd.globals.alt_config_file='/etc/collectd.conf'
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googlevideo.com/omr-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.cot/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/edf.fr/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/DDD/omr_dst_bypass_wan1,omr6_dst_bypass_wan1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='96h'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='RBS850'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='3C:37:86:FC:A8:9B'
dhcp.@host[0].ip='192.168.100.245'
dhcp.@host[0].leasetime='infinite'
dhcp.@host[1]=host
dhcp.@host[1].name='RBR850'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='3C:37:86:F4:95:1F'
dhcp.@host[1].ip='192.168.100.150'
dhcp.@host[1].leasetime='infinite'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].Port='30255'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@domains[0]=domains
dscp.@domains[0].name='googlevideo.com'
dscp.@domains[0].class='cs4'
[... 1945 lines cut ...]
wireless.radio2.type='mac80211'
wireless.radio2.channel='36'
wireless.radio2.hwmode='11a'
wireless.radio2.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio2.htmode='VHT80'
wireless.radio2.disabled='1'
wireless.default_radio2=wifi-iface
wireless.default_radio2.device='radio2'
wireless.default_radio2.network='lan'
wireless.default_radio2.mode='ap'
wireless.default_radio2.ssid='OpenWrt'
wireless.default_radio2.encryption='none'
wireless.radio3=wifi-device
wireless.radio3.type='mac80211'
wireless.radio3.channel='36'
wireless.radio3.hwmode='11a'
wireless.radio3.path='pci0000:00/0000:00:1c.5/0000:3a:00.0'
wireless.radio3.htmode='VHT80'
wireless.radio3.disabled='1'
wireless.default_radio3=wifi-iface
wireless.default_radio3.device='radio3'
wireless.default_radio3.network='lan'
wireless.default_radio3.mode='ap'
wireless.default_radio3.ssid='OpenWrt'
wireless.default_radio3.encryption='none'

patimages, 2020-03-22:
Is it all you needed ?

Ysurac, 2020-03-22:
I don't see an error.

Ysurac, 2020-03-22:
I would need System log of the router, in the state menu. When the VPN is down.

patimages, 2020-03-22:
Ok, will do when in error. Thanks anyway for the support !!

bambuca, 2020-03-22:
I had similar problem. I checked the log and there was
```
Sun Mar 22 22:02:51 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev tun0 ingress
Sun Mar 22 22:02:51 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Mar 22 22:02:51 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev tun0 root
Sun Mar 22 22:02:51 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
Sun Mar 22 22:02:51 2020 user.notice SQM: ERROR: cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4tun0 root
Sun Mar 22 22:02:51 2020 user.notice SQM: ERROR: cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
```
So I disabled SQM and restart router and now VPN works.

bambuca, 2020-03-23:
I have the same error. Now I get worked IPv6 (thanx to Ysurac), but VPN tunnels are down
[image]

bambuca, 2020-03-23:
[image]

After restart it's now more than two hours without problem. Nothing changed from my side.

patimages, 2020-03-23:
same here : need to restart the VPS ... The thing is that all work even with the VPN down so could be that more people are affected but do not realise ....

Ysurac, 2020-03-23:
When you have the case, instead of rebooting VPS, can you try to run ```systemctl restart glorytun-tcp@tun0``` on it ?

patimages, 2020-03-23:
will do thanks !

uryupinsk, 2020-03-24:
On VPN down with DSVPN, I tried `systemctl restart dsvpn-server@tun0` and `systemctl restart dsvpn-server@dsvpn0` and none of them worked. VPN goes up with a restart of the router.

Ysurac, 2020-03-24:
With dsvpn it's not the same problem. You can try a ```/etc/init.d/dsvpn restart``` on the router.

uryupinsk, 2020-03-24:
Do you have any idea of what is causing troubles with DSVPN?

Ysurac, 2020-03-24:
No idea yet, I'm not able to reproduce the issue for now.

xabix, 2020-03-25:
Hello I have similar issue (have to reboot VPS and OMR to get things back).

OMR
> Wed Mar 25 16:58:28 2020 user.notice omr-bypass: OMR-ByPass is running
> Wed Mar 25 16:58:29 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:29 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:29 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:30 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:32 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:32 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:33 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:34 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:35 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:36 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 16:58:36 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 151.101.129.164, 1.1.1.1, 1.0.0.1)
> Wed Mar 25 16:58:36 2020 user.notice Shadowsocks: Rules DOWN
> Wed Mar 25 16:58:36 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:38 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 16:58:38 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 16:58:41 2020 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
> Wed Mar 25 16:58:51 2020 user.notice OMR-VPS: Can't get token, try later
> Wed Mar 25 16:58:57 2020 daemon.info omr-tracker-ss: Server (149.91.xxx.xxx) seems down, no answer to ping
> Wed Mar 25 16:59:05 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 16:59:09 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
> Wed Mar 25 16:59:13 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 16:59:22 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 16:59:30 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 16:59:38 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 16:59:46 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 16:59:54 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:00:02 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:00:11 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:00:19 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:00:27 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:00:35 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:00:43 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:00:43 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 17:00:43 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
> Wed Mar 25 17:00:51 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:01:00 2020 daemon.err /usr/bin/ss-local[18248]: getpeername: Socket not connected
> Wed Mar 25 17:01:04 2020 daemon.err /usr/bin/ss-redir[18250]: remote recv: Operation timed out
> Wed Mar 25 17:01:05 2020 daemon.err /usr/bin/ss-redir[18249]: remote recv: Operation timed out
[... cut ...]
> Mar 25 17:10:41 XabiX shorewall[12827]: Setting up Traffic Control...
> Mar 25 17:10:41 XabiX shorewall[12827]: Preparing iptables-restore input...
> Mar 25 17:10:41 XabiX shorewall[12827]: Running /sbin/iptables-restore --wait 60...
> Mar 25 17:10:41 XabiX shorewall[12827]: IPv4 Forwarding Enabled
> Mar 25 17:10:41 XabiX shorewall[12827]: done.
> Mar 25 17:10:41 XabiX systemd[1]: Reloaded Shorewall IPv4 firewall.
> Mar 25 17:10:41 XabiX systemd[1]: Reloading Shorewall IPv4 firewall.

Any idea?

Merci

gnosoz, 2020-03-25:
not sure if it applies but I am using a modified omr-tracker as the
standard one was failing on some tunnels due to awk command not resulting
in a correct usable variable.

basically trying to reach an invalid IP as it concatenated routerIP+vpsIP
or ethXip+vpsIP.
this happens when I set all 5 connections to MPTC via vpn

in particular the following line
                       OMR_TRACKER_DEVICE_ROUTE=$(ip -4 r list dev
"$OMR_TRACKER_DEVICE" | grep via | grep -v default | grep -v metric | grep
-v / | awk '{print $1}' | tr -d "\n")

corrected into the following
                       OMR_TRACKER_DEVICE_ROUTE=$(ip -4 r list dev
"$OMR_TRACKER_DEVICE" | grep via | grep -v default | grep -v metric | grep
-v / | awk '{print $1; exit}' | tr -d "\n")

in my case I managed to backtrack the error and correct the script as per
above whilst yours seems similar it is in another point / file and I cannot
replicate the error

Ysurac, 2020-03-25:
@xabix there is something strange in time of your VPS log:
```
Mar 25 17:19:02 XabiX shorewall[5742]: Setting up Martian Logging...
Mar 25 16:56:41 XabiX ss-server[830]: getpeername: Transport endpoint is not connected
```
Shadowsocks and glorytun doesn't like when time are not synchronized or change like that.

xabix, 2020-03-25:
> @xabix there is something strange in time of your VPS log:
>
> ```
> Mar 25 17:19:02 XabiX shorewall[5742]: Setting up Martian Logging...
> Mar 25 16:56:41 XabiX ss-server[830]: getpeername: Transport [...]

Great thank you that maybe the issue. I will check and sync the VPS ntp with the one of OMR to avoid such issue! Bonne soirée

patimages, 2020-03-25:
solved for me with the latest version. Many thanks !

closed 2020-03-25 by patimages