Ysurac/openmptcprouter #918: Constant need to restart the interface on 4G router

G000R000, 2020-03-22:
## Expected Behavior

Having bonded internet connections 24/7 (ADSL + 4G)

## Actual Behavior

After some time (i can't really say when), my connection is only the ADSL (wan1). I found that i have to manually restart the network>interface of my 4G router in order to have both connections bonded. I do that once a day i'd say, or when i need to download faster

Both routers have standard configurations and i do not mess with them. Maybe i'm missing something quite simple, but if you have any tips, please help

## Steps to Reproduce the Problem

  1. Test my internet speed -> connections are not bonded anymore
  2. Restart the network interface on WAN2 (my 4G router)
  3. Profit the powerful internet

## Specifications

  - OpenMPTCProuter version: 0.54.6
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: RPI3

Ysurac, 2020-03-22:
When you have the problem, check in Network->MPTCP and MPTCP fullmesh tab that all wan IPs are here, check also using the ```multipath``` command via SSH on the router that all WANs interfaces are set as default (everything else should be set as deactivated). Check also in State->System Log and kernel log that you have no errors.

G000R000, 2020-03-22:
Hello,
I have the problem right now (seeing only 6mb on speedtest (it's my dsl limit) :

- MPTCP and MPTCP fullmesh tab that all wan IPs are here : both connections IP are there 
[image]

 - system log :  
[image]

- kernel log : not sure what to give you, as theres a lot of lines
[image]

- command via SSH on the router that all WANs interfaces are set as default : 
[image]

I can restart the interface to give you the "after" if  you want, or more details if needed

Ysurac, 2020-03-22:
I see no errors...
You can try another Multipath TCP scheduler in Network->MPTCP like BLEST.

gnosoz, 2020-03-22:
i am finding the best results for hybrid connections 4g + cabled to be
achieved by fullmesh + ECF + bbr

G000R000, 2020-03-22:
Thanks to you both, i'll try BLEST and ECF : here are my new settings (i'll try BLEST later), i can keep you posted if you want.

Maybe something else in wrong in these settings ? WAN1 is DSL, WAN2 is 4G :

[image]

[image]

G000R000, 2020-03-28:
After a few days, i tried both TCP scheduler but it didn't work, i still have to restart the wan2 interface manually to have both connections "up"

closed 2020-03-28 by G000R000