Ysurac/openmptcprouter #920: Cant't add a new port forward

artiflo, 2020-03-22:
## Expected Behavior

After upagrade to 0.54.7 and 0.1015 I cant't add a new port forward.

## Actual Behavior

Can't add a new port forward because when I try to select an Internal IP Adress I have object call [object DocumentFragment]

Same issue when I try to edit a existent one.
[image]

## Steps to Reproduce the Problem

  1. Go to Network> Firewall > Port Forwards
  2. Click on add to add new port Forward
  3. Click on dropdown menu "Internal IP adress"

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: RPI3/x86_64)

Ysurac, 2020-03-22:
Refresh browser cache.

artiflo, 2020-03-22:
Sorry, indeed that was browser cache.

closed 2020-03-22 by artiflo