Ysurac/openmptcprouter #921: Bypass Houseparty and Facetime

iditude, 2020-03-22:
Hello,

In times when we're all stuck at home due to COVID19 (stay safe guys), is there a way to direct houseparty and facetime app to some wan (ADSL, to avoid burning the 4g allowance)? 

I can't find them in the list of application of OMR Bypass and it sounds like each of them  use complex ports / IP that can't easily be blocked. 

Thanks

gnosoz, 2020-03-22:
you could rate limit per port / interface therefore closing down the pipe
on the 4g connection for the UDP facetime ports only.

   - 80 (TCP) no limit
   - 443 (TCP) no limit
   - 3478 through 3497 (UDP)
   - 5223 (TCP)
   - 16384 through 16387 (UDP)
   - 16393 through 16402 (UDP)

https://techhub.hpe.com/eginfolib/networking/docs/switches/K-KA-KB/15-18/5998-8160_ssw_mcg/content/ch05.html

not sure if this works, haven't tried it yet on openmptcprouter and maybe
with qos/shorewall you can just achieve the desired result.

it will affect any other applications that might use those ports as well.

maybe with shorewall is easier as you could create a rule where all traffic
to/from those ports is allowed only through an interface instead of the
group used by aggregation.

iditude, 2020-03-24:
Thanks. That probably works. I ended up blacklisting the entire device as the above is a bit beyond my capabilities 

Cheers

closed 2020-03-24 by iditude