Ysurac/openmptcprouter #922: VPS IPv6

bambuca, 2020-03-22:
## Expected Behavior

Working IPv6 connections from VPS.

## Actual Behavior

After restarting VPS there is limited time when IPv6 works. I can ping6 ipv6.google.com and "ip -6 neigh" show me that the state is REACHABLE. After few seconds it turns into STALE, DELAY, PROBE, FAILED.

```
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router REACHABLE
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router REACHABLE
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router REACHABLE
kl7788as@enigma:~$ date
Sun Mar 22 23:35:39 CET 2020
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router STALE
kl7788as@enigma:~$ date
Sun Mar 22 23:35:44 CET 2020
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router STALE
kl7788as@enigma:~$ date
Sun Mar 22 23:35:46 CET 2020
kl7788as@enigma:~$
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router STALE
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router DELAY
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router PROBE
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18 lladdr 00:00:5e:00:02:01 router PROBE
kl7788as@enigma:~$ ip -6 neigh
2a01:430:200:16::1 dev ens18  router FAILED
```

## Steps to Reproduce the Problem

  1. Restart VPS
  2. Wait few second
  3. Try to ping via IPV6.

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: RPI4

Ysurac, 2020-03-23:
Who is the VPS provider ?
Check that shorewall6 on the VPS is properly configured (like /etc/shorewall6/params.net with the correct interface).

bambuca, 2020-03-23:
VPS is maste.cz (czech rep)
/etc/shorewall6/params.net contains (same content as /etc/shorewall/params.net)
``
NET_IFACE=ens18
``

Ysurac, 2020-03-23:
Check ```dmesg``` on the VPS if something is blocked and shouldn't be.
You can try to to disable shorewall6 to check if it's the problem (```systemctl stop shorewall6```).

bambuca, 2020-03-23:
dmesg contains these lines with ipv6:
```

[   12.903270] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:00:5e:00:02:01:86:dd SRC=2a01:0430:0200:0016:0000:0000:0000:0001 DST=2a01:0430:0200:0016:0000:0000:0000:0010 LEN=72 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=135 CODE=0 
[   13.903232] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:00:5e:00:02:01:86:dd SRC=2a01:0430:0200:0016:0000:0000:0000:0001 DST=2a01:0430:0200:0016:0000:0000:0000:0010 LEN=72 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=135 CODE=0 
[   14.903294] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:00:5e:00:02:01:86:dd SRC=2a01:0430:0200:0016:0000:0000:0000:0001 DST=2a01:0430:0200:0016:0000:0000:0000:0010 LEN=72 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=135 CODE=0 
[   61.123604] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:01:e8:d8:4b:f1:86:dd SRC=fe80:0000:0000:0000:0201:e8ff:fed8:4bf1 DST=fe80:0000:0000:0000:e88c:54ff:fe35:b235 LEN=64 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=136 CODE=0 
[   62.143722] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:01:e8:d8:4b:f1:86:dd SRC=fe80:0000:0000:0000:0201:e8ff:fed8:4bf1 DST=fe80:0000:0000:0000:e88c:54ff:fe35:b235 LEN=64 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=136 CODE=0 
[   63.174205] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:01:e8:d8:4b:f1:86:dd SRC=fe80:0000:0000:0000:0201:e8ff:fed8:4bf1 DST=fe80:0000:0000:0000:e88c:54ff:fe35:b235 LEN=64 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=136 CODE=0 
[   66.113992] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:01:e8:d8:4b:f1:86:dd SRC=fe80:0000:0000:0000:0201:e8ff:fed8:4bf1 DST=fe80:0000:0000:0000:e88c:54ff:fe35:b235 LEN=72 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=135 CODE=0 
[   67.113945] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:01:e8:d8:4b:f1:86:dd SRC=fe80:0000:0000:0000:0201:e8ff:fed8:4bf1 DST=fe80:0000:0000:0000:e88c:54ff:fe35:b235 LEN=72 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=135 CODE=0 
[   68.113975] Shorewall:net-fw:DROP:IN=ens18 OUT= MAC=ea:8c:54:35:b2:35:00:01:e8:d8:4b:f1:86:dd SRC=fe80:0000:0000:0000:0201:e8ff:fed8:4bf1 DST=fe80:0000:0000:0000:e88c:54ff:fe35:b235 LEN=72 TC=224 HOPLIMIT=255 FLOWLBL=0 PROTO=ICMPv6 TYPE=135 CODE=0 
```
Then I disabled shorewall6 and IPv6 is working.

Ysurac, 2020-03-23:
This looks like ICMPv6 is blocked from net to fw, but there is a rule that allow that in /etc/shorewall6/rules...

bambuca, 2020-03-23:
this is my rules:
```
#
# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
# Copyright (C) 2006-2014,2007 by the Shorewall Team
#
# This library is free software; you can redistribute it and/or
# modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation; either
# version 2.1 of the License, or (at your option) any later version.
#
# See the file README.txt for further details.
#------------------------------------------------------------------------------
# For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    vpn             net
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#	Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#	Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#	DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT)	$FW		vpn
#
#	Redirect all port from 1 to 64999 to the VPN client from the network
#
#DNAT		net		vpn:10.0.0.2	tcp	1-64999

```

Ysurac, 2020-03-23:
It's /etc/shorewall/rules, for IPv6 it's /etc/shorewall6/rules

Ysurac, 2020-03-23:
(if it's really rules in shorewa6 then replace with this one: https://www.openmptcprouter.com/server/shorewall6/rules )

bambuca, 2020-03-23:
It was shorewall6/rules. I replaced it with your file and now it works like a charm.
Thank you very very very much.

closed 2020-03-23 by bambuca