Ysurac/openmptcprouter #923: Bonding based on WAN weight

ccmks, 2020-03-23:
Hello,

I don't know if this feature has been implemented. If so, tell me how to enable it. If not, then you can put this as future request!

I have 4 WAN are connected to OpenMPTCPRouter.
WAN1 - Fiber 100/100 Mbps
WAN2 - Fiber 30/6 Mbps
WAN3 - LTE Speed varies depend on the congestion, average speed 25/25 Mbps
WAN4 - LTE Speed varies depend on the congestion, average speed 20/16 Mbps

VPS speed - 1Gbps VULTR

So, when I am bonding all of the WAN, I am expecting the speed will be added up, up to 90% of the actual total all WAN combined. However, I am not seeing that happening

When the bonding happen, sometimes I noticed , sometimes the following WAN only getting:
WAN1 - 30 Mbps
WAN2 - 20 Mbps
WAN3 - 45 Mbps (during best hour)
WAN4 - 16 Mbps

The rest of WAN is under utilized the bandwidth available and WAN3 take the most of bonding weight.

Basically, the weight on bonding is not proportion based on the respective WAN speed. How can I adjust the weight.

In this case, my LTE is not unlimited and Fiber connection is unlimited. If LTE is being utilized to its  full bandwidth, whereas the fiber isn't, then I could be quickly running out of data allowance on my LTE. How can I adjust, that if WAN1 and WAN2 weight is higher then the rest of my WAN, so when MPTCP happen, they will prioritize those 2 fiber WAN over the LTE WAN.

I used to have bonding software before and they allow you to adjust the weight on which WAN you want to utilize the most.

openmptcprouter v0.54.7 r0+12509-04a21c26a0
VPS Version 0.1015 4.19.104-mptcp
Platform: Hyper-V core 2019 on custom made hardware with Quad port NIC

Thank you

Ysurac, 2020-03-23:
OpenMPTCProuter doesn't do bonding at all. It use MPTCP that aggregate connections.
It use connection with lowest latency first, then the connection with a bigger latency,...
There is, for now, no way to prioritize a connection.

In Network->MPTCP you can try another Multipath TCP scheduler like ECF or BLEST.

ccmks, 2020-03-23:
Aggregate and bonding are the same thing, isn't it?

What is the difference between ECF, BLEST or Default?

Ysurac, 2020-03-23:
No it's not exactly the same thing. When we talk about bonding, this often refer to Link Bonding 802.3ad that doesn't work the same way at all. With MPTCP a single TCP connection can be balanced on multiple links.

For TCP scheduler, you have to test or read the code. BLEST and ECF should be better when there is an aggregated link that loose some packets.

gnosoz, 2020-03-23:
throughput with MPCTP should consists of the sum of the aggregated lines
or am I mistaken?

Ysurac, 2020-03-23:
True

ccmks, 2020-03-24:
That's my understanding of bonding, it sums up all line bandwidth.

How many maximum internet can be aggregated per instance?

Ysurac, 2020-03-24:
8

ccmks, 2020-03-24:
What is the recommended hardware sizing (specs) if I want to aggregate 8 ISP with each respective speed 100 Mbps up and down?

Ysurac, 2020-03-24:
For 800Mb/s a CPU with AES hardware support and a quite good CPU. I don't exactly know, I don't have enough bandwidth to test.

ccmks, 2020-03-25:
How about the memory?

Ysurac, 2020-03-25:
1024Mo should be enough

ccmks, 2020-03-26:
Can you tell me the description or explanation on each option for the following picture that I circled in red?
[image]

I might be able to select the best option based on the suitable explanation.

cron410, 2020-04-04:
@ccmks did you test them all out? Let us know what you find.

ccmks, 2020-04-05:
> @ccmks did you test them all out? Let us know what you find.

I tested them, but I couldn't tell the difference since they were all almost have same behavior.

gnosoz, 2020-04-05:
Congestion control and packet scheduling are two important components for
MPTCP design. Congestion control is in charge of controlling induced
network load, while packet scheduling is responsible for the distribution
of data over multiple paths and improper scheduling decisions might
introduce higher delay. Therefore, congestion control algorithms and packet
schedulers are two components which greatly impact the performance of
MPTCP.

Full article here:https://helda.helsinki.fi/handle/10138/228831

closed 2020-09-14 by github-actions[bot]