Ysurac/openmptcprouter #924: Missing BLEST

kevin39, 2020-03-23:
## Expected Behavior

Having BLEST congestion control

## Actual Behavior
No BLEST. Other methods are still available : reno, bbr, cubic, lia, olia, wvegas, balia, mtcpdesync

## Steps to Reproduce the Problem

I was using raspberry 3. I switched the sd card to a fresh new Rpi 4. Everything is workine fine but BLEST congestion control is now missing. Is it a normal behavior on Rpi 4 ?
I didn't updated to the last version because everything was working fine but on the Rpi 4, BLEST was available.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.52.3 r0+11006-713561a10b
  - OpenMPTCProuter VPS version: 0.1001 4.19.67-mptcp
  - OpenMPTCProuter platform: Raspberry Pi 4 Model B Rev 1.2

Ysurac, 2020-03-23:
BLEST is a Multipath TCP scheduler not a TCP Congestion Control like bbr, olia,...

kevin39, 2020-03-23:
My bad ! You're right.

Sorry.

closed 2020-03-23 by kevin39