Ysurac/openmptcprouter #925: omr-tracker-ss : Failed to establish connection after Fullmesh subflows for each pair of IP addresses

xabix, 2020-03-23:
## Expected Behavior

Hello,

I have been doing some research to fine tune my setup and decided to increase the "Fullmesh subflows for each pair of IP addresses" from 1 (default) to 4.

The article I read: https://netsysci.cut.ac.cy/wp-content/uploads/2017/09/122943_mptcp.pdf. Maybe as I have 3 WANs I should be 3 and not 4. At least increasing this value seems a good idea :)

My config is with 3 WANs (2 ADSL & 1 LTE).

Since then I have no drawing in Status but also the following errors:

`Mon Mar 23 13:52:03 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 212.27.48.10)
Mon Mar 23 13:52:05 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan3 (wan3)
Mon Mar 23 13:52:05 2020 user.notice mptcp: Reloading mptcp config due to ifup of loopback (lo)
Mon Mar 23 13:52:05 2020 user.notice SQM: Starting SQM script: simple.qos on tun0, in: 0 Kbps, out: 0 Kbps
Mon Mar 23 13:52:05 2020 user.notice SQM: Using generic sqm_start_default function.
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Mon Mar 23 13:52:05 2020 daemon.warn [7498]: <warn>  Invalid sysfs path read for net/TMP_IFB_4_SQM
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: add network interface TMP_IFB_4_SQM: event processed
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: event reported: action=add, name=TMP_IFB_4_SQM, subsystem=net
Mon Mar 23 13:52:05 2020 daemon.warn [7498]: <warn>  Invalid sysfs path read for net/TMP_IFB_4_SQM
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: remove network interface TMP_IFB_4_SQM: event processed
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=TMP_IFB_4_SQM, subsystem=net
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: add network interface ifb4tun0: event processed
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Mon Mar 23 13:52:05 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Mon Mar 23 13:52:05 2020 user.notice SQM: simple.qos was started on tun0 successfully
Mon Mar 23 13:52:05 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Mon Mar 23 13:52:07 2020 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: {"result":"done","reason":"no changes","route":"shadowsocks"}
Mon Mar 23 13:52:08 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:52:08 2020 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: {"result":"done","reason":"changes applied"}
Mon Mar 23 13:52:10 2020 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: {"result":"done","reason":"changes applied"}
Mon Mar 23 13:52:12 2020 daemon.info procd: - init complete -
Mon Mar 23 13:52:12 2020 daemon.info vnstatd[20264]: Info: vnStat daemon 2.4 started. (pid:20264 uid:0 gid:0 64-bit)
Mon Mar 23 13:52:12 2020 daemon.info vnstatd[20264]: Info: Monitoring (10): wan3 (1000 Mbit) wan2 (1000 Mbit) wan1 (1000 Mbit) tun0 (10 Mbit) teql0 (1000 Mbit) ifb4wan2 (1000 Mbit) ifb4wan1 (1000 Mbit) ifb1 (1000 Mbit) ifb0 (1000 Mbit) eth0 (1000 Mbit)
Mon Mar 23 13:52:15 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Mon Mar 23 13:52:16 2020 user.notice post-tracking-post-tracking: Get status and settings for wan1... Failed
Mon Mar 23 13:52:17 2020 user.notice post-tracking-post-tracking: Get status and settings for wan3... Failed
Mon Mar 23 13:52:20 2020 user.notice post-tracking-post-tracking: Set server vps (149.xx.xx.xx) default route via 192.168.2.1
Mon Mar 23 13:52:23 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:52:38 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:52:54 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:53:09 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:53:24 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:53:39 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:53:53 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Mon Mar 23 13:53:54 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:54:09 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:54:20 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Failed
Mon Mar 23 13:54:24 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:54:39 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:54:54 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:55:07 2020 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Mon Mar 23 13:55:09 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:55:20 2020 user.notice post-tracking-post-tracking: Get status and settings for wan3...
Mon Mar 23 13:55:25 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:55:34 2020 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Mon Mar 23 13:55:40 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection
Mon Mar 23 13:55:44 2020 user.notice post-tracking-post-tracking: Get status and settings for wan1...
Mon Mar 23 13:55:48 2020 user.notice post-tracking-post-tracking: Get status and settings for wan3... Failed
Mon Mar 23 13:55:55 2020 daemon.err omr-tracker-ss[14577]: Failed to establish connection`

## Steps to Reproduce the Problem

  1.  "Fullmesh subflows for each pair of IP addresses" from 1 (default) to 4.

## Specifications

  - OpenMPTCProuter version: v0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: x86_64

xabix, 2020-03-23:
[image]
[image]

xabix, 2020-03-23:
Not sure why but the Default VPN was set to Dead Simple VPN. I put Glorytun TCP and it works back.
[image]

Any idea of the issue and why it didn't work with DSVPN?

Ysurac, 2020-03-23:
no

xabix, 2020-03-23:
So the logs shared don't help to identify the issue. From a latency perspective what would be your recommendation as of the "Default VPN"

closed 2020-03-24 by xabix