Ysurac/openmptcprouter #928: Difference and Use of ip.openmptcprouter.com vs hostname.openmptcprouter.com

xabix, 2020-03-24:
## Expected Behavior

Hello Ysurac,

I see in my Pi-hole on the VPS a lot of queries to ip or hostname @ openmptcprouter.com.
I assume one of them is for OMR to check for a newer release but I am surprised to see both of them on my top 3.

Any explication of the difference between ip. and hostname. and is it correct that they both go to the same IP or could it be that something is not right in my setup.

[image]

> dig hostname.openmptcprouter.com
> ; <<>> DiG 9.16.0-Ubuntu <<>> hostname.openmptcprouter.com
> ;; global options: +cmd
> ;; Got answer:
> ;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 39266
> ;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1
> ;; OPT PSEUDOSECTION:
> ; EDNS: version: 0, flags:; udp: 65494
> ;; QUESTION SECTION:
> ;hostname.openmptcprouter.com.  IN      A
> ;; ANSWER SECTION:
> hostname.openmptcprouter.com. 4634 IN   A       94.23.252.192
> ;; Query time: 0 msec
> ;; SERVER: 127.0.0.53#53(127.0.0.53)
> ;; WHEN: mar. mars 24 17:02:37 CET 2020
> ;; MSG SIZE  rcvd: 73

> dig ip.openmptcprouter.com
> ; <<>> DiG 9.16.0-Ubuntu <<>> ip.openmptcprouter.com
> ;; global options: +cmd
> ;; Got answer:
> ;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 44107
> ;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1
> ;; OPT PSEUDOSECTION:
> ; EDNS: version: 0, flags:; udp: 65494
> ;; QUESTION SECTION:
> ;ip.openmptcprouter.com.                IN      A
> ;; ANSWER SECTION:
> ip.openmptcprouter.com. 5186    IN      A       94.23.252.192
> ;; Query time: 0 msec
> ;; SERVER: 127.0.0.53#53(127.0.0.53)
> ;; WHEN: mar. mars 24 17:02:45 CET 2020
> ;; MSG SIZE  rcvd: 67

## Specifications

OpenMPTCProuter version: v0.54.7
OpenMPTCProuter VPS version: 0.1015
OpenMPTCProuter platform: x86_64

Ysurac, 2020-03-24:
ip.openmptcprouter.com is used to get IP of WANs and to get actual exit IP from the router.
hostname.openmptcprouter.com is used to get the reverse from the API script, this is used by obfuscation. I will remove this call and add it as variable in API conf.

xabix, 2020-03-24:
Thank you so it's all OK with my setup (aka getting the same DNS resolution on both) as well as in terms of frequency?

Have a good evening

Ysurac, 2020-03-24:
Yes, no problems.

closed 2020-03-25 by xabix