Ysurac/openmptcprouter #929: Plusieurs problèmes avec openmptcrouter

jeandelamarquise, 2020-03-24:
## Expected Behavior

I'm trying to put an ip adress on specific device (like amazon fire stick) because netflix and prime video blocked me with ovh... (i think)
Need to find solution for netflix, prime video and freebox tv.

## Actual Behavior
I think i found where to do so but openmptcrouter said change not apply. I click on the button and wait... but nothing happened and there is always some changes not applied.

## Steps to Reproduce the Problem

  1. put a mac adress with specific ip with specific wlan
  2. click on change not apply
  3. not working...

## Specifications

  - OpenMPTCProuter version:  v0.54.2 r0+12464-a017773a92
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: openmptcprouter v0.54.2 r0+12464-a017773a92

jeandelamarquise, 2020-03-24:
[image]
[image]
[image]
[image]

Ysurac, 2020-03-24:
The filesystem is read only so changes can't be saved.
This happen when filesystem is corrupted after power is unplugged or when SD card is died.
You can try a reboot, if it can be repaired this will be done.

jeandelamarquise, 2020-03-24:
Ok, i reboot and dont have the filesystem read only.I don't find how to Do for Netflix, prime video and freebox tv and my fire stick.Can you help me ? In french ? :DThanks for all. This changes my life :) need to steak it with your help.Billy

Ysurac, 2020-03-24:
Use OMR-ByPass and set Protocol/Services. OpenMPTCProuter must be used as DNS on clients.

jeandelamarquise, 2020-03-24:
I think that and i try to out for example my firestick on the same lan adresses that my freebox and set to wlan1.I put à screenshot about that. Is that ok ?How can i do to make freebox tv to work ?

closed 2020-09-14 by github-actions[bot]