Ysurac/openmptcprouter #930: No Server ping response after 1 second on LTE while ping is OK

xabix, 2020-03-24:
## Expected Behavior

Hello (again sorry)

Here is my setup:
[image]

## Actual Behavior

Not sure why I have 2 entries with different metrics for 192.168.2.1
> default via 10.255.255.1 dev tun0 
> default via 192.168.2.1 dev wan1 metric 3 
> default via 192.168.9.1 dev wan3 metric 4 
> default via 192.168.3.1 dev wan2 metric 5 
> 10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
> 127.0.0.0/8 dev lo proto static scope link metric 1 
> 149.91.xx.xx via 192.168.2.1 dev wan1 metric 1 
> 149.91.xx.xx via 192.168.2.1 dev wan1 metric 3 
> 149.91.xx.xx via 192.168.9.1 dev wan3 metric 4 
> 149.91.xx.xx via 192.168.3.1 dev wan2 metric 5 
> 192.168.2.0/24 dev wan1 scope link metric 3 
> 192.168.3.0/24 dev wan2 scope link metric 5 
> 192.168.9.0/24 dev wan3 scope link metric 4 

system log:
[image]

kernel log:
[image]

Wireshark says:
[image]

icmp is OK:
[image]

## Steps to Reproduce the Problem

Not sure

## Specifications

-OpenMPTCProuter version: v0.54.7
-OpenMPTCProuter VPS version: 0.1015
-OpenMPTCProuter platform: x86_64

Ysurac, 2020-03-24:
The ICMP in your trace is the test of the gateway, not to the server.
This happen that some packets get lost on 4G, this is not a big problem, only an information.

xabix, 2020-03-24:
How can I test manually then the ICMP to the server to try to understand why there is packet loss?
Is a ping to my VPS?

By the way shouldn't I move to a tcp type of connection over 4G to avoid packet loss interfering?

Ysurac, 2020-03-24:
It's a ping to the VPS from the interface specified (so a ```ping -B -I wan3 ipoftheserver```).
I don't understand what you mean with tcp over 4G.
ICMP ping lost is not a big problem, 4G will always lose some packets.

xabix, 2020-03-24:
I think there is something more than packet loss, no?

> root@OpenMPTCProuter:~# ping -B -I wan3 149.91.xx.xx
> PING 149.91.88.173 (149.91.88.173) from 192.168.9.2 wan3: 56(84) bytes of data.
> ^C
> --- 149.91.88.173 ping statistics ---
> 17 packets transmitted, 0 received, 100% packet loss, time 16563ms

I was thinking that as a keep alive a TCP type of connection would be more stable with retransmission than ICMP or UDP, so I was wondering if there was another way to ping the server than ICMP which could potentially overcome such messages.

Ysurac, 2020-03-24:
In omr-tracker you can choose what you want.

xabix, 2020-03-24:
100% packet loss is strange. I have selected httping but this the same message. If I have these messages, I assume the WAN is not being used at OMR is considering that the server is not responding right? So I am running on 2 WANs vs 3 ?

Ysurac, 2020-03-24:
As I said, the message in status is only an information.
Multipath is not disabled on wan3, so the interface is used.

xabix, 2020-03-24:
ok merci. Not sure of the utility of this message then if it s not affecting anything. It seems we can't deactivate this to avoid such messages.

THe comment says: Always ping gateway, then test connection by ping, httping or dns. None mode only ping gateway.
What is the interest to select ping if we always ping the gateway, you mention that this is a ping to the server and not the gateway. Is there a way to only ping the gateway but not the server? or this doesn't make sense?

closed 2020-03-24 by xabix

Ysurac, 2020-03-24:
The warning is only an information, losing too much ICMP is still not normal. But it's only a warning that can help for debug.
OMR-Tracker ping gateway by default and IPs if gateway is down. If for a strange reason you only want to ping gateway then select none, but it's not a very good idea.

xabix, 2020-03-24:
ok thank you so much. Is there a way for me to debug further to understand why so much loss? As I have a special antenna and not so far from the radio station i am surprised. If I reboot I think it will get back but would like to ensure the pb is really on the radio or my modem beforehand.

Ysurac, 2020-03-24:
As it's 100% loss with your ping, no idea. Maybe ICMP is filtered...

xabix, 2020-03-24:
thus while I wanted to try another protocol :)

xabix, 2020-03-24:
Ok I ll reboot to check!

xabix, 2020-03-24:
Look after reboot OMR all fine again
--- 149.91.88.173 ping statistics ---
9 packets transmitted, 0 received, 100% packet loss, time 8332ms

root@OpenMPTCProuter:~# reboot
root@OpenMPTCProuter:~# 

BusyBox v1.31.1 () built-in shell (ash)

   ___                 __  __ ___ _____ ___ ___              _           
  / _ \ _ __  ___ _ _ |  \/  | _ \_   _/ __| _ \_ _ ___ _  _| |_ ___ _ _ 
 | (_) | '_ \/ -_) ' \| |\/| |  _/ | || (__|  _/ '_/ _ \ || |  _/ -_) '_|
  \___/| .__/\___|_||_|_|  |_|_|   |_| \___|_| |_| \___/\_,_|\__\___|_|  
       |_|                                                               
------------------------------------------------------------------------------
   (r0+12509-04a21c26a0)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.54.7

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Sat Mar 21 18:10:50 UTC 2020
-----------------------------------------------------
root@OpenMPTCProuter:~# ping -B -I wan3 149.91.88.173
PING 149.91.88.173 (149.91.88.173) from 192.168.9.2 wan3: 56(84) bytes of data.
64 bytes from 149.91.88.173: icmp_seq=1 ttl=54 time=27.0 ms
64 bytes from 149.91.88.173: icmp_seq=2 ttl=54 time=24.9 ms
^C
--- 149.91.88.173 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 24.928/25.977/27.027/1.049 ms