Ysurac/openmptcprouter #931: Disneyplus OMR Bypass

zicochaos, 2020-03-24:
Any way to bypass it?  tried with this AS

ASN11251
ASN25932
ASN22604
ASN8137

but stills is not working :(

Ysurac, 2020-03-24:
I can't make tests on that because it's not yet available in France.

rubaz100, 2020-03-24:
You don't have to bypass it. I use it without bypass. It works fine for me with my vps / vpn

gnosoz, 2020-03-24:
I am on it ;)
it seems that both the app and the browser contact differenti ips for the
negotiation / auth / geolocalizzazione than the country ASNs

by firewall app i found that you need to allow following domains for italy:
Disney-plus.net
disneyplus.com
bamgrid.com

i attach pictures for your reference.

gnosoz, 2020-03-24:
i have my server in another country and i needed to bypass, maybe same for
him?

now it works fine with the conf i provided that is for Italy, not sure
other countries setup as domains might be different...

zicochaos, 2020-03-24:
hey 
adding this three domains resolves issue :) Thanks guys

gnosoz, 2020-03-24:
Glad it helps!
Enjoy disneyplus ;)

Ysurac, 2020-03-25:
I will add them as service for next release.

victorleduc, 2020-04-07:
Hi Ysurac, Disney+ is now available in France, and I tried to subscribe but I have the following error :

POST https://global.edge.bamgrid.com/token 400
{"error":"unauthorized_client","error_description":"unreliable-location"}

I tried adding to the OMR-Bypass the domains listed here : https://support.opendns.com/hc/en-us/articles/360037591112-Domains-to-Allow-for-Disney-Plus which contains the bamgrid.com we're talking about.
I also added the ASN listed above.

Nothing fixed it :/

I tried to bypass to my 2 connexions which are a VDSL Free and 4g Free mobile and should work fine.

If you want to test you don't have to subscribe, just the homepage https://www.disneyplus.com/ will get you (maybe) the error.

Do you have any insite for this issue ?

Keep up the good work, I'm using OMR for more than 1 year and i'm very happy :)

Ysurac, 2020-04-07:
I tested by adding domains Disney-plus.net, disneyplus.com and bamgrid.com to omr-bypass and it's working (no more the token error). Check that you are using OpenMPTCProuter IP as DNS and that browser DNS cache is empty.

gnosoz, 2020-04-07:
on a more positive note i discovered that the geo-token is require on login
only and after that is done you can switch to full on MPTCP.

this test has been done on mobile app:
  - phone 4g
 -  login to the app
 - activate wifi
-  Enjoy full speed ;)

I will try to figure out which domani needs to be on omr-bypass so all can
enjoy full speed all time.

victorleduc, 2020-04-08:
Hi !

Thanks I just had to set my OMR IP as DNS on my computer and it did the tricks !

For information for others, at the end I only added **bamgrid.com** to the bypass and it is enough as you says @gnosoz, i was able to subscribe and watch!

jeandelamarquise, 2020-04-08:
Hi all,

Can someone explain me how to Do that ?
I added the 3 domain in one-bypass but it's not working.

Ysurac, 2020-04-08:
@jeandelamarquise OpenMPTCProuter IP must be used as DNS. You also have to reset local DNS browser cache (restart it).

jeandelamarquise, 2020-04-08:
@Ysurac 
Je suis désolé. Je ne maîtrise pas encore tout et je ne comprends pas comment faire tout ça. Peux tu me guider pas à pas ?

victorleduc, 2020-04-08:
Salut, tu as juste besoin du domaine bamgrid.com dans OMR-Bypass.
Ensuite dans les réglages de ton ordinateur, ou de l'appareil qui va accéder à Disney, tu dois indiquer comme premier serveur DNS l'IP locale de ton installation OMR, qui est l'IP que tu vois dans ta barre d'URL de ton navigateur quand tu est en train de configurer OMR. (Surement quelque chose en 192.168.x.x)

Après pour ajouter aux DNS de ton appareil ca dépend de ton système. Tu peux par exemple trouver une explication pour Windows ici : https://le-routeur-wifi.com/dns-google-opendns-cloudflare-changer-windows/
Tu dois mettre comme DNS préféré l'IP de ton OMR, et en auxiliaire tu peux par exemple mettre 8.8.8.8 qui est le DNS de google

jeandelamarquise, 2020-04-09:
@victorleduc 
Merci pour l'explication. J'aimerais configurer Disney+ sur un firestick d'Amazon.
J'ai réussi pour netflix et amazon prime vidéo.
J'ai tenté comme tu le dis d'entrer en dns 1 l'adresse ip de mon omr mais ça met une adresse bizarre (ipv6 ?)
J'ai encore besoin d'aide

victorleduc, 2020-04-09:
Tu peux utiliser ce tuto pour changer ton adresse DNS sur un firestick : https://proprivacy.com/guides/change-dns-firestick

Je n'en ai pas trouvé en français j'espère que ca ira.

Je n'ai pas compris pour l'histoire de l'ipV6, si tu suis le tuto il va bien te demander une ipv4 normalement

jeandelamarquise, 2020-04-09:
@Ysurac 
Je viens de penser à un truc. J'ai installé  openmptcprouter v0.54.7 et je viens de voir dans un autre post qu'il faut mettre à jour le vps. Est-ce que c'est pour ça que je n'y arrive pas ?

jeandelamarquise, 2020-04-09:
> Tu peux utiliser ce tuto pour changer ton adresse DNS sur un firestick : https://proprivacy.com/guides/change-dns-firestick
> 
> Je n'en ai pas trouvé en français j'espère que ca ira.
> 
> Je n'ai pas compris pour l'histoire de l'ipV6, si tu suis le tuto il va bien te demander une ipv4 normalement

L'anglais ça peut aller.
J'ai réussi à trouver où changer les dns mais au lieu de mettre 192.168.10.1 (mon omr), il met une ip 2a01:e00::1 à la place

jeandelamarquise, 2020-04-09:
Merci à vous tous ! Ça fonctionne !! Je sais pas si c'était le coup du vps à updater mais ça fonctionne :)
Merci @Ysurac pour cet omr !!!

Chatissimus, 2020-04-11:
Bonjour, 
vous avez vu passer quelque chose pour changer les DNS des chromecast ? je trouve pas.

j'ai essayé pour Netflix , blocage des DNS google, route statique vers le DNS OMR, rien de concluant, netflix me détecte toujours sur OVH via la chromecast (et me bloque) 
pas de problème via d'autre appareils

Ysurac, 2020-04-11:
Pour bloquer les DNS de google depuis un chromecast ayant l'IP 192.168.100.6, il faut ajouter dans Réseau->Parefeu puis onglet règles spécifiques : 
```
iptables -I FORWARD -s 192.168.100.6 --destination 8.8.8.8 -j REJECT
iptables -I FORWARD -s 192.168.100.6 --destination 8.8.4.4 -j REJECT
```
Normalement il utilise les DNS distribués par DHCP. Sinon il doit être possible aussi de rediriger vers l'IP du routeur au lieu de celles de google via iptables.

Chatissimus, 2020-04-11:
ok merci, je vais regarder

j'ai déjà mis des routes statique entre les IP des DNS google et celle du routeur, sur le routeur qui gère mon réseau local et sur l'OMR, et ca marche, mais pas tout le temps
je suis perplexe sur le "pas tout le temps"

closed 2020-09-14 by github-actions[bot]