Ysurac/openmptcprouter #932: DHCP exclude MAC

gaalos, 2020-03-25:
## Expected Behavior
Hi want to exlude MAC adress from DHCP openmptcprouter
For exemple i'm using Orange Router with Orange TV
My Orange TV must use ORANGE Router only

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0

Maybe it's request feature like 
Static DHCP with other gateway 
Or Exclude mac 

I can use external DHCP server but i prefer manage it there

Best regards

bnhf, 2020-03-26:
Having multiple gateways on the same LAN requires some specialized setup on your part:

First, you don't want more than one active DHCP server, so turn off DHCP on your Orange router. Second, set a static address on the Orange TV that's within the range of the Orange router's subnet.

Do those two things and your Orange TV will use the Orange router, and everything else on your network will get IP addresses from the DHCP server on your OpenMPTCProuter.

gaalos, 2020-03-26:
I can't use static ip on my orange TV :/ and the orange TV refuse other DHCP :/ 
Best regards

bnhf, 2020-03-26:
Maybe you could do something like this:

https://bokunokeiken.wordpress.com/2015/06/27/how-to-block-device-on-openwrt-based-on-mac-address/

But aren't you still going to have a problem with the Orange router's DHCP competing with the OpenMPTCProuter's DHCP? You didn't say, but I'm assuming everything is Ethernet connected here?    The other option that comes to mind would be to use a managed switch and VLANs to keep your traffic separated.

gaalos, 2020-03-26:
@bnhf 
unfortunatly i don't have L2 switch ^^ 
The orange box block all DHCP except the orange router (Yes this is Sh*t)

bnhf, 2020-03-26:
Worth a try to block the MAC address of the Orange TV on the OpenMPTCProuter using the technique in the link I sent you, no?

gaalos, 2020-03-26:
@bnhf this is exactly what i want ;) i'll reply soon ;)

cron410, 2020-04-04:
@gaalos please let us know how it worked out.

closed 2020-09-14 by github-actions[bot]