Ysurac/openmptcprouter #935: vnstat complains about tun0 speed of 10Mbit

xabix, 2020-03-26:
## Expected Behavior

Hello,

I have such error messages:
[image]

From this thrread it looks like some tweaking should be done to be solved:
https://github.com/vergoh/vnstat/issues/81

Is there anything to be done in OMR sw or should it be done manually?

Merci

Ysurac, 2020-03-26:
```tun0``` shouldn't be added to vnstat. You added it manually ?

xabix, 2020-03-26:
No, the issue came to the fact that I added a wan4 through "Setting Wizard" with SQM and then I deleted it but it looks like than in SQM menu the entry stayed but was then referring to tun0 instead of wan4.
Not a big issue but it took me some time to figure out and had instabilities.

closed 2020-03-26 by xabix