Ysurac/openmptcprouter #936: Odd disable/reset log messages

darthclide, 2020-03-26:
## Expected Behavior

Logs should only show a few extraneous "info" messages, in addition to my Verizon getting failed status setting messages every so often (the tower is far away, so I barely manage 3 bars)

## Actual Behavior
Every 1-2 hours I get the following message in the log:
[image]

## Steps to Reproduce the Problem

  1. Restart router
  2. Wait a little while

## Specifications

  - OpenMPTCProuter version:  version 0.54.5
  - OpenMPTCProuter VPS version: Version 0.1009 4.19.80-mptcp
  - OpenMPTCProuter platform: RPI4
  - Anker 4-Port Ultra Slim USB 3.0 Hub A7518 (this allows injection of power into the hub from an AC outlet)
  - ZTE Velocity 2
  - Verizon Jetpack 8800L
  - Motorola Moto E6
  - DSL plugged in using this USB adapter: 
https://www.amazon.com/Linksys-Ethernet-Chromebook-Ultrabook-USB3GIG/dp/B00LIW8TBG/ref=sr_1_3?crid=XE74SNCH5OTK&dchild=1&keywords=linksys+usb+3.0+ethernet+adapter&qid=1585263809&sprefix=linksys+usb+3.0%2Caps%2C164&sr=8-3

Jetpack 8800L and Motorola Moto E6 are plugged into the USB hub.
ZTE Velocity 2 and DSL (USB ethernet adapter) are plugged directly into the RPI4.

1. Is there a command I can use to translate this "usb 2-1" into a device name like "usb0" or "usb1"?
2. Any ideas on why it is trying to disable a device, and then reset it?

darthclide, 2020-03-27:
Extra info: Sometimes the log shows only 1 of these events occurring on it's own. Attached some photos to show:
[image]
[image]
[image]

At this point if you could confirm that this isn't hurting my hardware it would relieve me greatly... I don't want to wake up one day with a few or all of my devices bricked.

darthclide, 2020-03-28:
After some research this issue seems to be related to kernel issues. And while some places recommended switching off UAS and turning on usb-storage, I didn't want to take such drastic measures if all I need is an update. (My eeprom version is from July 2019 apparently)

I tried running:
rpi-eeprom-update -a
But I get the following result:
[image]

I found another user with this issue from early January 2020, and your response said it would be fixed in the next release. Version 0.54.5 came out well after early January, so could you help getting this update? If I am lucky, it will fix these resets? Or if you have a better idea on what is happening, I am all ears.

Ysurac, 2020-03-28:
Can you check if ```/usr/bin/rpi-eeprom-config``` exist or not ?

darthclide, 2020-03-28:
[image]
It would appear so? Can you confirm before we go further if you think this reset is potentially bricking any of my devices?

Ysurac, 2020-03-28:
It's a firmware update, like all firmware this can brick the device but there is some recovery way.
I updated mine without any issue, you only need to reboot to apply the new firmware after running the update script.

darthclide, 2020-03-28:
But why didn't the command come back and tell me to restart? It is acting like it didn't work.

Further, I was more worried about these "reset" messages. Do you think they are ruining my devices? I don't want to get my hopes up that I can stop the resets :/

Ysurac, 2020-03-28:
Should not, but you can try to remove device and add them one by one to check it's related to a device.

darthclide, 2020-03-28:
Well I am pretty sure it is the USB hub, but I was hoping you had more knowledge about why it would be resetting it. It is a powered HUB (looked for it specifically on ebay since many hubs don't allow you to inject power from an AC outlet) so it shouldn't be a lack of power.

darthclide, 2020-04-01:
I moved the HUB to the 2.0 interface and so far the errors are gone. I guess there needs to be a kernel update that allows you to plug both usb 2.0 and usb 3.0 devices into a usb 3.0 hub without causing havoc. That is out of your control though.

cron410, 2020-04-04:
Thanks for the thorough troubleshooting. Please close the issue @darthclide

darthclide, 2020-04-04:
Oh I apologize. I thought only administrators of a github page were allowed to do that.

closed 2020-04-04 by darthclide