Ysurac/openmptcprouter #937: VPN tunnel DOWN

chris77170, 2020-03-27:
## Expected Behavior
Connect VPS

## Actual Behavior
No connection to VPS

## Steps to reproduce the problem
No real steps, updates have been made on OMR and VPS
But the OMR indicates:
VPN tunnel DOWN
Can't get public IP address from ShadowSocks
Can't contact Server Admin Script
Version 0.1008 ...

## Characteristics

- OpenMPTCProuter version: 0.54.07
- OpenMPTCProuter VPS version: 0.1015
- OpenMPTCProut x86_64 platform

[daemon.log](https://github.com/Ysurac/openmptcprouter/files/4392775/daemon.log)
[image]
[OMR_journal_system.txt](https://github.com/Ysurac/openmptcprouter/files/4392800/OMR_journal_system.txt)

Scubad1v3r, 2020-03-27:
Same trouble on OMR "VPN tunnel DOWN"
But after rebooting VPS (Milkywan), VPS is good working and aggregate multiple Free and Free Mobile Internet connections are good working

Just "VPN tunnel DOWN" message on OMR is printed

Ysurac, 2020-03-27:
@chris77170 Try another VPN.
@Scubad1v3r what is the VPN used ?

chris77170, 2020-03-27:
I tested with the different VPNs.
Always the same answer despite restarting the OMR and waiting

Scubad1v3r, 2020-03-27:
I don't know what VPN is normal used - I've change anything from initial config

Ysurac, 2020-03-27:
Can you set to "glorytun-tcp" as VPN in the wizard on the router and give me on the VPS the result of ```ip a``` I think VPN doesn't get IP correctly...

Scubad1v3r, 2020-03-27:
How to? "Can you set to "glorytun-tcp" as VPN in the wizard on the router"

Ysurac, 2020-03-27:
In System->OpenMPTCProuter checkbox "Advanced settings"

Scubad1v3r, 2020-03-27:
Je trouve pas...

Ysurac, 2020-03-27:
So you didn't change the VPS, glorytun-tcp is the default. Only give me the ```ip a``` result from the VPS.

Scubad1v3r, 2020-03-27:
`1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc prio state UP group default qlen 1000
    link/ether b6:34:7e:d2:36:b0 brd ff:ff:ff:ff:ff:ff
    inet 45.13.104.179/32 brd 45.13.104.179 scope global eth0:1
       valid_lft forever preferred_lft forever
    inet6 2a0b:cbc0:1101:1::38/64 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::b434:7eff:fed2:36b0/64 scope link 
       valid_lft forever preferred_lft forever
3: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 qdisc fq state UNKNOWN group default qlen 500
    link/none 
    inet 10.255.253.1/24 brd 10.255.253.255 scope global mlvpn0
       valid_lft forever preferred_lft forever
    inet6 fe80::4574:ae43:6515:ee4a/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
4: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 500
    link/none 
5: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc prio state UNKNOWN group default qlen 500
    link/none 
    inet 10.255.255.1/30 brd 10.255.255.3 scope global gt-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::be43:51fb:6372:6460/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
6: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UNKNOWN group default qlen 100
    link/none 
    inet 10.255.252.1 peer 10.255.252.2/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::74b9:2b9f:faee:4436/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
7: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UNKNOWN group default qlen 100
    link/none 
    inet 10.255.250.1 peer 10.255.250.2/32 scope global tun1
       valid_lft forever preferred_lft forever
    inet6 fe80::9bbc:658c:1d91:32e7/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
8: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
9: omr-6in4-user0@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1480 qdisc noqueue state UNKNOWN group default qlen 1000
    link/sit 10.255.251.1 peer 10.255.251.2
    inet6 fe80::aff:fb01/64 scope link 
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/126 scope link 
       valid_lft forever preferred_lft forever
10: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 qdisc fq state UNKNOWN group default qlen 500
    link/none 
    inet 10.255.251.1 peer 10.255.251.2/32 scope global dsvpn0
       valid_lft forever preferred_lft forever
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::e850:86d6:136d:a4ba/64 scope link stable-privacy 
       valid_lft forever preferred_lft forever
`

Ysurac, 2020-03-27:
So at least for you, IP are configured (not for glorytun UDP, I will try to fix that), but you VPN seems to be dsvpn instead of glorytun-tcp that is the default.

chris77170, 2020-03-27:
do you need me to give you the same info ?

Ysurac, 2020-03-27:
yes

Scubad1v3r, 2020-03-27:
Ok
I've manually desactived VPN->DSVPN and activated VPN->Glorytun
the message seems to have disappeared for now...

Thanks

chris77170, 2020-03-27:
In VPN-> Glorytun "Activated" and "Started"

[ip_a.txt](https://github.com/Ysurac/openmptcprouter/files/4393861/ip_a.txt)

Ysurac, 2020-03-27:
@chris77170 what is the VPN selected in the OpenMPTCProuter ? You may have 2 VPN enabled...

chris77170, 2020-03-27:
I only have Glorytum active

Scubad1v3r, 2020-03-27:
"VPN tunnel DOWN" again...

Ysurac, 2020-03-27:
I need State->System log (use pastebin, or paste here do not attach a file, I don't like that :) )

Scubad1v3r, 2020-03-27:
`Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI24, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI25, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI25
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI25, subsystem=tty
Fri Mar 27 16:54:38 2020 daemon.err omr-tracker[11336]: RTNETLINK answers: Permission denied
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI26, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI26
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI26, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI27, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI27
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI27, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI28, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI28
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI28, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI29, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI29
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI29, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI30, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI30
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI30, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI31, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI31
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI31, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice Shadowsocks: Reload omr-bypass rules
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI32, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI32
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI32, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice omr-bypass: Starting OMR-ByPass...
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI33, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI33
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI33, subsystem=tty
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI34, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI34
Fri Mar 27 16:54:38 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI34, subsystem=tty
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI35, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI35
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI35, subsystem=tty
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI36, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI36
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI36, subsystem=tty
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyI37, subsystem=tty, sysfspath=/sys/devices/virtual/tty/ttyI37
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI37, subsystem=tty
Fri Mar 27 16:54:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Fri Mar 27 17:25:31 2020 daemon.err /usr/bin/ss-local[18935]: getpeername: Socket not connected
Fri Mar 27 17:27:30 2020 daemon.notice wpa_supplicant[10756]: wlan0: WPA: Group rekeying completed with c6:7a:d3:64:55:e8 [GTK=CCMP]
Fri Mar 27 17:27:55 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Fri Mar 27 17:28:29 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Fri Mar 27 17:29:27 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out`

Ysurac, 2020-03-27:
You seems to have very high latency and wan2 was down.

Scubad1v3r, 2020-03-27:
I have rebooted wan2 but it's up
Latency is between 47ms up to 100ms for 4G (Wan1 Wwan and TetheringWAN) and 60/70ms for ADSL (Wan2)

Ysurac, 2020-03-27:
Bigger latency was found.
You can try to set "Master" to "no change" in System->OpenMPTCProuter and advanced settings tab, and set in Network->MPTCP master to wan2 (others should be set to enable)

Scubad1v3r, 2020-03-27:
Done but but msg VPN tunnel DOWN always appears

Ysurac, 2020-03-27:
I need again system log since you changed master.

Scubad1v3r, 2020-03-27:
OMR OpenMPTCProuter
État
Vue d\'ensemble
Pare-feu
Routes
Journal système
Journal du noyau
Processus
Graphiques temps-réel
Surveillance du trafic VnStat
Système
Système
OpenMPTCProuter
Administration
Logiciels
Démarrage
Tâches Régulières
Configuration des LED
Sauvegarde / Mise à jour du micrologiciel
Redémarrage avancé
Redémarrage
VPN
DSVPN
MLVPN
OpenVPN
Glorytun
Services
iPerf
Shadowsocks-libev
Wake on LAN
uHTTPd
OMR-Tracker
UPnP
Paramètres e-mail
OMR-Bypass
Recursive DNS
Nginx High Availability
Réseau
Interfaces
Sans-fil
DHCP et DNS
Noms d\'hôtes
Routes statiques
Pare-feu
Diagnostics
SNMPd
Macvlan
SQM QoS
OMR-DSCP
MPTCP
Quota
Déconnexion
Journal système
Fri Mar 27 19:38:21 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev eth1
Fri Mar 27 19:38:23 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.30.1 dev wan2
Fri Mar 27 19:38:25 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev wlan0
Fri Mar 27 19:38:26 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.1.203 dev wan1
Fri Mar 27 19:38:27 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev eth1
Fri Mar 27 19:38:28 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.30.1 dev wan2
Fri Mar 27 19:38:30 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev wlan0
Fri Mar 27 19:38:31 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.1.203 dev wan1
Fri Mar 27 19:38:32 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev eth1
Fri Mar 27 19:38:34 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.30.1 dev wan2
Fri Mar 27 19:38:36 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev wlan0
Fri Mar 27 19:38:37 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.1.203 dev wan1
Fri Mar 27 19:38:38 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev eth1
Fri Mar 27 19:38:39 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.30.1 dev wan2
Fri Mar 27 19:38:41 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev wlan0
Fri Mar 27 19:38:42 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 192.168.1.203 dev wan1
Fri Mar 27 19:38:43 2020 user.notice post-tracking-post-tracking: Master up : Replace default route by 172.20.10.1 dev eth1
[... 546 lines cut ...]
Fri Mar 27 21:29:01 2020 daemon.err /usr/bin/ss-local[18935]: getpeername: Socket not connected
Fri Mar 27 21:29:25 2020 daemon.err /usr/bin/ss-local[18935]: getpeername: Socket not connected
Fri Mar 27 21:34:29 2020 user.notice post-tracking-post-tracking: Get status and settings for wwan...
Fri Mar 27 21:34:42 2020 user.notice post-tracking-post-tracking: Get status and settings for TetheringWAN...
Fri Mar 27 21:35:38 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Fri Mar 27 21:40:21 2020 daemon.err /usr/bin/ss-local[18935]: getpeername: Socket not connected
Fri Mar 27 21:40:27 2020 daemon.err /usr/bin/ss-redir[18936]: remote recv: Operation timed out
Fri Mar 27 21:40:27 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Fri Mar 27 21:40:29 2020 daemon.err /usr/bin/ss-local[18935]: getpeername: Socket not connected
Fri Mar 27 21:40:58 2020 daemon.err /usr/bin/ss-local[18935]: getpeername: Socket not connected
Fri Mar 27 21:41:03 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Fri Mar 27 21:41:11 2020 daemon.err /usr/bin/ss-local[18935]: getpeername: Socket not connected
Fri Mar 27 21:41:30 2020 daemon.err /usr/bin/ss-local[18935]: getpeername: Socket not connected
Fri Mar 27 21:41:41 2020 daemon.err /usr/bin/ss-redir[18936]: remote recv: Operation timed out
Fri Mar 27 21:42:31 2020 daemon.notice wpa_supplicant[10756]: wlan0: WPA: Group rekeying completed with c6:7a:d3:64:55:e8 [GTK=CCMP]
Fri Mar 27 21:43:39 2020 user.notice post-tracking-post-tracking: Get status and settings for wwan... Done
Fri Mar 27 21:43:50 2020 user.notice post-tracking-post-tracking: Get status and settings for TetheringWAN... Done
Fri Mar 27 21:51:44 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Fri Mar 27 21:53:52 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Fri Mar 27 21:57:11 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Fri Mar 27 21:57:31 2020 daemon.notice wpa_supplicant[10756]: wlan0: WPA: Group rekeying completed with c6:7a:d3:64:55:e8 [GTK=CCMP]
Fri Mar 27 21:57:48 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Fri Mar 27 21:58:08 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Fri Mar 27 21:58:11 2020 daemon.err /usr/bin/ss-redir[18937]: remote recv: Operation timed out
Powered by openmptcprouter v0.54.7 r0+12509-04a21c26a0

chris77170, 2020-03-28:
Hello Ysurac,
Do you have news on this concern ...

Ysurac, 2020-03-28:
You can try in Network->MPTCP to disable all interfaces and set one as master (the interface with lower latency), then add one after one to check if the problem come with a specific interface.
You can also try to disable fastopen in System->OpenMPTCProuter and advanced settings tab.

Scubad1v3r, 2020-03-28:
No change - msg VPN tunnel DOWN always appears

Ysurac, 2020-03-28:
You tested all in less than 7 minutes ? really ? ;)

Ysurac, 2020-03-28:
If the VPN is down, even with one connection set as master, then try a ```systemctl restart glorytun-tcp@tun0``` on the VPS, if it's still down, try a ```/etc/init.d/glorytun restart``` on the router. If it's still down check on the router system log about glorytun. If there is nothing special, give me full configuration, without keys, available in System->OpenMPTCProuter and all parameters tab.

Ysurac, 2020-03-28:
And a ```ps aux | grep glorytun``` from the VPS with a ```ip a``` to check if glorytun is running and configured on the VPS. A ```ps w | grep glorytun``` and ```ip a``` from the router to check same on the router.

chris77170, 2020-03-28:
Ysurac "You can try in Network->MPTCP to disable all interfaces and set one as master (the interface with lower latency), ..... OpenMPTCProuter and advanced settings tab."
No change - msg VPN tunnel DOWN always appears

Ysurac, 2020-03-28:
Use one interface as master, disable all others in Network->MPTCP, and try what I said before on the VPS and router.

Scubad1v3r, 2020-03-28:
> You tested all in less than 7 minutes ? really ? ;)

oui
déconnecter si j'ai bien compris laisser un seul wan master désactiver les 3 autres lancer un speedtest et réactualiser la page état - a faire 3x ca prends pas une heure non plus

Scubad1v3r, 2020-03-28:
Bon j'ai plus rien qui fonctionne ...
Réseau free et free mobile trop instables
je remets dernière backup du routeur et reprendrais cette aprem les vérifications

chris77170, 2020-03-28:
Always the same problem, using an interface as master, deactivating all the others ;
VPN tunnel DOWN
Can't get public IP address from ShadowSocks
Can't contact Server Admin Script

chris77170, 2020-03-28:
systemctl restart glorytun-tcp@tun0 on the VPS, and, try a /etc/init.d/glorytun restart
don't give anything either

chris77170, 2020-03-28:
VPS
ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc prio state UP group default qlen 1000
    link/ether 2e:c3:ef:c1:55:be brd ff:ff:ff:ff:ff:ff
    inet 45.13.104.173/32 brd 45.13.104.173 scope global eth0:1
       valid_lft forever preferred_lft forever
    inet6 2a0b:cbc0:1101:1::31/64 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::2cc3:efff:fec1:55be/64 scope link
       valid_lft forever preferred_lft forever
3: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 qdisc fq state UNKNOWN group default qlen 500
    link/none
    inet 10.255.253.1/24 brd 10.255.253.255 scope global mlvpn0
       valid_lft forever preferred_lft forever
    inet6 fe80::fec0:fb9e:84a5:44b0/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
4: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UP group default qlen 500
    link/none
    inet 10.255.254.1/30 brd 10.255.254.3 scope global gt-udp-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::be17:2f26:2b16:41a7/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
6: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UNKNOWN group default qlen 100
    link/none
    inet 10.255.250.1 peer 10.255.250.2/32 scope global tun1
       valid_lft forever preferred_lft forever
    inet6 fe80::cde2:b370:5db0:c6f8/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
7: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UNKNOWN group default qlen 100
    link/none
    inet 10.255.252.1 peer 10.255.252.2/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::31cb:19b6:ef40:5c2f/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
8: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 qdisc fq state UNKNOWN group default qlen 500
    link/none
    inet 10.255.251.1 peer 10.255.251.2/32 scope global dsvpn0
       valid_lft forever preferred_lft forever
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::e235:757e:fa6d:491a/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
9: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UNKNOWN group default qlen 500
    link/none
    inet 10.255.255.1/30 brd 10.255.255.3 scope global gt-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::ce86:4839:ccc5:704/64 scope link stable-privacy
       valid_lft forever preferred_lft forever

Ysurac, 2020-03-28:
@chris77170 I would need system log of the router and all parameters without keys.

chris77170, 2020-03-28:
for a VPS
ps aux | grep glorytun
root       516  0.0  0.0   2692   908 ?        SLs  Mar27   0:00 glorytun bind 0.0.0.0 65001 keyfile /etc/glorytun-udp/tun0.key dev gt-udp-tun0 chacha persist
root     30141  0.0  0.0   2960   860 ?        Ss   11:41   0:00 glorytun-tcp listener keyfile /etc/glorytun-tcp/tun0.key dev gt-tun0 port 65001 mptcp chacha20 retry count -1 const 5000000 timeout 10000 keepalive count 5 idle 20 interval 2 buffer-size 32768 multiqueue
christi+ 31551  0.0  0.0   6148   888 pts/0    S+   11:49   0:00 grep glorytun

ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc prio state UP group default qlen 1000
    link/ether 2e:c3:ef:c1:55:be brd ff:ff:ff:ff:ff:ff
    inet 45.13.104.173/32 brd 45.13.104.173 scope global eth0:1
       valid_lft forever preferred_lft forever
    inet6 2a0b:cbc0:1101:1::31/64 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::2cc3:efff:fec1:55be/64 scope link
       valid_lft forever preferred_lft forever
3: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1444 qdisc fq state UNKNOWN group default qlen 500
    link/none
    inet 10.255.253.1/24 brd 10.255.253.255 scope global mlvpn0
       valid_lft forever preferred_lft forever
    inet6 fe80::fec0:fb9e:84a5:44b0/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
4: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UP group default qlen 500
    link/none
    inet 10.255.254.1/30 brd 10.255.254.3 scope global gt-udp-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::be17:2f26:2b16:41a7/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
6: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UNKNOWN group default qlen 100
    link/none
    inet 10.255.250.1 peer 10.255.250.2/32 scope global tun1
       valid_lft forever preferred_lft forever
    inet6 fe80::cde2:b370:5db0:c6f8/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
7: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state UNKNOWN group default qlen 100
    link/none
    inet 10.255.252.1 peer 10.255.252.2/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::31cb:19b6:ef40:5c2f/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
8: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 qdisc fq state UNKNOWN group default qlen 500
    link/none
    inet 10.255.251.1 peer 10.255.251.2/32 scope global dsvpn0
       valid_lft forever preferred_lft forever
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::e235:757e:fa6d:491a/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
9: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc mq state UNKNOWN group default qlen 500
    link/none
    inet 10.255.255.1/30 brd 10.255.255.3 scope global gt-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::ce86:4839:ccc5:704/64 scope link stable-privacy
       valid_lft forever preferred_lft forever

chris77170, 2020-03-28:
OMR journal system
Sat Mar 28 11:18:35 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:18:40 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:18:42 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:18:48 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:18:50 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:18:55 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:18:56 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:00 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:19:04 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:08 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:19:12 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:15 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:19 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:19:20 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:27 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:19:29 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:35 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:37 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:37 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:19:45 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:45 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:19:53 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:19:55 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:19:55 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:01 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:03 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:20:09 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:14 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:20:16 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:18 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:22 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:20:26 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:32 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:20:34 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:36 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:40 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:20:42 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:50 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:50 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:20:56 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:58 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:20:59 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:21:06 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:21:09 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:21:15 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:21:16 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:21:17 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:21:23 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:21:27 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:21:31 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:21:35 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 11:21:36 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 11:21:39 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
[... cut ...]
Sat Mar 28 12:05:19 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 12:05:22 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 12:05:24 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
Sat Mar 28 12:05:30 2020 daemon.err /usr/bin/ss-local[12583]: getpeername: Socket not connected
Sat Mar 28 12:05:34 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)

chris77170, 2020-03-28:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@domain[0]=domain
dhcp.@domain[0].name='OPNsense'
dhcp.@domain[0].ip='192.168.100.224'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
[... 1315 lines cut ...]
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth1' 'eth2' 'eth3' 'eth4'
vnstat.@vnstat[0].backup='0'

Sorry for the delay but not simple with a rotten connection

Ysurac, 2020-03-28:
The VPS is updated to 0.1015 ?

chris77170, 2020-03-28:
Yes

chris77170, 2020-03-28:
Ysurac, milkywan is under attack right now, wouldn't the problem come from there?

Ysurac, 2020-03-28:
Maybe, but the admin API seems to not answer. Possible that ping work but http doesn't...

chris77170, 2020-03-28:
I can ping the vm but can't connect to ssh at the moment

Ysurac, 2020-03-28:
ok, so this can really be the problem :)
I will add in status page a check for http answer.

chris77170, 2020-03-28:
Ok, as soon as the situation improves, I will give you feedback.
Thanks again.

Scubad1v3r, 2020-03-28:
So do I

chris77170, 2020-03-30:
Hello,
I finally have my router again which communicates with my vps.
In fact the concern came from the fact that I had the redirection of all the ports of the vps to the router.
It was Hugues de Milkyan who pointed this out to me following the impossibility of connecting to my VM.
What I don't really understand since I had it before and it worked ...
Thanks for the time!

Ysurac, 2020-03-30:
You manually added the redirection ?

chris77170, 2020-03-30:
no the box was checked in
System -> OpenMPTCRouter -> Advanced settings
VPS
Redirect all ports from the server to this router

I just added a rule in
Network -> Firewall -> Port forwarding, in order to have a forwarding of all the ports towards my internal firewall

chris77170, 2020-03-30:
on the other hand I had to try again several times in order to uncheck the box

Ysurac, 2020-03-30:
The all redirect ports open port from 1 to 64999 and OMR use port > 65000 so this shouldn't make any problems. But it's always a better idea to redirect only needed ports :)

chris77170, 2020-03-30:
This is what I saw in the VPS rules
For port forwarding this is what I will do in the future as everything is working fine now. And that the integration of my firewall works

closed 2020-04-02 by chris77170