Ysurac/openmptcprouter #941: Port Forward IPV4

soken35, 2020-03-30:
Salut ;-)

J'ai mis à jour mon VPS et mon routeur sous ESXI, tout fonctionne très bien depuis. 

J'ai décidé de passer le cap a l'ipv6 et tout fonctionne. Seulement, dès que j'active le serveur DHCP pour l'ipv6 toutes mes redirections de port IPV4 ne fonctionnent plus. 

Quelqu'un aurait une idée ?

ps: Merci  Ysurac pour tout le boulot de titan que tu nous fais. Je ne peux plus me passer de ta solution :)

bambuca, 2020-04-02:
Please write in english. There could be more non France speaking people with same question or issue. 
Thank you

cron410, 2020-04-04:
I ran it through GTranslate. 
```
Hi ;-)

I updated my VPS and my router under ESXI, everything has been working fine since.

I decided to pass the course and activated the ipv6 and everything works. Only, as soon as I activate the DHCP server for ipv6, all my IPV4 port redirects no longer work.

Someone would have any idea ?

ps: Thank you Ysurac for all the titanic work that you do to us. I can't do without your solution anymore :)
```

There's a heavy user base in France because this project is somehow related to OVH who sells this service commercially. I hope they pay YSurac for this fantastic project.

Ysurac, 2020-04-04:
This project is not really related to OVH. More info: https://github.com/Ysurac/openmptcprouter/issues/911

closed 2020-08-12 by Ysurac