Ysurac/openmptcprouter #942: wan2 (slave) is barely utilized

gitxmax, 2020-03-30:
## Expected Behavior
total speed is close to wan1 +wan2

  In English please.

## Actual Behavior
wan2 is not really utilized, when wan1 is set as master and wan2 as enabled

## Steps to Reproduce the Problem

  1. Clean install of both router and vpn firmwares
  2. wan1 is eth0 wan2 is usb0 (both set in Macvlan)
  3. Running various speed tests

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: v0.1015
  - OpenMPTCProuter platform: x86_64

Hello,
my wan1 is a broadband connection, attached to eth0 in macvlan, maxiing out at 4mbps, but unlimited
wan2 is a 4g router, connected via usb, macvlan set to usb0 but limited to 14gb/month

My idea is to use wan2, as a backup, when the wan1 speed is not sufficient.

the problem is, whatever connection I set to "Master"  is giving a full speed, the other "slave" connection is almost not used, perhaps 10% at maximum.

So, if I set wan1 as master and wan2 as enabled, I get total speed ~=wan1 speed
if I set wan2 as master and wan1 as enabled, I get total speed ~=wan2 speed

openmptcprouter/status shows everything GREEN
VPN is installed on 1gb ram google cloud instance with all ports open

Tried running the built-in speed test, it confirm the situation, wan2 is running at 10% of the real speed, when wan1 is master.

bnhf, 2020-03-31:
Macvlan may be your problem here. Since each of your WANs is on a separate physical interface it's not required and may be causing issues. Macvlan is used when you have a single Ethernet port and you'd like to share it using an unmanaged switch.

Try deleting the Macvlan entries, and then check your interface assignments under Network - Interfaces to be sure wan1 and wan2 are assigned to the correct physical interfaces rather than the virtual Macvlan interfaces. It's worth checking that each remains assigned to the wan firewall zone too.

Couldn't hurt to reboot the router after making the changes.

gitxmax, 2020-04-01:
Thanks.
I removed all the instances from Macvlan and made sure everything is correct in Interfaces. Multiple reboots done also. Sadly, nothing has really changed, still only the Master deals with all the traffic...

[image]
[image]
[image]

Ysurac, 2020-04-01:
Check in Network->MPTCP and MPTCP fullmesh that you have all WAN IPs.
Check also in State->System log if you have no error.

bnhf, 2020-04-01:
Do you have a monthly Quota set on either interface -- or are you trying to use SQM?
Are download speed and upload speed both set to 0 for both interfaces in the OMR Wizard?

gitxmax, 2020-04-01:
Haven't really change any default settings, so yes, no quotas, no sqm..

[image]
[image]
[image]
[image]

Errot log: https://drive.google.com/open?id=1vo53nDmIp6lK7u7qVoDsuv9Rvg6RzIwp
Full settings: https://drive.google.com/open?id=1dBW0XzFGSa3P3Z-OoQpNZfBlyhT54Vox

gitxmax, 2020-04-05:
Just want to post an update
I tried the second eth2 connection, instead of the usb0 and all seems to work fine.
I guess, it has something to do with the usb connection..

Ysurac, 2020-04-05:
How did you test eth2 ? The router have a eth connection ?

gitxmax, 2020-04-05:
nope, attached a different broadband router, using a different ISP

closed 2020-09-14 by github-actions[bot]