Ysurac/openmptcprouter #943: Ping not working

bournehlm, 2020-03-31:
## Expected Behavior

  PC connected to router can't ping. I can ping 192.168.100.1, ping 8.8.8.8 failed, but I can access internet, only can't ping, i don't why.
  After that problem come out, I need to reflash the router.

## Actual Behavior

## Steps to Reproduce the Problem

  1.PC(win10) connected to the rpi3 router, dhcp get ip 192.168.100.123.
  2.PC ping 192.168.100.1 success.
  3.PC ping public ip suchas 8.8.8.8 failed, no response.
  4. Glorytun and ss working fine, PC can access internet.

## Specifications

  - OpenMPTCProuter version: (v0.52.3)
  - OpenMPTCProuter VPS version: (0.1001)
  - OpenMPTCProuter platform: (RPI3 64bits)

Ysurac, 2020-03-31:
Ping use the VPN, so this should use glorytun TCP.
I don't support old releases, so if you want help you have to update router and VPS.

bournehlm, 2020-03-31:
It seems the glorytun working fine, the VPN state is ok at the state page.
And I can ping 8.8.8.8 on the router, i think it's also use the glorytun VPN.
I will try the latest version, thanks.

closed 2020-09-14 by github-actions[bot]