Ysurac/openmptcprouter #944: Getting tips on how to best configure OMR for games to avoid latency problems

gwarf, 2020-03-31:
I would like to understand what is the best/recommended way of configuring OMR so that tings will work well for games. 👾 
Until know I've been using the default setup with DSVPN, but my kids are from time to time getting some latency/ping problems and are harassing me... 🦈 😢 
I've looked at #268, #769 and it seems that disabling the VPN could help (so I'm now testing this), there is also the mention of SQM but I haven't much knowledge about this...

So is there any recommended setup or tips?

For what it's worth I'm using 3 ADSL connections with a RPI3 OMR and scaleway VPS.

Thanks!

Ysurac, 2020-03-31:
I'm using glorytun TCP without issues with latency in games (like CSGO). I'm using a scaleway VPS too.
Disabling the VPN is often not a good idea. OMR-ByPass can help if you don't need aggregation and want to use a specific interface.
SQM can help, you have to setup it via the wizard (and lower speed if you have still some problems).
What is the game ?

gwarf, 2020-03-31:
OK, will try Glorytun TCP and see how it goes.
I've been using OMR-ByPass for some other stuff, I see ways to force traffic via a specific gateway for everything (There is no UDP protocol in the list and I'm not sure about the server lists that I could allow to have some more granularity) but I would like to let them use the aggregation for most of the usage, especially with regards to downloading updates/games.

The problem are with CS:GO and Fortnite.

Ysurac, 2020-03-31:
It would not be a good idea to bypass everything on UDP.
What is the latency of each connection in the status page ?

gwarf, 2020-03-31:
23 to 30 depending on the wind, more often around 30.

(Incidently I see that one link (RED by SFR) is with an MTU of 1492 while the two others ones are at 1500 (Sosh) but not sure if it could have an impact/if it should haromonized)

closed 2020-09-14 by github-actions[bot]