Ysurac/openmptcprouter #948: Way to select VPN interfaces (like MPTCP concept with master/backup/handover)

dur3x, 2020-04-01:
## Expected Behavior
In the MPTCP options I can select an interface as master and another one as backup or handover.
So it's fine if works for **TCP** protocol; the second interface is only used when there is a problem on the first one.  
But for the **UDP** protocol I'm not able to do a similar scenario; I want(/prefer) to use MLVPN or glorytun-udp but it looks that all the time that uses all interfaces.

Why I want to use this kind of setup (master/backup)? Simply because I have some good interfaces (which are stable) but sometimes it happens that some problems occur on these ones and then I want to automatically use the second one in the worst case. I don't want to use it all the time the worst one because if I do active-active with this one I have more global bad issues.

Some ideas:
- re-evaluate tcp vpn options (glorytun-tcp, deadvpn, openvpn) but in my previous experiences I got better result with udp ones (my tests were done with sensitive applications: jitsi/skype)
- search/try to use options in mlvpn/glorytun

Ysurac, 2020-04-02:
This should work with Glorytun UDP since latest release, but I never tested this.

Chatissimus, 2020-04-02:
Hello, 
I use it like that on lastest release

- Glorytun UDP
- one master interface on an unstable ADSL connexion
- one handover interface on 4G 

and yes the 4G interface seems well used only when my ADSL interface has congestion (which is an issue in my side cause the 4g interface tend to be used too much when high activity :) )

example , my ADSL upload vary between 200kb to 900kb depending on the wind, the 4G (WAN1) well used on peak
[image]

dur3x, 2020-04-07:
Thanks @Ysurac I will test that when you will release a new version ;-)
@Chatissimus thank you for your feedback ! It looks what I want

closed 2020-09-14 by github-actions[bot]