Ysurac/openmptcprouter #949: Bad upload via openmtpcprouter

kevin39, 2020-04-02:
Hi,
Download is OK
My upload is a lot better without using openmtpcrouter. Is there some tunning possible ?

With OpenMTCPRouter : 
Very poor and unstable upload (screenshot 1)

Without OpenMTCPRouter : (directly connected to my box)
Upload ~30mbs  (screen shot 2)

Do you have an idea ?

[image]
[image]

Ysurac, 2020-04-02:
Please use Issue template...

kevin39, 2020-04-02:
Sorry. Fixed.

Ysurac, 2020-04-02:
You can try in Network->MPTCP another Multipath TCP scheduler like BLEST and try other TCP Congestion Control.

kevin39, 2020-04-02:
I'm already using BLEST.
Anyway, I tried once again all settings. All are having same behavior with my upload.

bnhf, 2020-04-02:
Given the modest increases you're seeing on the download side, and the instability you're reporting on the upload side, perhaps there's not really that much risk in upgrading? Falling back to a single connection looks to be better than what you have going now. It's always best to be on the latest release with a leading-edge package like this.

closed 2020-04-05 by kevin39

kevin39, 2020-04-16:
FYI : 
- Bad upload test using nperf.com
- Upload speed using speedtest.net and test.my/upload are OK

Anyway, I was having a bad upload (~1mbs).
I improved my antenna positionning and my upload is now ~8mbs (anyway, nperf test is still very unstable).