Ysurac/openmptcprouter #950: VPN Tunnel Down with GloryTUN

cron410, 2020-04-04:
Expected:
No errors on Status page when all connections are up. 

Actual:
```
VPN tunnel DOWN
Version 0.54.7
Load: 0.11 0.21 0.19
Uptime: 0h 51m 34s
```
GloryTUN seems to be operational from Logs. See below.

## Steps to Reproduce the Problem

  1. Upgrade Router and VPS
  2. Reboot both

## Specifications

  - OpenMPTCProuter version: Version 0.54.7
  - OpenMPTCProuter VPS version: Version 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: x86_64

## Logs
```
Sat Apr  4 09:53:52 2020 daemon.err /usr/bin/ss-local[17851]: getpeername: Socket not connected
Sat Apr  4 09:53:57 2020 daemon.err /usr/bin/ss-local[17851]: getpeername: Socket not connected
Sat Apr  4 09:53:57 2020 daemon.err /usr/bin/ss-local[17851]: getpeername: Socket not connected
Sat Apr  4 09:54:02 2020 daemon.err /usr/bin/ss-local[17851]: getpeername: Socket not connected
Sat Apr  4 09:54:07 2020 daemon.err /usr/bin/ss-local[17851]: getpeername: Socket not connected
Sat Apr  4 09:54:11 2020 daemon.err glorytun[31393]: 3.14.20.8.65001: connected
Sat Apr  4 09:54:12 2020 daemon.info glorytun[31393]: STARTED tun0
Sat Apr  4 09:54:12 2020 daemon.info glorytun[31393]: STOPPED tun0
Sat Apr  4 09:54:13 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 103.224.182.242)
Sat Apr  4 09:54:13 2020 daemon.info omr-tracker-ss: Reload Shadowsocks rules
Sat Apr  4 09:54:13 2020 user.notice Shadowsocks: Rules UP
Sat Apr  4 09:54:13 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sat Apr  4 09:54:13 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Apr  4 09:54:13 2020 daemon.err glorytun[31393]: 3.14.20.8.65001: connected
Sat Apr  4 09:54:14 2020 user.notice omr-bypass: Reload dnsmasq...
Sat Apr  4 09:54:14 2020 daemon.info dnsmasq[2520]: read /etc/hosts - 4 addresses
Sat Apr  4 09:54:14 2020 daemon.info dnsmasq[2520]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Sat Apr  4 09:54:14 2020 daemon.info dnsmasq-dhcp[2520]: read /etc/ethers - 0 addresses
Sat Apr  4 09:54:14 2020 user.notice omr-bypass: OMR-ByPass is running
```

cron410, 2020-04-04:
Apparently my second LTE connection is not able to pass OMRVPN traffic. Maybe AT&T is doing some deep packet inspection and blocking? My previous router version was v0.53.3 and corresponding VPS version from a few weeks ago.

[image]

```
Sat Apr  4 10:31:23 2020 user.notice omr-tracker: Launching...
Sat Apr  4 10:31:23 2020 user.notice omr-tracker: Launched
Sat Apr  4 10:31:24 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Sat Apr  4 10:31:24 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.1.1.1)
Sat Apr  4 10:31:56 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Sat Apr  4 10:31:56 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.3.1 dev eth1
Sat Apr  4 10:32:21 2020 user.notice post-tracking-post-tracking: eth2 switched off because ping from 192.168.0.232 error (9.9.9.9,1.1.1.1,1.0.0.1)
Sat Apr  4 10:32:21 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev eth2
Sat Apr  4 10:33:06 2020 authpriv.info dropbear[16107]: Child connection from 10.1.1.107:1945
Sat Apr  4 10:33:10 2020 authpriv.notice dropbear[16107]: Password auth succeeded for 'root' from 10.1.1.107:1945
Sat Apr  4 10:33:43 2020 user.notice post-tracking-post-tracking: Set server vps (3.14.20.8) route via 192.168.0.1 metric 4
Sat Apr  4 10:33:43 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2...
Sat Apr  4 10:33:45 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
```
  
```
root@OpenMPTCProuter:~# ping -I eth2 9.9.9.9
PING 9.9.9.9 (9.9.9.9) from 192.168.0.232 eth2: 56(84) bytes of data.
64 bytes from 9.9.9.9: icmp_seq=1 ttl=53 time=84.5 ms
64 bytes from 9.9.9.9: icmp_seq=2 ttl=53 time=125 ms
64 bytes from 9.9.9.9: icmp_seq=3 ttl=53 time=86.6 ms
64 bytes from 9.9.9.9: icmp_seq=4 ttl=53 time=85.6 ms
64 bytes from 9.9.9.9: icmp_seq=5 ttl=53 time=125 ms
^C
--- 9.9.9.9 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4003ms
rtt min/avg/max/mdev = 84.485/101.404/125.441/19.443 ms
root@OpenMPTCProuter:~# ping -I eth1 9.9.9.9
PING 9.9.9.9 (9.9.9.9) from 192.168.3.232 eth1: 56(84) bytes of data.
64 bytes from 9.9.9.9: icmp_seq=1 ttl=52 time=46.9 ms
64 bytes from 9.9.9.9: icmp_seq=2 ttl=52 time=52.9 ms
64 bytes from 9.9.9.9: icmp_seq=3 ttl=52 time=44.0 ms
64 bytes from 9.9.9.9: icmp_seq=4 ttl=52 time=33.3 ms
64 bytes from 9.9.9.9: icmp_seq=5 ttl=52 time=43.6 ms
^C
--- 9.9.9.9 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4004ms
rtt min/avg/max/mdev = 33.324/44.155/52.874/6.346 ms
```

Ysurac, 2020-04-04:
Latency on eth2 is high.
You can try another VPS in System->OpenMPTCProuter and advanced setting checkbox.

gitxmax, 2020-04-04:
Probably not very related, but I had all sorts of connection problems, until I upgraded the vps from 1core 1gb to 2core 2gb. Tons better now.

Ysurac, 2020-04-04:
This can be related, for some high speed a good VPS is needed (and virtual core can be sometimes really slow). In default OMR configuration, I expect VPS to not be the limit for anything (speed, encryption,...) so if in fact it is the limit many problems can happen.

bnhf, 2020-04-04:
> Apparently my second LTE connection is not able to pass OMRVPN traffic. Maybe AT&T is doing some deep packet inspection and blocking? My previous router version was v0.53.3 and corresponding VPS version from a few weeks ago.

No issues here with those same two providers (plus Verizon), on the same versions of OMR and the same hardware base:
[image]

bnhf, 2020-04-04:
> Probably not very related, but I had all sorts of connection problems, until I upgraded the vps from 1core 1gb to 2core 2gb. Tons better now.

Certainly something worth looking at for any user. In my case, I'm well within the limits of 1vCore and 1GB of RAM with a Vultr VPS. Hopefully most providers have similar stats available to evaluate whether their VPS has sufficient horsepower for the job they're asking it to do:
[image]

cron410, 2020-04-04:
I'm running my "VPS" on a personal server I installed in my friend's rack at a datacenter. It's running on a 2 core VM in ESXi with 4GB ram and has it's own 1GB direct connection to a major internet provider. I get on average 800MB/s and 5ms ping to google and akamai with it. I didn't have the issue until I upgraded the VPS script and router today.

@bnhf  can you search your system log on the router for `omrvpn` and post any results you find with `wan1` ?   (if wan1 is your AT&T connection)

bnhf, 2020-04-04:
> @bnhf can you search your system log on the router for `omrvpn` and post any results you find with `wan1` ? (if wan1 is your AT&T connection)

No omrvpn lines in my system log. Did a reboot of the router, just to see if any appeared and got these two:

`Sat Apr  4 12:18:56 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sat Apr  4 12:19:01 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)`

Everything still looks on the OMR status page, and AT&T passes the MPTCP check under Network - MPTCP - MPTCP Support Check. MPTCP Fullmesh looks correct too.

bnhf, 2020-04-04:
Also, worth checking to confirm no other VPNs are enabled besides Glorytun. Also, confirm your WAN Network Interfaces are all still assigned to the wan firewall zone -- their header labels should all be pink. I have, on occasion, had zone assignments disappear when adding and subtracting WANs on OMR.
[image]

cron410, 2020-04-04:
Yep GloryTUN is enabled and I have all other VPNs disabled. OpenVPN and DSVPN are also disabled at the VPS from running the launch script with DSVPN=disabled etc.
  
Here's an example screen from my router:
[image]

All my WANs are in the red firewall zone, my Interfaces page matches. I haven't touched my interfaces at all.

As a test I switched over to my AWS Lightsail instance that I've been fighting with since yesterday and has been running for about an hour now. It works so that means my VM on the dedicated server is borked. I'll have to build a new one.

AWS VPS active:
[image]

Side note, I do NOT recommend AWS Lightsail at all for hosting this unless you like doing a bunch of dirty hacks just to get a functioning MPTCP VPS. I'll describe those misadventures in another ticket.

Ysurac, 2020-04-05:
This doesn't fully work, there is the "Direct output" warning. So there is still an issue.
I need your configuration, without keys, available in System->OpenMPTCProuter and full parameters tab.

cron410, 2020-04-07:
I verified both VPS have the same GloryTUN key in `/etc/glorytun-tcp/tun0.key`

Here's the full config from OpenWRT, I copied a bit more from the page than was needed, oh well.
https://pastebin.com/iKq1B0b0

Here is Glorytun config on OpenWRT router:
```
root@OpenMPTCProuter:~# cat /etc/config/glorytun

config glorytun 'vpn'
        option dev 'tun0'
        option mptcp '1'
        option proto 'tcp'
        option chacha20 '1'
        option mtuauto '1'
        option enable '1'
        option host '127.0.0.1'
        option key '##########censored#########'
        option port '65001'
        option localip '10.255.255.2'
        option remoteip '10.255.255.1'
```

Here's excerpt from the end of the system log on OpenWRT.
```
Mon Apr  6 21:37:19 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:20 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:20 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:20 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:20 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:21 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:21 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:21 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:21 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:22 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:23 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:23 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:23 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:24 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:24 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:24 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:24 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:25 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:25 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:25 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:25 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:26 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:26 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:26 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:26 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:27 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:27 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:27 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:27 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:28 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:28 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:28 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:28 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:29 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:29 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:29 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:29 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:30 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:30 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:30 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:30 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:31 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:31 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:31 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:31 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:32 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:32 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:32 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:32 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:33 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:33 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:33 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:33 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:34 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:34 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:34 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:34 2020 daemon.info glorytun[8695]: STOPPED tun0
Mon Apr  6 21:37:35 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: connected
Mon Apr  6 21:37:35 2020 daemon.err glorytun[8695]: read: Connection reset by peer
Mon Apr  6 21:37:35 2020 daemon.err glorytun[8695]: 127.0.0.1.65001: key exchange failed
Mon Apr  6 21:37:35 2020 daemon.info glorytun[8695]: STOPPED tun0
```

Ysurac, 2020-04-07:
You use multiple VPS ?

cron410, 2020-04-07:
I do right now because I've had so many problems since the upgrade. Is this not tested? I set one to Server Master when I'm reinstalling on the other.

Ysurac, 2020-04-07:
It's tested but not as much as normal configuration. Maybe keys are not sync anymore, try to save the wizard again.

cron410, 2020-04-07:
I ran the VPS script again on "vps" which is my dedicated server. Glortun was not running after script ran so I started it manually with `systemctl` and walked away frustrated. Came back to the computer after 30 minutes and checked router system log, still showing `Key Exchange Failed`, then walked away for another 5 minutes and it was working. This is driving me crazy.

cron410, 2020-04-19:
Even after reinstalling the local router I could not connect with GloryTUN TCP and DSVPN with both of my VPS machines. GloryTUN UDP is working great though!

GloryTUN TCP
```
Sat Apr 18 20:31:03 2020 daemon.info glorytun[31163]: STOPPED tun0
Sat Apr 18 20:31:04 2020 daemon.err glorytun[31163]: 127.0.0.1.65001: connected
Sat Apr 18 20:31:04 2020 daemon.err glorytun[31163]: read: Connection reset by peer
Sat Apr 18 20:31:04 2020 daemon.err glorytun[31163]: 127.0.0.1.65001: key exchange failed
Sat Apr 18 20:31:04 2020 daemon.info glorytun[31163]: STOPPED tun0
Sat Apr 18 20:31:05 2020 daemon.err glorytun[31163]: 127.0.0.1.65001: connected
Sat Apr 18 20:31:05 2020 daemon.err glorytun[31163]: read: Connection reset by peer
Sat Apr 18 20:31:05 2020 daemon.err glorytun[31163]: 127.0.0.1.65001: key exchange failed
Sat Apr 18 20:31:05 2020 daemon.info glorytun[31163]: STOPPED tun0
```

DSVPN
```
Sun Apr 19 00:36:01 2020 daemon.info dsvpn[26966]: Connecting to 127.0.0.1:65401...
Sun Apr 19 00:36:01 2020 daemon.info dsvpn[26966]: Trying to reconnect
Sun Apr 19 00:36:01 2020 daemon.err dsvpn[26966]: Client connection failed: Connection refused
Sun Apr 19 00:36:04 2020 daemon.info dsvpn[26966]: Connecting to 127.0.0.1:65401...
Sun Apr 19 00:36:04 2020 daemon.err dsvpn[26966]: Client connection failed: Connection refused
Sun Apr 19 00:36:04 2020 daemon.info dsvpn[26966]: Trying to reconnect
```

After changing to GloryTUN UDP
```
Sun Apr 19 00:45:41 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun Apr 19 00:45:41 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Apr 19 00:45:42 2020 user.notice OMR-VPS: Get config from VPS...
Sun Apr 19 00:45:42 2020 user.notice OMR-VPS: Glorytun restart...
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Network device 'tun0' link is down
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Sun Apr 19 00:45:42 2020 daemon.info glorytun-udp: starting glorytun-udp vpn instance vpn
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Network device 'tun0' link is up
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sun Apr 19 00:45:42 2020 daemon.err glorytun-udp[18177]: running on device tun0 as pid 18177
Sun Apr 19 00:45:42 2020 daemon.notice netifd: Interface 'omrvpn' is now up
```

[image]

Everything is going great now. I just wish I knew what setting I changed to cause this mess. I'm glad GloryTUN UDP is working now, it didn't in the past and I was very much looking forward to it. Thanks for all your hard work @Ysurac

closed 2020-04-19 by cron410