Ysurac/openmptcprouter #951: Discord Voice and video not work

vinceducat, 2020-04-04:
## Expected Behavior

  Discord soft for used by my children for contact there friend during game party doesnt work
https://discordapp.com
## Actual Behavior

voice and video dont work : saying probleme of UDP connexion

## Steps to Reproduce the Problem

  1. OMR active
  2. discord soft install on a computer on the lan
  3.

## Specifications

  - OpenMPTCProuter version: 0.54.8
  - OpenMPTCProuter VPS version: 1015
  - OpenMPTCProuter platform: RPI3

Please help ;-)

try OMR bypass for discordapp.com it doesnt work

Ysurac, 2020-04-04:
Never tested discord...
This doesn't work in which side ? send, receive or both ?

vinceducat, 2020-04-04:
Both :-(

vinceducat, 2020-04-04:
Discord seems to be base on Webrtc ... https://support.discordapp.com/hc/fr/articles/115001310031-Erreurs-avec-la-connexion-vocale

vinceducat, 2020-04-04:
make the test on test.webrtc ... pb of reflexivity

some information but not qn expert https://github.com/webrtc/testrtc/issues/176 perhaps can help you

Ysurac, 2020-04-04:
Maybe some port to open for discord. I have no idea how I can test discord.

vinceducat, 2020-04-04:
i will check for discord port ..

you think opening port in OMR or VPS ?

i think test could me made using the https://test.webrtc.org/

Ysurac, 2020-04-04:
redirecting port from the VPS to OMR. But I have really no idea.

vinceducat, 2020-04-04:
Paramètres du VPS
vps
Rediriger tous les ports du server vers ce routeur
Désactiver les redirections de ports définis dans la partie pare-feu du serveur vers ce routeur

the 2 options must be on ?

Ysurac, 2020-04-04:
no and it's better to define the port that need to be redirected instead of redirecting all ports.

vinceducat, 2020-04-04:
ok ;-)
 

in witch menu can i do this ?

Ysurac, 2020-04-04:
Network->Firewall
You can follow doc: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

vinceducat, 2020-04-04:
thanks i will check that...

doc says that discord need large range of udp port 50000-65535 :-(

Ysurac, 2020-04-04:
Port 65000 to 65535 are reserved for OMR API, VPN,....

vinceducat, 2020-04-04:
oups ... 

that s the problem .

i made some test, i have a french ISP from Free Freebox V5 and when i in direct not throught OMR all is ok.
with OMR i have no problem with putting my computer lan ip in omr bypass by ip source..

perhaps need to define something in port or in protocole section ... to restrict bypass only for Discord...

Ysurac, 2020-04-04:
in status page all is green and the VPN was not disabled ?

vinceducat, 2020-04-04:
yes all is good.

vinceducat, 2020-04-04:
perhaps test other vpn than Glory TCP ?

cron410, 2020-04-05:
Have you tried putting `gateway.discord.gg` in OMR-Bypass?

vinceducat, 2020-04-05:
> 

> 
> 
> Have you tried putting `gateway.discord.gg` in OMR-Bypass?

in "Domain" sub menu in OMR-Bypass ??

vinceducat, 2020-04-05:
i try to add it in domain sub menu discord voice et audio doesnt work.

Ysurac, 2020-04-05:
Do you know a way I can test without having any friend on discord ?

vinceducat, 2020-04-05:
not really... il will think about that ..

vinceducat, 2020-04-05:
if you have an account on discord i have one too .. we could be friend for test :-)

vinceducat, 2020-04-18:
@Ysurac always my problem with discord

global omr-bypass work but i will try to limite bypass to "discord" so i find that perhaps this could help me to set-up the omr-bypass or to find a solution 

https://medium.com/tenable-techblog/lets-reverse-engineer-discord-1976773f4626

vinceducat, 2020-04-18:
new test - new info

video and audio ok when i m disabling ShadowSocks and leave glorytun tcp on ....

vinceducat, 2020-04-18:
@Ysurac discord seems to used port 50000-65535 is it possible to change shadowsocks port ? ie 49999 to make a test ?

Ysurac, 2020-04-18:
This test is working or not: https://support.discordapp.com/hc/en-us/articles/360020641332-Mic-Testing ?

vinceducat, 2020-04-18:
yep it s good ;-)

Th0rGer, 2020-05-28:
[image]
have you tried to let rtp traffic bypass mptcp?

vinceducat, 2020-05-28:
yes i try dont change anything but i will retry

vinceducat, 2020-05-30:
retry doesnt work...

closed 2020-10-03 by github-actions[bot]