Ysurac/openmptcprouter #952: Problème débit

couili, 2020-04-04:
## Expected Behavior

Problème de débit en download, aucun problème avec l'upload.

## Actual Behavior
Depuis la dernière version, j'ai de gros problèmes de débit. Je pense que le problème se situe au niveau du VPS. J'ai essayé de tout réinstallé à ce niveau là, j'ai bien obtenu mon débit normal, mais pas pendant longtemps. 

## Steps to Reproduce the Problem

  1. Faire un test de débit
  2. Avoir que 30 Mbits au lieu de environ 70-80 en download via mptcp.
  3. Avoir 70 mégas via 4G et 16 via ADSL.

## Specifications

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0
  - OpenMPTCProuter VPS version: Version 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-04-04:
Tout est vert dans la page status  ?
Que donne ```omr-test-speed``` et ```omr-test-speed wan1``` si l'interface réelle est wan1, pareil pour toute les interfaces. Ctrl+C pour arrêter le test après 1-2 minutes au moins.

couili, 2020-04-04:
Effectivement tout est vert (j'ai juste l'erreur VPN tunnel down de temps en temps en cas de forte utilisation, mais en général ça revient à la normal tout seul).
[image]

Voici le résultat du omr-test-speed :

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3 10.0G    3  348M    0     0  2572k      0  1:07:56  0:02:18  1:05:38 2868k

omr-test-speed wan1 :
root@OpenMPTCProuter:~# omr-test-speed eth1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  2 10.0G    2  240M    0     0  1904k      0  1:31:47  0:02:09  1:29:38 1926k

omr-test-speed wan2 :
root@OpenMPTCProuter:~# omr-test-speed eth2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  6 10.0G    6  623M    0     0  4617k      0  0:37:50  0:02:18  0:35:32 4008k

vinceducat, 2020-04-04:
de mon cote j aimerai aussi comprendre...

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 10.0G    1  189M    0     0  1479k      0  1:58:05  0:02:11  1:55:54 1613k^C
root@OpenMPTCProuter:~# omr-test-speed usb0
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 10.0G    4  495M    0     0  2683k      0  1:05:06  0:03:08  1:01:58 3195k^C
root@OpenMPTCProuter:~# omr-test-speed eth1
curl: (45) Couldn't bind to 'eth1'

root@OpenMPTCProuter:~# omr-test-speed wan1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 49.0M    0     0   262k      0 11:07:00  0:03:11 11:03:49  301k^C

il semble que omr-test-speed soit le debit agrege mais qu il soit bien inferieur au debit de la somme de usb0 eth1 et wan 1

Ysurac, 2020-04-05:
Il faut tester l'agrégation, donc juste ```omr-test-speed```, en enlevant/désactivant chaque interface (pour désactiver ça peut se faire dans Réseau->MPTCP, une interface doit toujours être maitre).

Sur la plateforme du routeur il me faudrait plus de détails pour vous deux : une VM ? autre chose ? CPU ? mémoire ?
Il me faudrait aussi le fournisseur du VPS afin d'éliminer cette cause possible.

couili, 2020-04-05:
1er test uniquement avec ADSL et 2nd uniquement avec 4G.

root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  1 10.0G    1  122M    0     0  1730k      0  1:41:01  0:01:12  1:39:49 1820k^C
root@OpenMPTCProuter:~# omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 10.0G    4  476M    0     0  6768k      0  0:25:49  0:01:12  0:24:37 6577k^C

Pour mon routeur, il tourne sur une VM, avec 4 vCPU et 4GB de RAM. Et mon VPS est chez Scaleway.
Ce qui est curieux, c'est que j'ai l'impression que c'est aléatoire. Ce matin, en débit agrégé, j'ai 90 mégas ...

Ysurac, 2020-04-05:
Peut être un soucis de liaison saturé entre un fournisseur et le VPS alors ou alors beaucoup de paquets perdu sur une liaison ce que MPTCP n'apprécie pas forcément.
Dans Réseau->MPTCP tu peux essayer le scheduler BLEST ou ECF, ça passe normalement mieux quand il y a une connexion 4G dans le lot.
Sinon changer le TCP Congestion Control peut parfois améliorer les choses.

couili, 2020-04-05:
J'y ai pensé c'est vrai, je suis en contact avec bouygues pour qu'ils regardent de leur côté que tout aille bien. Chez scaleway, ils m'ont fait changé d'hyperviseur, mais pas mieux, donc je ne pense pas que cela vienne de chez eux.
Je test tout ça et je reviens vers toi.
Merci de ton aide !

vinceducat, 2020-04-05:
OMR RPI3
VPS VM chez Milkywan

resulat liaison 4g seul via usb tethering puis adssl seul

omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 40.2M    0     0  3670k      0  0:47:36  0:00:11  0:47:25 3612k^C
 omr-test-speed
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0 10.0G    0 84.8M    0     0  1195k      0  2:26:12  0:01:12  2:25:00 1196k^C

je vais tester pour scheduler et tcp cogestion control....
merci pour ton aide

closed 2020-04-23 by couili