Ysurac/openmptcprouter #954: Disk management and more

gitxmax, 2020-04-05:
Have a couple of questions..
I have openmptcpcomputer  installed on a 120GB ssd.
Is it possible to expand the root partition, from within the system?
Since openmptcpcomputer is missing the mounting points tab and the block-mount package, what's the right way of managing the disk(s)? I would like to use some nas features and can't figure out how to do it here.
Thanks

Ysurac, 2020-04-05:
You can use parted to resize rootfs partition, or compile the image yourself.
For NAS, there is many packages available. OpenMPTCProuter is based on OpenWrt.

salchris, 2020-04-21:
i'm following tutorial on openwrt for usb storage etc etc... but they all need block-mount, which is not available in SW repositories..

How do i install block command? Thank you

BTW.. openmptcprouter rocks! Thank you

closed 2020-09-14 by github-actions[bot]