Ysurac/openmptcprouter #956: Enhancement in debug logs: getpeername: Socket not connected

xabix, 2020-04-06:
## Expected Behavior

Hello,

Can this error be more precise to know which wan is failling?

1/ daemon.err /usr/bin/ss-local[19207]: getpeername: Socket not connected
2/ daemon.err /usr/bin/ss-redir[19209]: remote recv: Operation timed out
3/ user.notice post-tracking-post-tracking: omrvpn down because gateway down

In general, is there a way to increase the log level to have more insight on the errors?

Would be nice to log the interface on which these messages relate to?

## Specifications

  - OpenMPTCProuter version: v0.54.7 r0+12509-04a21c26a0

Ysurac, 2020-04-06:
Difficult to add more details to these errors:
* ss-local and ss-redir are related to shadowsocks (not possible to be more verbose here)
* omrvpn down because gateway down mean that the gateway of interface omrvpn doesn't answer to ping, the gateway is the local address of the VPN on the VPS side. So the VPN is down.
* The "tunnel down use shadowsocks for UDP" mean that the VPN doesn't answer, so shadowsocks is used for UDP and this mean no aggregation for UDP

VPN down error are related to the connection between WANs and the VPS.

xabix, 2020-04-06:
My setup is to use 
[image]

Therefore when Glory fails there is an automatic fall back to Shadowsocks in UDP. Is there a way to control the fall back? what I mean is having 3 different WANs with ADSL and 4G, I would be suprised that none of them work at the same time. So the issue should be in the VPS side (as locally if OMR is failing nothing will work whatever the WANs and the VPS).

If I look on the VPS I see:

> Apr  6 15:09:32 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:09:35 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:10:10 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:10:39 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:11:47 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:11:52 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:11:53 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:12:26 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:12:27 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:14:24 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:14:26 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:14:50 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:15:10 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:15:26 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:16:08 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:16:22 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:16:23 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:17:12 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:17:18 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:17:20 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:17:41 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:17:51 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:17:52 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:18:05 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:18:08 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:18:11 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:18:53 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:18:59 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:21:03 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:22:59 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:23:25 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:23:35 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:23:43 XabiX ss-server[896]: remote recv: Connection reset by peer
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    password true
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config...
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config... shadowsocks
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config... glorytun
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config... dsvpn
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config... iperf3
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config... openvpn
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config... mlvpn
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config... mptcp
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config... ipv6
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    get server IPv4
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    get hostname
> Apr  6 15:24:01 XabiX omr-admin.py[1985]: DEBUG:    Get config: done
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    password true
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    password true
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config...
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config... shadowsocks
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config... glorytun
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config... dsvpn
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config... iperf3
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config... openvpn
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config... mlvpn
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config... mptcp
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config... ipv6
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    get server IPv4
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    get hostname
> Apr  6 15:24:02 XabiX omr-admin.py[1985]: DEBUG:    Get config: done
> Apr  6 15:24:03 XabiX omr-admin.py[1985]: DEBUG:    Get config...
> Apr  6 15:24:03 XabiX omr-admin.py[1985]: DEBUG:    Get config... shadowsocks
> Apr  6 15:24:03 XabiX omr-admin.py[1985]: DEBUG:    Get config... glorytun
[... 30 lines cut ...]
> Apr  6 15:24:03 XabiX omr-admin.py[1985]: DEBUG:    get server IPv4
> Apr  6 15:24:03 XabiX omr-admin.py[1985]: DEBUG:    get hostname
> Apr  6 15:24:03 XabiX omr-admin.py[1985]: DEBUG:    Get config: done
> Apr  6 15:25:27 XabiX ss-server[896]: server recv: Connection reset by peer
[... cut ...]
> Apr  6 15:24:57 XabiX kernel: [595867.980907] Shorewall:net-fw:DROP:IN=eth0 OUT= MAC=a6:ee:69:bf:0d:d8:00:1c:73:44:59:c6:08:00 SRC=185.175.93.11 DST=149.91.xxx.xxx LEN=40 TOS=0x00 PREC=0x00 TTL=250 ID=39988 PROTO=TCP SPT=57059 DPT=38413 WINDOW=1024 RES=0x00 SYN URGP=0 

By the way, I noticed the same type of logs between message, syslog and daemon etc... maybe worth reducing the number of fils logging the shorewall deny logs as they seem duplicated in 3 files. I don't think this as any impact on the VPS side though.
[image]

Ysurac, 2020-04-06:
You can configure what you want on VPS side... Log are the default config.
You can set MPTCP master as the interface that is more stable (ADSL, fiber,...). But, as I said, the problem can be between WANs and VPS.

xabix, 2020-04-07:
Thanks Ysurac! As alwayes very helpful.

I will not touch the log level in the VPS as I don't want to mess up :) And yes I have my Master interface set to ADSL Orange which is most stable and quick WAN so not sure what can happen betweeen my WANs and the VPS :). Anyway I may give a try with another VPS. I am using https://www.pulseheberg.com/vps

xabix, 2020-04-07:
Hello,
I still face the issue:
Apr  7 13:49:24 XabiX ss-server[896]: remote recv: Connection reset by peer

Fullmesh

> Index, Address-ID, Backup, IP-address, if-idx
> IPv4, next v4-index: 5
> 2, 3, 0, 192.168.2.2 7
> 3, 4, 0, 192.168.3.2 8
> 4, 5, 0, 192.168.9.2 9
> IPv6, next v6-index: 2

WAN1
> 1: 192.168.2.1 
> 2: * 
> 3: 193.249.214.206  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
> 4: 193.252.100.34  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
> 5: 193.251.126.10  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
> 6: 193.252.98.102  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
> 7: 193.253.13.206 IP::TTL IP::CheckSum 
> 8: * 
> 9: 178.255.102.44 IP::TTL IP::CheckSum 
> 10: 149.91.xxx.xxx TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 
> 
> Result:
> < Ethernet (14 bytes) :: DestinationMAC = 7a:cb:0a:6b:6f:ec , SourceMAC = 48:83:c7:38:0d:60 , Type = 0x800 , >
> < IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 53 , Protocol = 0x6 , CheckSum = 0x950d , SourceIP = 149.91.xxx.xxx , DestinationIP = 192.168.2.2 , >
> < TCP (20 bytes) :: SrcPort = 65101 , DstPort = 36415 , SeqNumber = 1592583445 , AckNumber = 2097620317 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 65535 , CheckSum = 0x4791 , UrgPointer = 0 , >
> < TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
> < TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 9884525243855844440 , >

WAN2

> 1: 192.168.3.1 
> 2: 194.149.169.69 IP::TTL IP::CheckSum 
> 3: 194.149.172.102 IP::TTL IP::CheckSum 
> 4: 37.77.34.38 IP::TTL IP::CheckSum 
> 5: 37.77.34.75 IP::TTL IP::CheckSum 
> 6: 178.255.102.44 IP::TTL IP::CheckSum 
> 7: 149.91.xxx.xxx TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 
> 
> Result:
> < Ethernet (14 bytes) :: DestinationMAC = 9a:e3:fe:56:63:f3 , SourceMAC = 00:07:cb:4f:97:51 , Type = 0x800 , >
> < IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 56 , Protocol = 0x6 , CheckSum = 0x910d , SourceIP = 149.91.xxx.xxx , DestinationIP = 192.168.3.2 , >
> < TCP (20 bytes) :: SrcPort = 65101 , DstPort = 42359 , SeqNumber = 1996887494 , AckNumber = 1279959361 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 65535 , CheckSum = 0xd74d , UrgPointer = 0 , >
> < TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
> < TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 2149819598405958871 , >

WAN3:

> 1: 192.168.9.1 
> 2: * 
> 3: * 
> 4: 10.64.21.83 TCP::DataOffset TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
> 5: * 
> 6: * 
> 7: * 
> 8: * 
> 9: * 
> 10: * 
> 11: * 
> 12: * 
> 13: * 
> 14: * 
> 15: * 
> 16: * 
> 17: * 
> 18: 149.91.xxx.xxx TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::DiffServicesCP IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 
> 
[... 51 lines cut ...]
[27245.980346]  irq_exit+0x62/0x90
[27245.980535]  do_IRQ+0x7f/0xe0
[27245.980723]  common_interrupt+0xf/0xf
[27245.980919]  </IRQ>
[27245.981096] RIP: 0010:native_safe_halt+0x17/0x20
[27245.981390] Code: 00 a8 08 75 c3 e9 77 ff ff ff 90 90 90 90 90 90 90 90 90 55 8b 05 69 24 bb 00 48 89 e5 85 c0 7e 07 0f 00 2d fb 1d 5f 00 fb f4 <5d> c3 0f 1f 80 00 00 00 00 55 8b 05 49 24 bb 00 48 89 e5 85 c0 7e
[27245.981979] RSP: 0018:ffffffff82203e48 EFLAGS: 00000246 ORIG_RAX: ffffffffffffffd1
[27245.982326] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 000000000001a440
[27245.982567] RDX: 000000000296b062 RSI: ffffffff82233250 RDI: 0000000000000000
[27245.982807] RBP: ffffffff82203e48 R08: 0000000000000124 R09: 0000000000000000
[27245.983047] R10: 8080808080808080 R11: 0000000000000000 R12: 0000000000000000
[27245.983291] R13: 0000000000000000 R14: ffff88801ffd58c0 R15: 0000000000000000
[27245.983533]  ? tsc_verify_tsc_adjust+0x39/0xc0
[27245.983739]  default_idle+0x10/0x20
[27245.983934]  arch_cpu_idle+0x10/0x20
[27245.984214]  default_idle_call+0x27/0x30
[27245.984415]  do_idle+0x164/0x180
[27245.984607]  cpu_startup_entry+0x5f/0x70
[27245.984807]  rest_init+0x99/0x9b
[27245.984998]  start_kernel+0x554/0x561
[27245.985195]  x86_64_start_reservations+0x29/0x2b
[27245.985403]  x86_64_start_kernel+0x71/0x74
[27245.985606]  secondary_startup_64+0xa4/0xb0
[27245.985809] ---[ end trace 8f8ebb1ae243217a ]---
```

xabix, 2020-04-15:
@Ysurac Hi Yannick, any change to tell me if the call trace above can help to pinpoint the issue? otherwise I may close the issue and wait for the new kernel 5.4 :). When do you expect to release 0.55?

Merci

Ysurac, 2020-04-15:
This looks like an issue with the VM, like not enough resources really available.

closed 2020-06-15 by xabix