Ysurac/openmptcprouter #957: unable to mptcprouter page at 192.168.100.1

rajava2010, 2020-04-06:
## Expected Behavior

Expects to open the mptcprouter web page for further configuration @ https://192168.100.1

## Actual Behavior

Downloaded VirtualBox mptcprouter image and started in Mac OS X host 10.13.6,  with 2 adapters 1) with bridged / promiscous mode 2) Host-Only Adapter with 192.168.100.1/24 , after booting not able to access the https://192.168.100.1 from the host. 

## Steps to Reproduce the Problem

  1. check above in actual behaviour.
  2.
  3.

## Specifications

  - OpenMPTCProuter version:  openmptcprouter-v0.54.7-r012509-04a21c26a0-x86-64-combined-ext4-efi.vdi 
  - OpenMPTCProuter VPS version: 0.1013
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-04-06:
Why this host only adapter ?

rajava2010, 2020-04-06:
Even tried without host-only adapter but still failed which is why I created the adapter and tried but both fails.

Ysurac, 2020-04-06:
Set one interface as bridge, promiscous as Intel PRO/1000 MT and add an IP in the 192.168.100.0/24 network, like 192.168.100.2 to your local computer.

rajava2010, 2020-04-06:
It worked now after manually setting my wifi en0 interface to 192.168.100.2  and VirtualBox VM Adapter to Bridged Adapter with Promiscous Mode - Allow All. Thanks for your help.

closed 2020-04-06 by rajava2010