Ysurac/openmptcprouter #958: VPN down. Aggregation not working correctly

uryupinsk, 2020-04-07:
## Expected Behavior

I expect the VPN (Glorytun TCP or DSVPN) connection to be up at all time. The aggregation needs to be fully working on 2 WAN.

## Actual Behavior

Since OMR 0.54.7 and VPS 0.1015, there are very serious VPN problems, both on Glorytun TCP or DSVPN. 

With DSVPN, the VPN is down erratically after a few minutes and does not go up again without a complete reboot of the router. With Glorytun TCP, it restarts itself from time to time without any apparent reason.

As for the aggregation, it works on actual browsing or downloading at all times. But, after a few minutes, when I download a file on a server exposed to the Internet, or through a Wireguard or OpenVPN site-to-site tunnel, only the master WAN is used. I have to reboot the router for the 2 WAN to be aggregated again.

## Steps to Reproduce the Problem

  1. Start OMR router with Glorytun TCP or DSVPN.
  2. Wait a few minutes
  3. Start a download on a nginx server exposed on the Internet or through a OpenVPN or Wireguard tunnel
 4. Only master WAN is used
 5. Reboot OMR router
 6. All WAN are fully aggregated again

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform: x86_64
  - VPS provider: Scaleway

Ysurac, 2020-04-07:
I need the log when the VPN is detected as down (In State->System log).

uryupinsk, 2020-04-07:
Here are the OMR router system log and settings :

- [OMR System Log](https://github.com/Ysurac/openmptcprouter/files/4443443/omr.system.log.txt)
- [OMR settings](https://github.com/Ysurac/openmptcprouter/files/4443476/omr.settings.txt)

Ysurac, 2020-04-07:
Try Another MPTCP scheduler and use 1 subflow by ip.

uryupinsk, 2020-04-07:
Same problem after switching to BLEST and using 1 subflow per IP.

I saw that the same problem is occurring on OverTheBox with Free LTE WANs. The OTB team said that it could be caused by an MTU too large and thus tell some users to put it down to 1432, which I did, but same problem again.

What's very strange is that SpeedTest.net, nPerf and DSLreports are all aggregating correctly.

Ysurac, 2020-04-07:
I have a Free LTE WANs and I don't have this issue.
Check in Network->MPTCP and MPTCP Fullmesh tab that you have all WANs IPs when you have the problem. Also check the result of ```multipath``` via SSH on the router.

uryupinsk, 2020-04-07:
Result of MPTCP Fullmesh tab:

```
Index, Address-ID, Backup, IP-address, if-idx
IPv4, next v4-index: 6
3, 4, 0, 192.168.10.141 11
5, 6, 0, 192.168.20.2 26
IPv6, next v6-index: 1
```

Result of `multipath`:

```
eth0 is deactivated
ifb0 is deactivated
ifb1 is deactivated
ifb4tun0 is deactivated
ifb4wan1 is deactivated
lo is deactivated
sit0 is deactivated
teql0 is deactivated
tun0 is deactivated
wan1 is in default mode
wan2 is in default mode
```

Ysurac, 2020-04-07:
All seems ok.
You can try to enable Multipath Debug in Network->MPTCP and check if you have error in system log after that.

uryupinsk, 2020-04-07:
Here is the [system log with mptcp debug activated](https://github.com/Ysurac/openmptcprouter/files/4444957/system.log.mptcp.debug.txt) when the problem is occurring.

uryupinsk, 2020-04-08:
I made a few more tests today. 

When I'm doing a speedtest, the SQM QoS is working correctly and keeping my ping at stable values. However, when I'm uploading files with FTP or HTTP by making multiple connections, it seems the QoS is not working at all and my ping is exploding at values > 1000 ms.

I have cake/piece_of_cake on WAN1 with download and upload speeds set at 80% of total value, nothing on WAN2 because it's LTE.

Ysurac, 2020-04-08:
You tried some other Queuing disciplines ?

uryupinsk, 2020-04-10:
I tested all queuing disciplines available but none of them is working. I cannot understand why QoS is working when doing a Speedtest or downloading a file, but not when I'm uploading files.

closed 2020-06-19 by uryupinsk