Ysurac/openmptcprouter #959: OpenMPTCProuter v0.55

Ysurac, 2020-04-07:
This issue is dedicated to OpenMPTCProuter v0.55.

For now nothing is tagged, it's only available as snapshot and not yet fully tested.

New features:
* Kernel 5.4 based
* interfaces omr-bypass and MPTCP ported to javascript
* Load balancing support when no VPS available

Nothing is tested yet.

xabix, 2020-04-30:
How is the new release looking? Would be happy to move to a newer kernel to see if we get better performances on virtualized environment. 
Beta 1 is there so not too long before we get something stable :)

brazzo97, 2020-05-01:
hi hyper-v is nomore supported ? thanks

Ysurac, 2020-05-02:
Hyper-V and ova images are converted from .img images via a script only for release.

VoightKampff, 2020-05-03:
Thank you for making the release process more public. I installed beta 2 and I can report it works as expected for a basic setup aggregating two WANs. What are the areas you would like testing/feedback on the most?

Ysurac, 2020-05-03:
For now it's general testing, if it's working for a basic usage it's good. New features and some MPTCP patches will be added for next beta.
What is the platform you use ? x86_64, RPI,... ? You tried an update or a new install ?

brazzo97, 2020-05-03:
can confirm that it works well on hyperv

VoightKampff, 2020-05-03:
# Info
I'm using a fresh image - `openmptcprouter-v0.55beta2-r0+13089-3fdb08681b-x86-64-generic-ext4-combined.img` on KVM/libvirt on an ancient hp 6005 with an `AMD Athlon(tm) II X2 B24 Processor` (it doesn't have aes, if that matters at all). I gave the VM 1 gig of ram.

The VPS is running a fresh install of debian 10 @ the Warsaw ovh DC. 1 vCore / 2  GB RAM / 40  GB NVMe / 250 Mbps, which is the lowest "normal" option they have in the range. (OVH changed their offer recently)

# Bugs

## Multiple 404 errors in the admin panel
The message is always in the following format
```
404 Not Found
Sorry, the object you requested was not found.

Unable to dispatch: /cgi-bin/luci/admin/system/openmptcprouter
```
If the interface has tab buttons, you can still get to a sub-page, just the default is broken. But if there are no tabs, the page is completely inaccessible.
This occurs in these areas:
* System -> OpenMPTCProuter 
* Services -> OMR tracker
* Services -> proxy dns...
* Services -> Nginx...
* Services -> Email
* Network -> Quota
* Network -> Snmp

## Debug UI polling feature (presumably) is left on
There is a label in the top right corner of the screen that says `Refreshing` or, if you click it `Paused` that seems to disable polling for ui updates.

## Encryption setting in wizard is not always honored
Go to Wizard -> Advanced settings -> ShadowSocks settings. Disable ShadowSocks. Set encryption to `None`. Note the label which says `Encryption method is also used for Glorytun`. This setting does not seem to be applied to glorytun's config on the client side. In `VPN settings` below, if `Default VPN` is set to `Glorytun UDP`, the status screen will show `Direct connection`. Running ` glorytun-udp show` returns `cipher: chacha20poly1305`. If I go to VPN -> Glorytun and manually disable the `chacha20` checkbox, the tunnel starts working. If I change `Default VPN` to `Glorytun TCP`, the `chacha20` checkbox in VPN -> Glorytun is still checked. I do not know how to check the status of the old tcp version of glorytun from the console or if that is even possible. I have not checked other VPN options.

## Display issues on the status page for huawei modems
* There is a tiny label to the right of the signal bars, which reads `lte` (you have to check it in the DOM inspector, otherwise it's impossible to read)
* The `operator` label is possibly showing only the first letter, in my case `O` - it should probably read `Orange`.
* It would be nice to have even more information, such as the frequency band, signals, sinr, etc. Perhaps enabled by a checkbox in the `Advanced Settings` section?

## MPTCP over VPN does not work
When I select the `MPTCP over VPN` option on both of my WAN connections, the status page shows `No ping to gateway in x seconds` warnings for both wan connections  (I didn't copy the exact text - going from memory). All traffic is going to the VPS through a single WAN connection bypassing the openvpn tunnels that were created, despise the status page indicating that the WANs are now the openvpn endpoints, not the real wan connections. After I restarted, the `no ping...` message was no longer shown, and the status page was all "green" and indicated that traffic would be going through the openvpn tunnels, but still all of the traffic was going to the VPS through a single WAN omitting the openvpn tunnels.

An additional issue is that after I unchecked the `MPTCP over VPN` checkboxes, MPTCP was set to `disabled` on all interfaces and I had to manually set it back to `master` and `enabled` in the Network->MPTCP screen.

I am certain this feature worked correctly when it was first introduced, because I remember testing it for a while when one of my ISPs was intermittently blocking MPTCP. (Thank you very much for implementing this for me :) ) I do not remember the specific version number, but it was probably the first version that had it.

## Questions
* No documentation on the `Ballancing` option in Master interface selection feature. Was `Redundant` simply renamed to `Ballancing`?
* How does the `Load balancing support when no VPS available` feature work? How does it compare to mwan3, which is a pretty popular multi wan package for vanilla openwrt?

Ysurac, 2020-05-03:
404 errors are fixed in my local repo (a typo...)
I will try to fix other bugs for next beta.

Redundant was renamed to balancing and it's related to the load balancing support. I think mwan3 do that in the same way, no idea. Multiples routes are created to the VPS when available and as default routes using all connections when no VPS available. So when the VPS is available, this should help for failover and without any VPS at least this provide a dirty load balancing.

salchris, 2020-05-06:
rpi4 - updating SW lists

Bonding works... still testing. Thank you.
(blockd and blockmount present! thank you)

[image]

Ysurac, 2020-05-06:
Package list should be fixed now.

salchris, 2020-05-06:
not yet

`Collected errors:
 * opkg_download: Failed to download http://openmptcprouter.com:80/release/5.4/targets/bcm27xx/bcm2711/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://openmptcprouter.com:80/release/5.4/packages/aarch64_cortex-a72/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://openmptcprouter.com:80/release/5.4/packages/aarch64_cortex-a72/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://openmptcprouter.com:80/release/5.4/packages/aarch64_cortex-a72/openmptcprouter/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://openmptcprouter.com:80/release/5.4/packages/aarch64_cortex-a72/packages/Packages.gz, wget returned 8.
`

VoightKampff, 2020-05-06:
@Ysurac What kind of performance should I be getting with the updated version of glorytun (0.3.4)? I've peeked into the OTB repository, and I didn't see any reference to shadowsocks in the master branch there. They might be relying just on glorytun. However, I am getting pretty bad performance when I disable shadowsocks and set glorytun to UDP in 0.55beta2. glorytun reports that it's using 2 paths, so that's correct, and it seems I am getting aggregation. But the bandwidth gets distributed equally between both my LTE connections, and it is only around 50% of the slower one for both connections (I checked this from the modems status pages). So the total aggregated bandwidth is pretty much only what my slower LTE connection offers. Does glorytun only work well if the connections are "similar" in terms of bandwidth/latency/etc. ?

Ysurac, 2020-05-06:
@salchris this time this should be fixed.

salchris, 2020-05-06:
yes it is! thank you

Ysurac, 2020-05-06:
@VoightKampff It's true, OTB seems to use only glorytun UDP in the repo. For now, I always get bad perf with glorytun UDP, I have no idea in which case (if any) glorytun UDP work well.

VoightKampff, 2020-05-06:
@Ysurac It seems that omr-huawei was broken by this change https://github.com/Ysurac/openmptcprouter-feeds/commit/d76151eef58949a6d07b38e37152f48552bbaab2
```
root@OpenMPTCProuter:~# /bin/omr-huawei 192.168.8.1 all
;;connected;lte
root@OpenMPTCProuter:~# ./omr-huawei_beta2 192.168.8.1 all
100;O;;connected;lte
```

I didn't test anything else yet. Maybe tomorrow.

Ysurac, 2020-05-06:
@VoightKampff I found the issue, this will be fixed in next release.

xabix, 2020-05-06:
Having issues with the Beta 3 on the VPS, I get:

> May  6 23:46:29 XabiX omr-admin.py[1467]: DEBUG:    Get config...
> May  6 23:46:29 XabiX omr-admin.py[1467]: DEBUG:    Get config... shadowsocks
> May  6 23:46:29 XabiX omr-admin.py[1467]: ERROR:    Exception in ASGI application
> May  6 23:46:29 XabiX omr-admin.py[1467]: Traceback (most recent call last):
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/uvicorn/protocols/http/httptools_impl.py", line 385, in run_asgi
> May  6 23:46:29 XabiX omr-admin.py[1467]:     result = await app(self.scope, self.receive, self.send)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/uvicorn/middleware/proxy_headers.py", line 45, in __call__
> May  6 23:46:29 XabiX omr-admin.py[1467]:     return await self.app(scope, receive, send)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/fastapi/applications.py", line 149, in __call__
> May  6 23:46:29 XabiX omr-admin.py[1467]:     await super().__call__(scope, receive, send)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/starlette/applications.py", line 102, in __call__
> May  6 23:46:29 XabiX omr-admin.py[1467]:     await self.middleware_stack(scope, receive, send)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/starlette/middleware/errors.py", line 181, in __call__
> May  6 23:46:29 XabiX omr-admin.py[1467]:     raise exc from None
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/starlette/middleware/errors.py", line 159, in __call__
> May  6 23:46:29 XabiX omr-admin.py[1467]:     await self.app(scope, receive, _send)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/starlette/exceptions.py", line 82, in __call__
> May  6 23:46:29 XabiX omr-admin.py[1467]:     raise exc from None
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/starlette/exceptions.py", line 71, in __call__
> May  6 23:46:29 XabiX omr-admin.py[1467]:     await self.app(scope, receive, sender)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/starlette/routing.py", line 550, in __call__
> May  6 23:46:29 XabiX omr-admin.py[1467]:     await route.handle(scope, receive, send)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/starlette/routing.py", line 227, in handle
> May  6 23:46:29 XabiX omr-admin.py[1467]:     await self.app(scope, receive, send)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/starlette/routing.py", line 41, in app
> May  6 23:46:29 XabiX omr-admin.py[1467]:     response = await func(request)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/fastapi/routing.py", line 197, in app
> May  6 23:46:29 XabiX omr-admin.py[1467]:     dependant=dependant, values=values, is_coroutine=is_coroutine
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/lib/python3.7/dist-packages/fastapi/routing.py", line 148, in run_endpoint_function
> May  6 23:46:29 XabiX omr-admin.py[1467]:     return await dependant.call(**values)
> May  6 23:46:29 XabiX omr-admin.py[1467]:   File "/usr/local/bin/omr-admin.py", line 613, in config
> May  6 23:46:29 XabiX omr-admin.py[1467]:     shadowsocks_key = data["port_key"][str(shadowsocks_port)]
> May  6 23:46:29 XabiX omr-admin.py[1467]: KeyError: 'port_key'

Ysurac, 2020-05-07:
What do you have in ```/etc/shadowsocks-libev/manager.json``` on the VPS ?

xabix, 2020-05-07:
> /etc/shadowsocks-libev/manager.json

root@XabiX:~# ls -l /etc/shadowsocks-libev/manager.json
-rw-r--r-- 1 root root 531 May  6 22:30 /etc/shadowsocks-libev/manager.json

Ysurac, 2020-05-07:
I think this was an update. I fixed the script for that.

xabix, 2020-05-07:
> I think this was an update. I fixed the script for that.

How can I update to fix the issue then?
[image]

Ysurac, 2020-05-07:
By reinstalling the VPS part from scratch or adding a comma at the line beginning by 65101 in ```/etc/shadowsocks-libev/manager.json```.

xabix, 2020-05-07:
> By reinstalling the VPS part from scratch or adding a comma at the line beginning by 65101 in `/etc/shadowsocks-libev/manager.json`.

[image]

Since I could save the Glory key, the only message I get now is:
May  7 10:26:56 XabiX omr-service[928]: /etc/glorytun-tcp/post.sh: 3: .: : not found

How can I troubleshoot further?

Ysurac, 2020-05-07:
The comma must be added at the end of the line : 
```"65101": "WpHPWs4LTD3ZL_xxx",```
For the post error, run the VPS script again.

xabix, 2020-05-07:
> The comma must be added at the end of the line :
> `"65101": "WpHPWs4LTD3ZL_xxx",`
> For the post error, run the VPS script again.

Did keeping only one instance (and not to lines). But the script doesn't finish

> Synchronizing state of haveged.service with SysV service script with /lib/systemd/systemd-sysv-install.
> Executing: /lib/systemd/systemd-sysv-install enable haveged
> E: Could not get lock /var/lib/dpkg/lock-frontend - open (11: Resource temporarily unavailable)
> E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?

weltmeyer, 2020-05-08:
the post.sh problem is existing for me too, fresh VM directly installed script.

after some research i found that:
`for intf in /etc/glorytun-udp/*; do`
returns the full path
so, the readlink does:
`readlink -f /etc/glorytun-udp//etc/glorytun-udp/tun0`
which is wrong, as it should do
`readlink -f /etc/glorytun-udp/tun0`
same is for the omr6-4 parts.

caller for glorytun-udp:
omr-service, line 22
`/etc/glorytun-udp/post.sh /etc/glorytun-udp/${intf}`
intf is already /etc/glorytun-udp/tun0

VoightKampff, 2020-05-11:
I noticed that I am getting a lot of drop outs / flapping. Here's the system log: https://gist.github.com/VoightKampff/c0922bb0d2c12441a903ef94bfb5963c
I'm also running 0.54.7 and it reports no errors such as `kern.err kernel: [340259.286623] __mptcp_init4_subsockets mptcp_add_sock failed ret: -1` in 0.54.7. It is of course running with a different VPS instance, but in the same data center from the same provider (OVH Warsaw DC).

vobo70, 2020-05-16:
Somebody answered in my topic on Banana PI R2 forum as related to very slow LAN speed:

I saw yesterday a new patch related to VLAN was applied in the latest 5.4 kernel tree and I guess it will be available in the next release.

https://patchwork.kernel.org/patch/11552461/

Allow DSA to add VLAN entries even if VLAN filtering is disabled, so enabling it will not block the traffic of existent ports in the bridge

I’m not sure if this is related to the traffic slowness I have not tested it.

Ysurac, 2020-05-16:
True, patches seems to be already applied :)
I will test speed.

vobo70, 2020-05-16:
I made some test and connected this card:
USB 3.0 ASIX Elec. Corp. AX88179
USB 3.0 to 1GbE LAN adapter
and the speed is OK:

maciek@bananapipro2:~$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.22 port 43650 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  18.6 MBytes   155 Mbits/sec    0    844 KBytes
[  5]   1.00-2.00   sec  16.8 MBytes   141 Mbits/sec    0   1.63 MBytes
[  5]   2.00-3.01   sec  16.2 MBytes   135 Mbits/sec    0   2.42 MBytes
[  5]   3.01-4.00   sec  15.0 MBytes   127 Mbits/sec    0   3.20 MBytes
[  5]   4.00-5.01   sec  16.2 MBytes   136 Mbits/sec    0   3.48 MBytes
[  5]   5.01-6.02   sec  16.2 MBytes   135 Mbits/sec    0   3.48 MBytes
[  5]   6.02-7.01   sec  16.2 MBytes   137 Mbits/sec    0   3.61 MBytes
[  5]   7.01-8.02   sec  16.2 MBytes   135 Mbits/sec    0   3.79 MBytes
[  5]   8.02-9.02   sec  16.2 MBytes   137 Mbits/sec    0   3.79 MBytes
[  5]   9.02-10.00  sec  16.2 MBytes   138 Mbits/sec    0   3.79 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   164 MBytes   138 Mbits/sec    0             sender
[  5]   0.00-10.19  sec   164 MBytes   135 Mbits/sec                  receiver

I understand that the problem is related to onboard Realtek LAN controlller.

Adorfer, 2020-05-17:
> Somebody answered in my topic on Banana PI R2 forum as related to very slow LAN speed:

that's not a problem of OpenMPTCProuter, nor OpenWRT, that's an issue of the RPI-Platform. it becomes better (still not even 80% linespeed) on RPI4.

Ysurac, 2020-05-17:
He is talking about BPI-R2, not RPI. I need to make tests to find why it's slow.

xabix, 2020-05-17:
FYI when 'updating the VPS script' at the end I got:

> [Sun May 17 20:31:09 UTC 2020] Create new order error. Le_OrderFinalize not found. {
>   "type": "urn:ietf:params:acme:error:rateLimited",
>   "detail": "Error creating new order :: too many certificates already issued for: ovh.net: see https://letsencrypt.org/docs/rate-limits/",
>   "status": 429
> }

Ysurac, 2020-05-18:
@xabix if the script doesn't stop here it's not a big issue: cert are used for v2ray based on reverse of the VPS IP.

vobo70, 2020-05-20:
Banana PI R2 - openmptcprouter 0.55 beta 4
when I connect USB 3.0 SSD the device does not appear in /dev/sd* only I can see that in dmesg:
[    3.852621] usb-storage 2-1.1:1.0: no of_node; not parsing pinctrl DT
[    3.852771] usbcore: registered new interface driver usb-storage
(besides that the router hangs when I do it "online")
what log should I provide to help me?

vobo70, 2020-05-21:
Instead connecting SSD i put 32GB card to Banana PI R2.
But for 2-3 days i can see disconnecting varius clients form AP running on BPI_R2.
I was running beta4 now I moved to beta5 (clean install).
I will report if it still will happends.

JorgeMoort89, 2020-05-21:
Can be implemented Full Cone NAT for additional public IP?

vinceducat, 2020-05-21:
@Ysurac  any planned date to the 0.55 release ?

Ysurac, 2020-05-21:
@JorgeMoort89 I never tested Full Cone NAT, it was requested and validated but I never used this yet.

@vinceducat when it will be ready :) There is never planned date. If I don't find major bug in current beta, release candidate 1 will be ready and I will test it on all arch. If RC1 is ready to production then it will be very soon (I don't think so...). But with some luck 0.55 should be ready next week or beginning of June. For that I need to rent a new server to compile/host images and all packages.

Adorfer, 2020-05-21:
i 

> If I don't find major bug in current beta,

what kind of input do you need "for natv6 not working even with current beta and fresh install on debian10 vps with fully working public /64"? New issue?

vobo70, 2020-05-21:
> For that I need to rent a new server to compile/host images and all packages.

can we support you with that "case"?
(I'm not reach but it's only way to thank you for your work)

vinceducat, 2020-05-21:
@vobo70  nice question how can you help you @Ysurac  ?

kevin39, 2020-05-21:
There is a donate button on https://www.openmptcprouter.com/ :)

Personnally i already donated few days ago. He changed my life ;)

Ysurac, 2020-05-22:
@Adorfer This will be fixed in next beta (I have also an error)
@vobo70 I want to own the server, I want to host also a translation tool for the interface. Then I will try to make possible to have some small mirrors and I will ask help here ;)
@vinceducat not for now, I have many. But they may be nice for mirrors when this will be available
@kevin39 and thanks for that, this will help to pay the new server :)

salchris, 2020-05-22:
> There is a donate button on https://www.openmptcprouter.com/ :)
> 
> Personnally i already donated few days ago. He changed my life ;)

Yes.. i think this is the best way. 
A small donation is a must.

Thank you Ysurac.

Adorfer, 2020-05-22:
despite having joined one of the patreon plans: 
i can offer a (new) VPS with a dedicated NVME and some 12 cores  (out of 32/64HT). 
currently the host machine is used to build full chain OpenWRT (here: Freifunk-Gluon). 
For the benchmark: If the target's toolchain is already build, a full target takes 55min each. 
Platform either jenkins or plain debian.

Ysurac, 2020-05-22:
@Adorfer thanks for the offer. For now my issue is more related to disk space and memory. For the build process I will see how I can use multiples servers.

Adorfer, 2020-05-22:
i can offer 500GB SSD and 16GB of RAM dedicated for this vps. it's just lacking a bit of connectivity: just 30MBit/s upload (it's sitting a basement rack connected via -surprise, surprise- openmptcprouter).

krissfr, 2020-06-02:
Hi Yannick,

will you provide an ubuntu 20.04 VPS script when the 0.55 will be release ?

thank you

tentious, 2020-06-02:
I'm running latest Beta 7 with latest VPS admin script. With Shadowsocks enabled, I'm having trouble connecting to my Arlo cameras. This didn't happen in 0.54. Glorytun TCP is now working(my preferred VPN) and I'm not running all connections via Glorytun. If I enable Shadowsocks, everything seems to work perfectly, except for connecting to the Arlo cameras. Very strange and haven't noticed any other effects. Since I use Arlo cameras as baby monitors, this is a show stopper, so I've disabled Shadowsocks for now.

Ysurac, 2020-06-02:
@krissfr I tested, this seems to work for must part, so yes I will make the VPS script available for Ubuntu 20.04

@tentious Cameras are on the lan or on another network/cloud ?

tentious, 2020-06-02:
> @tentious Cameras are on the lan or on another network/cloud ?

Arlo cameras are cloud connected. So they upload to cloud and whatever device you use to view the cameras also connect to cloud. There is no direct connection to the camera on local LAN. I'm eventually planning to move to on-premise/local LAN cameras, as cloud connected cameras have more weird problems than the time and effort they save to make things easy.

Ysurac, 2020-06-02:
@tentious do you have IPv6 enabled ?
Else it's difficult for me to test, there is no demo available on arlo website.

tentious, 2020-06-02:
IPv6 is disabled.

Understood that's it's a difficult one to test. It could actually be a problem with the IP cameras uploading video, not necessarily the playing/download of the video stream. I will test further from my phone connected via LTE vs local LAN to see if it's the upload or download.

tentious, 2020-06-02:
> I will test further from my phone connected via LTE vs local LAN to see if it's the upload or download.

So it's definitely upload from the camera to Netgear/Arlo cloud that isn't working. After enabling Shadowsocks again and viewing from my phone connected over LTE WAN, not local LAN, I'm still not able to view the camera. After disabling shadowsocks Arlo cameras work both from LAN and WAN.

Would OMR Bypass directly route/bypass Shadowsocks if configured? This might be a good workaround.

dettofatto, 2020-06-04:
On my raspberry pi4 0.55 beta 7 is much more stable and better than last stable...

xabix, 2020-06-08:
Hello, 

I am on beta7. I tried to swap to another VPS but every time I select disable on the primary and save then the tick box is not save so both seem active.

Some logs if it can help (not sure why curl complains)
> Mon Jun  8 18:06:20 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
> Mon Jun  8 18:06:20 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
> Mon Jun  8 18:06:20 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
> Mon Jun  8 18:06:21 2020 daemon.err glorytun[28959]: 51.178.141.35.65001: connected
> Mon Jun  8 18:06:21 2020 daemon.info glorytun[28959]: STARTED tun0
> Mon Jun  8 18:06:22 2020 daemon.err /usr/bin/ss-local[2879]: getpeername: Socket not connected
> Mon Jun  8 18:06:23 2020 user.notice omr-bypass: Starting OMR-ByPass...
> Mon Jun  8 18:06:23 2020 user.notice OMR-VPS: Get config from VPS...
> Mon Jun  8 18:06:23 2020 user.notice OMR-VPS: Glorytun restart...
> Mon Jun  8 18:06:23 2020 daemon.notice netifd: Network device 'tun0' link is down
> Mon Jun  8 18:06:23 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
> Mon Jun  8 18:06:23 2020 daemon.notice netifd: Interface 'omrvpn' is now down
> Mon Jun  8 18:06:23 2020 daemon.info glorytun[28959]: STOPPED tun0
> Mon Jun  8 18:06:23 2020 daemon.err glorytun: Key empty
> Mon Jun  8 18:06:23 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
> Mon Jun  8 18:06:23 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
> Mon Jun  8 18:06:23 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
> Mon Jun  8 18:06:23 2020 user.notice OMR-VPS: Restart glorytun...
> Mon Jun  8 18:06:23 2020 daemon.err glorytun: Key empty
> Mon Jun  8 18:06:24 2020 user.notice omr-bypass: Restart dnsmasq...
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[25810]: exiting on receipt of SIGTERM
> Mon Jun  8 18:06:24 2020 user.notice omr-bypass: OMR-ByPass is running
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: started, version 2.81 cachesize 150
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: DNS service limited to local subnets
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using only locally-known addresses for domain test
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using only locally-known addresses for domain onion
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using only locally-known addresses for domain localhost
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using only locally-known addresses for domain local
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using only locally-known addresses for domain invalid
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using only locally-known addresses for domain bind
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using only locally-known addresses for domain use-application-dns.net
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using nameserver 127.0.0.1#5353
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: using only locally-known addresses for domain lan
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: read /etc/hosts - 4 addresses
> Mon Jun  8 18:06:24 2020 daemon.info dnsmasq[30612]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
> Mon Jun  8 18:06:25 2020 daemon.err omr-tracker-server[11057]: curl: option --max-time: expected a proper numerical parameter
> Mon Jun  8 18:06:25 2020 daemon.err omr-tracker-server[11057]: curl: try 'curl --help' for more information
> Mon Jun  8 18:06:26 2020 user.notice Shadowsocks: Rules UP
> Mon Jun  8 18:06:26 2020 user.notice Shadowsocks: Reload omr-bypass rules
> Mon Jun  8 18:06:26 2020 user.notice omr-bypass: Starting OMR-ByPass...
> Mon Jun  8 18:06:26 2020 daemon.err /usr/bin/ss-redir[2880]: remote recv: Connection refused
> Mon Jun  8 18:06:26 2020 user.notice OMR-VPS: Can't get token, try later
> Mon Jun  8 18:06:26 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
> Mon Jun  8 18:06:26 2020 user.notice omr-bypass: Reload dnsmasq...
> Mon Jun  8 18:06:26 2020 daemon.err omr-tracker-server[11057]: curl: option --max-time: expected a proper numerical parameter
> Mon Jun  8 18:06:26 2020 daemon.err omr-tracker-server[11057]: curl: try 'curl --help' for more information
> Mon Jun  8 18:06:26 2020 daemon.info dnsmasq[30612]: read /etc/hosts - 4 addresses
> Mon Jun  8 18:06:26 2020 daemon.info dnsmasq[30612]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
> Mon Jun  8 18:06:26 2020 user.notice omr-bypass: OMR-ByPass is running
> Mon Jun  8 18:06:27 2020 daemon.err /usr/bin/ss-local[2879]: getpeername: Socket not connected
> Mon Jun  8 18:06:27 2020 daemon.info omr-tracker-ss: Shadowsocks is down (can't contact via http 1.1.1.1, 1.0.0.1, 212.27.48.10, 104.16.1.1, 103.224.182.242, 198.27.92.1, 88.191.250.176, 151.101.129.164, 198.11.132.250)
> Mon Jun  8 18:06:27 2020 user.notice Shadowsocks: Rules DOWN
> Mon Jun  8 18:06:27 2020 daemon.err omr-tracker-server[11057]: curl: option --max-time: expected a proper numerical parameter
> Mon Jun  8 18:06:27 2020 daemon.err omr-tracker-server[11057]: curl: try 'curl --help' for more information
> Mon Jun  8 18:06:28 2020 daemon.err omr-tracker-server[11057]: curl: option --max-time: expected a proper numerical parameter
> Mon Jun  8 18:06:28 2020 daemon.err omr-tracker-server[11057]: curl: try 'curl --help' for more information
> Mon Jun  8 18:06:28 2020 user.notice post-tracking-post-tracking: Get status and settings for wan3... Done

Ysurac, 2020-06-08:
@xabix This will be fixed in next release

xabix, 2020-06-08:
Hello Ysurac,

Why when I set only DL SQM on my ADSL in OMR Assistant, I get 0 in SQM QoS. Could there be a bug?

[image]

Merci

[45 more replies]

closed 2021-03-30 by Ysurac