Ysurac/openmptcprouter #960: Firewall - Block website/Parental Control

kevin39, 2020-04-07:
## Expected Behavior

Hi,
My goal is to block some website using firewall rules.
I'm unable to block anything !
Is something I missed ? There is no wiki to setup custom rule.

## Actual Behavior

No outoging traffic blocked.

## Steps to Reproduce the Problem

  1. Create a new rule to block LAN -> VPN or LAN -> WAN 
  2. Block a specific IP using this url
  3. Nothing blocked (ping and curl are working)
 
## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.54.7 r0+12509-04a21c26a0 / LuCI Master git-20.069.54281-3fb8e3d
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1015
  - OpenMPTCProuter platform:  RPI4

bnhf, 2020-04-07:
Are you trying ping and curl from a device on the LAN, or from the router?

Also, if you have Pi-hole implemented, you could add those websites to your blacklist and force DNS queries to Pi-hole -- not as good as a firewall REJECT, just an alternative.

kevin39, 2020-04-07:
I'm using ping/curl from my LAN (my computer).

I saw Pi-Hole can be an alternative but I'm not using it.

Ysurac, 2020-04-07:
In Network->DHCP & DNS you can set a domain you want to block in DNS forwardings

kevin39, 2020-04-08:
Thanks but there is no way to filter using firewall ? (time restriction is what I'm interested in case of parental control)

vinceducat, 2020-04-08:
you can try that https://bokunokeiken.wordpress.com/2015/06/27/how-to-block-device-on-openwrt-based-on-mac-address/

vinceducat, 2020-04-18:
> Thanks but there is no way to filter using firewall ? (time restriction is what I'm interested in case of parental control)

do you find what u want ?

kevin39, 2020-04-18:
No

vinceducat, 2020-04-18:
really ?!

vinceducat, 2020-04-18:
> you can even manage the time of in extra parameters for example adding this string
> -m time –kerneltz –weekdays Mo,Tu,We,Th,Fri –timestart 00:30 –timestop 6:00
> 
> (firewall will use kernel time zone, and from monday to friday it will block traffic from 0:30 am to 6 am)

vinceducat, 2020-04-18:
https://openwrt.org/docs/guide-user/firewall/fw3_configurations/fw3_parent_controls

closed 2020-09-14 by github-actions[bot]