Ysurac/openmptcprouter #961: Connection refused when making connection , destination sending [RST,ACK] packet on [SYN]

rajava2010, 2020-04-08:
## Expected Behavior

TCP connection is not established.

## Actual Behavior

I had configured the router in VirtualBox on OS X with 1 interface having USB tethered to iPhone. Configured OS X Wifi Interface to DHCP Server 192.168.100.1. Trying to make a connection it says connection refused, Wireshark shows destination sending [RST,ACK] packet on [SYN]. name lookup works fine. 

Status page show ShadowSocks is NOT running.

## Steps to Reproduce the Problem

  1 .curl -v https://www.openmptcprouter.com 
  2.*connect to  94.23.252.192 port 443: Connection refused
  3.

## Specifications

  OpenMPTCProuter version: openmptcprouter-v0.54.7-r012509-04a21c26a0-x86-64-combined-ext4-efi.vdi
OpenMPTCProuter VPS version: 0.1013
OpenMPTCProuter platform: x86_64

Ysurac, 2020-04-08:
VPS script should be updated.
Did you set IP and key of the VPS API in the wizard ?

rajava2010, 2020-04-08:
Pardon my ignorance, which IP I need to set and how do I get the key for VPS API ?

Ysurac, 2020-04-08:
The VPS IP and the keys you get when you installed the VPS script (also available in /root/openmptcprouter_config.txt).

rajava2010, 2020-04-08:
Can the VPS be installed on the same Gues VirtualBox VM where the router is running or a separate VM is needed ? I am confused , this is what I am trying to do , I have Macbook and I have 2 USB tethered 4G iPhones which I need to aggregate and run by Corporate VPN and access the internet so that the aggregate bandwidth is enough for me to do faster uploads/downloads and access ? Please Help, Pardon me as I Am not very technical but I need to achieve this to keep myself working due to Covid-19 outbreak.

Ysurac, 2020-04-08:
The VPS need to be in a datacenter or somewhere where it can have more than the aggregated bandwidth to Internet.

rajava2010, 2020-04-09:
I was able to get a VPS through Google Cloud and it seems to work. But when I connect my office VPN, it seems to utilize only one interface, Is it working as expected ? Is there anyway I can make the VPN use both interface for better throughput ?

closed 2020-09-14 by github-actions[bot]