Ysurac/openmptcprouter #962: Not an issue but a Thank you

iditude, 2020-04-08:
I just want to thank you for the great work. Being confined at home due to COVID-19 is challenging enough, and with slow internet it makes working remotely a nightmare. 

After many tweaks and waiting for postal service to deliver new SIM cards, my family is finally able to surf with crazy speeds thanks to this project. 

Cheers for the great work.

rajava2010, 2020-04-09:
Where did you host your VPS ?

iditude, 2020-04-09:
OVH has good VPS. Overall works well except for Entertainment such as Netflix where you need OMR bypass.

vinceducat, 2020-04-09:
milkywan ;-)

rajava2010, 2020-04-13:
Does your Office VPN like Cisco AnyConnect works and load balances ? it doesn't seem to work for me, the VPN tunnel is routed only through one interface.

vinceducat, 2020-04-13:
> 
> 
> Does your Office VPN like Cisco AnyConnect works and load balances ? it doesn't seem to work for me, the VPN tunnel is routed only through one interface.

yes it load balance, for me adsl and 4g link, ... this job is done by OMR for me on a raspberry pi 3

Ysurac, 2020-04-14:
There is now load balancing support in OpenMPTCProuter for now, but it aggregate speed when needed :)

rajava2010, 2020-04-14:
there is "NO" or "NOW" load balancing support ? because my VPN tunnel gets routed through only one interface even there is demand for speed when mutliple downloads are happening ?

Ysurac, 2020-04-14:
It's a no, or a not now. This will be available in next release when VPS is down. And please don't hijack issue.

user747, 2020-04-15:
also please check out the donation links on the main page and consider a financial contribution. 
appreciate the frequent (sometimes hourly!) commits and fixes, especially upstream fixes. keep up the good work.

talalmash, 2020-04-18:
A massive thank you from Lebanon (a famous country for internet issues), we are heavily relying on fail over system as ADSL, microwave and 3g intermittently disconnect several times a day!

vinceducat, 2020-04-18:
@talalmash, i m curious what is microwave tech? 

n

talalmash, 2020-04-22:
We need a donate emblem for blogs and websites, there are two logo design, which will be the default? (blue back, vs white)

Ysurac, 2020-04-22:
which logo ?

talalmash, 2020-04-23:
> which logo ?

Router admin page has blue background inside the icon. I guess white is default although the blue  background looks nicer to me. I'm making a banner.

Ysurac, 2020-04-23:
Yes it's better with the blue background. The default is transparent background.

closed 2020-09-14 by github-actions[bot]