Ysurac/openmptcprouter #964: Désinstallation de Pi-Hole

mathchab, 2020-04-13:
## Expected Behavior
Supprimer Pi-Hole.

## Current Behavior
Plus d'accès internet.
GloryTun s'arrête.

"DNS issue"
"No output"

## Possible Solution
Script pour désinstaller Pi-Hole et reset OpenMPTCPRouter

## Steps to Reproduce the Problem

  1. Connexion SSH au VPS
  2. Sudo pihole uninstall
  3. Reset manuellement des DNS
  4. Quand on rapplique l'assistant OpenMPTCPRouter, le DNS redeviens l'IP de Pi-Hole
## Specifications

  - OpenMPTCProuter version: 54.7
  - OpenMPTCProuter VPS version: 1015.4.19104
  - OpenMPTCProuter platform: Pi3+

Ysurac, 2020-04-13:
Pour vérifier que Pi-Hole est installé sur le serveur, l'API regarde si le fichier ```/etc/pihole/setupVars.conf``` existe. Je ne sais pas ce que fait "pihole uninstall", j'ai jamais essayé.

mathchab, 2020-04-13:
Du coup, pas de procédure pour revenir en arrière et supprimer Pi-Hôle pour l'instant ?
À l'instant T, obligé de réinstaller Pi-Hôle pour ne pas tout cassé.
J'ai vu votre message trop tard.

Ysurac, 2020-04-13:
Si donc, en supprimant le fichier ```/etc/pihole/setupVars.conf``` si il est bien toujours présent après l'uninstall de pihole qui serait incomplet donc.

notloged, 2020-04-15:
Pour moi pihole uninstall a bien fonctionné

closed 2020-09-14 by github-actions[bot]