Ysurac/openmptcprouter #965: WAN1 active but not visible in Interface

bernic, 2020-04-13:
## Expected Behavior
under system/OpenMTCPRouter
both under wizard and state Wan1 interface should be visible with Wan2

## Current Behavior
Only WAN2 is visible
Which means setting for Wan1 cannot be set under the wizard

## Steps to Reproduce the Problem
Not sure it will help because I am encountering a lot of problem with my modem
My setting:
1 SAGEMCOM modem/router (=WAN1)
1 NETGEAR LTE modem LB2120 (=WAN2)
1 Atom nuc running OMR
1 netgear 3500WRL running advance tomato

All above network appliances are connected via ethernet to the SAGEMCOM modem

All connected devices of the house (like computers,NAS...) are connected through the Netgear 3500WRL

It seems that disabling DHCP on the SAGEMCOM modem is not stable and its connection is quickly lost

So I kept the DHCP enabled on it and disabled it on the OMR

the netgear 3500WRL router is on the same addressing as OMR so it uses OMR to conect to internet
If I physically disconnect the LTE modem
and reboot OMR than WAN1 disappear from the interface
(though it is still visible under /Network/interfaces)
and it is working otherwise I wouldn't be writting this ticket

## Context (Environment)

Affects me in understanding what goes wrong with the DHCP 
but I am far from being an expert. So I just thought that it would be useful to report.

## Specifications

  - OpenMPTCProuter version:  v0.54.7 
  - OpenMPTCProuter VPS version: 0.1015 4.19.54-mptcp
  - OpenMPTCProuter platform: <x86>

Ysurac, 2020-04-13:
Only interfaces with MPTCP enabled are visible in wizard & status page.
You need to enable MPTCP on the interface in Network->MPTCP.

bernic, 2020-04-13:
Yes it was active

bernic, 2020-04-13:
Well I think because I remember quickly checking
though I cannot be 100% affirmative
I have sloved it by reinstalling - I know its a bit  radical, but efficient

bernic, 2020-04-13:
It must have been active for Wan1
because I remember (I was writing this ticket) I log on my VSP account (not on the VSP - but on the managing account) and the provider did not like that I log on my account from the VSP Ip and asked me double authentication for this reason.

closed 2020-09-14 by github-actions[bot]