Ysurac/openmptcprouter #966: Wan1 is reported as DOWN but isn't

bernic, 2020-04-13:
## Expected Behavior
Report the real state of the interface

## Current Behavior
Wan1 is always reported as down after a few minutes of functioning (restart router and everything is reported as working for a few minutes)
This is reported in System/OpenMtcpRouter/ State(tab)
But you can tell it is not down as if you start downloading something on internet you can see a lot of traffic going through it in State/Real time traffic

## Possible Solution

## Steps to Reproduce the Problem

 I suppose it is linked to my devices especially my sagemcom modem
since there is no issue with my LTE modem:

My setting:
1 SAGEMCOM modem/router (=WAN1)
1 NETGEAR LTE modem LB2120 (=WAN2)
1 Atom nuc running OMR
1 netgear 3500WRL running advance tomato

All above network appliances are connected via ethernet to the SAGEMCOM modem

All connected devices of the house (like computers,NAS...) are connected through the Netgear 3500WRL

It would seem that SAGEMCOM modem as some reserved address : it cannot adress anything between 2 and 10 (168.192.0.10 is the first ip that can be given by its dhcp) but for instance it has a dlna service that is linked to 168.192.0.5

So I made sure that OMR is using 168.192.0.11 on Wan1
it seems that it solved a few issues (but I am not sure yet: everytime I though it was working it broke after a certain time)

## Context (Environment)

Slightly annoying because I am never sure I am running on my 4G connection alone because the Wan1 is effectively down or not.

## Specifications

OpenMPTCProuter version: v0.54.7
OpenMPTCProuter VPS version: 0.1015 4.19.54-mptcp
OpenMPTCProuter platform: x86

Ysurac, 2020-04-14:
To check state of an interface by default ICMP Ping is used. Check that ICMP is not blocked somewhere or use another method in omr-tracker for this interface.
You can also check in State->System log why the interface is show as down.

bernic, 2020-04-14:
Ok I will try to set another method
The strange thing with my modem is that at first it respond to ping , but after a while it stops.
I deactivated the firewall , and the portscan prevention, but it did not change anything.
I need to recheck but it seems that it respond to ping again if I turn OMR down and back up. So it doesn't look like a banning feature of some sort.

Is there a prefered way to send you the error log. I have plenty of error indeed but I am not sure if it refers to my issue

Ysurac, 2020-04-14:
On pastebin or paste it here.

bernic, 2020-04-14:
ok there are some strange results

[cid:a97e2da8-d2ec-4f23-bd13-6f4f66067d2b]
as you can see wan 1 uses a lot of bandwidth
but the log says

Tue Apr 14 17:03:15 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Tue Apr 14 17:03:30 2020 user.notice post-tracking-post-tracking: wan1 switched off because gateway down and ping from 192.168.0.11 error (4.2.2.1,8.8.8.8,51.25...)

here is the log in more details (most external ip are anonymized by regexp):

Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty21, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty21
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty21, subsystem=tty
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty5, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty5
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty5, subsystem=tty
Tue Apr 14 07:42:49 2020 daemon.err usbmuxd[10775]: [07:42:49.395][3] usbmuxd v1.1.1 starting up
Tue Apr 14 07:42:49 2020 user.notice post-tracking-post-tracking: omrvpn down
Tue Apr 14 07:42:49 2020 daemon.err usbmuxd[10775]: [07:42:49.409][3] Initialization complete
Tue Apr 14 07:42:49 2020 daemon.err usbmuxd[10775]: [07:42:49.409][3] Enabled exit on SIGUSR1 if no devices are attached. Start a new instance with "--exit" to trigger.
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:49 2020 user.notice SQM: Starting SQM script: simple.qos on tun0, in: 0 Kbps, out: 0 Kbps
Tue Apr 14 07:42:49 2020 user.notice SQM: Using generic sqm_start_default function.
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: add serial interface ttyI54: event processed
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI54, subsystem=tty
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty11, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty11
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty11, subsystem=tty
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty58, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty58
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty58, subsystem=tty
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: add serial interface ttyI55: event processed
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI55, subsystem=tty
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty3, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty3
Tue Apr 14 07:42:49 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty3, subsystem=tty
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty48, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty48
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: add serial interface ttyI56: event processed
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty48, subsystem=tty
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI56, subsystem=tty
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=ttyS8, subsystem=tty, sysfspath=/sys/devices/platform/serial8250/tty/ttyS8
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyS8, subsystem=tty
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: add serial interface ttyI57: event processed
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: event reported: action=add, name=ttyI57, subsystem=tty
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty38, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty38
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty38, subsystem=tty
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 07:42:50 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
[... cut ...]
Tue Apr 14 15:00:35 2020 daemon.err /usr/bin/ss-redir[17685]: remote recv: Connection reset by peer
Tue Apr 14 15:00:37 2020 user.notice post-tracking-post-tracking: wan1 switched off because gateway down and ping from 192.168.0.11 error (8.8.8.8,4.2.2.1,8.8.8.8,4.2.2.1)
Tue Apr 14 15:00:37 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev wan1
Tue Apr 14 15:01:08 2020 daemon.err /usr/bin/ss-redir[17685]: remote recv: Connection reset by peer
Tue Apr 14 15:04:4

Ysurac, 2020-04-14:
As log said, gateway of wan1 doesn't answer to ping and not possible to ping via wan1 the only 2 IPs available in omr-tracker (why only 2 ips ?).

bernic, 2020-04-14:
I first left the default setting but then added my VSP I did not copy past the line were the ip of my VSP appears.
the result was the same not possible to ping
However I do not understand then why wan 1 has routed 1.5Gb of data vs 200 for wan2

bernic, 2020-04-14:
I tried to ping wan 1 (192.168.0.1) from my netgear router -> no answer
I restarted OMR and tried again -> answer ok

but at the same time in the log I get that
Tue Apr 14 18:10:42 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.0.1 dev wan1
Tue Apr 14 18:10:44 2020 user.notice post-tracking-post-tracking: wan1 switched off because gateway down and ping from 192.168.0.11 error (4.2.2.1,8.8.8.8,4.2.2.1,8.8.8.8)
Tue Apr 14 18:10:44 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.0.1 dev wan1

I tried to ping from my router and no issue (yet - I guess that at some point it will not get any answer as before I restarted OMR)

bernic, 2020-04-14:
also note that my router is on the OMR network (with its wan ip set on 192.168.100.12)

bernic, 2020-04-14:
I am also easily accessing http://192.168.5.1 the LTE router interface but never the wire router interface http://192.168.0.1<http://192.168.5.1>

Ysurac, 2020-04-14:
I understand nothing to your network now...

bernic, 2020-04-14:
I don't know if it makes sens:

[cid:03d1c8b8-fa0f-4172-bb0f-2c548f3ac0dd]

bernic, 2020-04-14:
though I am not sure the lte router get this 192.168.0.16 adress
I just happen to see it on the sagemcom modem interface as there is only 3 devices listed on it:
-the netgear router on which I temporarily ste the wan ip to 192.168.0.12 to access the sagemcom interface (otherwise no access)
-OMR
and I can see a 3rd on with this ip 192.168.0.16 I supposed it was the LTE router though I never did set this adress my self and in the router the only ip that I set is 192.168.5.1
So I am a bit surprised I will double check with the mac adress

Ysurac, 2020-04-15:
I think the schema is missing.

bernic, 2020-04-15:
Here is my network schematic (hope you can see it):

Mean while I rebooted OMR but did not turn on the 4G modem router
and since then the SAGEMCOM MODEM ROUTER has been online all along with no issue, I can even access its interface to change the settings
So I start to believe there is an issue when both Modems are on.
when both are on I have access to both for only a short time after which I can only access the 4G LTE modem.
Pinging to or through SAGEMCOM seems not to work
but the interface also reports more data going through SAGEMCOM modem than 4G modem

At the same time if I change the Wan of my tomato router to 192.168.0.1 to be able to reach the sagemcom modem I notice that the dhcp server is leasing 3 ip:
192.168.0.11 (OMR)
192.168.0.12 (tomato router)
192.168.0.16 ??? only the 4G LTE is plugged on top of OMR and tomato router, but this adress could be given to a service of SAGEMCOM (like dlna server,... though I deactivated all the services I found)

[cid:34a26617-c623-43eb-8c90-3b1330ba9c73]

bnhf, 2020-04-15:
> My setting:
> 1 SAGEMCOM modem/router (=WAN1)
> 1 NETGEAR LTE modem LB2120 (=WAN2)
> 1 Atom nuc running OMR
> 1 netgear 3500WRL running advance tomato
>
> All above network appliances are connected via [...]

I'll suggest that some, if not all, of your issues are related to the way you're making your connections. Given that OMR is a router too, you need to separate your LAN and WAN traffic, either physically (using separate adapters), or virtually (using VLANs or Macvlan).

If your OMR router has sufficient Ethernet ports, you should make your connections like so:

[image]

If your OMR router doesn't have sufficient Ethernet ports then you could use an unmanaged switch to physically connect everything, but then you'll need to setup VLANs or Macvlan to keep the traffic sorted. Trying to use the SAGEMCOM router as that switch is likely to cause issues.

bernic, 2020-04-15:
Yes I thought so but I have no spare switch... and only one physical interface on the OMR

If I can get a switch, how should I set MacVlan, from what I saw they are already 2 macvlan attributed by ORM for each of the wan interface aren't they?

is there some doc you would recommend me to make this setup?

anyway a big thank you for your help

[bernic network]<https://user-images.githubusercontent.com/41088895/79355561-aeea5680-7ef2-11ea-9eea-e05822994eab.png>

bnhf, 2020-04-15:
> If I can get a switch, how should I set MacVlan, from what I saw they are already 2 macvlan attributed by ORM for each of the wan interface aren't they?

Macvlan allows you to create virtual adapters that share physical interfaces. So you could have virtual adapters named wan1 and wan2 (not to be confused with the network interfaces of the same name) each assigned to physical adapter eth0. Theoretically, you could also have a virtual adapter called lan assign to that same eth0, though I haven't tried it.

Under Network - Interfaces - LAN, then change the physical adapter assignment to the virtual adapter named lan. Re-run the OMR Wizard, and change the interface assignment of WAN1 to wan1, and WAN2 to wan2. Check that the IP addresses of the gateway routers are correct, and assign a static address to each in that gateway's same subnet. I think you'll find you can now turn off DHCP on the SAGEMCOM and LTE routers without the stability issues you reported previously.

bnhf, 2020-04-16:
> Theoretically, you could also have a virtual adapter called lan assign to that same eth0, though I haven't tried it.

Just wanted to expand on this a little bit. When I've used Macvlan, I've always had two adapters, one for LAN and one for all WANs. As I said in my previous post, I believe you can make it work with a single Etherent port too using Macvlan -- but maybe @Ysurac or somebody that has such a configuration in use can confirm this.

Ysurac, 2020-04-16:
Yes, macvlan can be used on a single ethernet port (like on RPI).

bernic, 2020-04-17:
HI
thank you for the answer
What I meant is I think everything is already correctly set and was wondering if I missed something, but from your answer I think I did not miss anything.
I have a lan and 2 wan interfaces already set If I con under MacVlan menu I can see that eth0 is assigned wan1 and wan2
I can see I have 3 different MAc adress for my one and only ethernet port: the lan mac adress and the 2 macvlan adresses.
As for the Dhcp servers I had already deactivated all dhcp server except on the tomato router and SAGEMCOM modem/router
I have further deactivated the SAGEMCOM dhcp server - no more dhcp except for the tomato router

So anyway I found a switch and plugged every thing on it :
The neatgear tomato router
The SAGEMCOM modem router
The 4G LTE modem router
the atom nuc OMR

and I could see and access all these devices from any computer behind the tomato router !!! Great... Except that
The SAGEMCOM is still reported as down from OMR interface and in the SAGEMCOM interface the wan interface is reported as ip 0.0.0.0  with 255.0.0.0 as subnet masq
Very stange because I can use the phone plugged on the modem without issue, so it must some how be connected.
I suppose there are some Vlan behind it that mangle things up.

I made a test: I restarted SAGEMCOM modem : no difference
I turned off the 4G modem and restarted the SAGEMCOM modem -> no more issue with SAGEMCOM

I turned the 4G modem back on -> back to previous state no more ping no more access to SAGEMCOM

But now I looked at the routing table and I think that Something is definitively going wrong
[cid:image002.png@01D614B9.84CBC3A0]
(The 198.168.1.1 is now the 4G router adress)

bernic, 2020-04-17:
I have these warning  in the SAGEMCOM modem at same time

16/04/2020 à 23:14:54
 DHCP WARNING - Non-critical field invalid in response ;CM-MAC=f4:6b:ef:5a:66:f0;CMTS-MAC=00:01:5c:87:59:a5;CM-QOS=1.1;CM-VER=...
 17/04/2020 à 12:18:11
 SYNC Timing Synchronization failure - Loss of Sync;CM-MAC=f4:6b:ef:5a:66:f0;CMTS-MAC=00:01:5c:87:59:a5;CM-QOS=1.1;CM-VER=3.0;
Even when removing my tomato router and Following the doc exactly I get same result
Wan1 is lost
The doc suppose we run OMR on a Raspberry Pi so I suppose this means it only has one physical port unless you add a shield.
So I suppose I should be in the same configuration then.

If I go in the OMR interface and restart wan1 interface I can reconnect to wan1 as normal but it only last a few seconds

Provenance : Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows 10

De : Eric Bayard<mailto:bayarder@hotmail.com>
Envoyé le :vendredi 17 avril 2020 13:10
À : Ysurac/openmptcprouter<mailto:reply@reply.github.com>
Objet :RE: [Ysurac/openmptcprouter] Wan1 is reported as DOWN but isn't (#966)

HI
thank you for the answer
What I meant is I think everything is already correctly set and was wondering if I missed something, but from your answer I think I did not miss anything.
I have a lan and 2 wan interfaces already set If I con under MacVlan menu I can see that eth0 is assigned wan1 and wan2
I can see I have 3 different MAc adress for my one and only ethernet port: the lan mac adress and the 2 macvlan adresses.
As for the Dhcp servers I had already deactivated all dhcp server except on the tomato router and SAGEMCOM modem/router
I have further deactivated the SAGEMCOM dhcp server - no more dhcp except for the tomato router

So anyway I found a switch and plugged every thing on it :
The neatgear tomato router
The SAGEMCOM modem router
The 4G LTE modem router
the atom nuc OMR

and I could see and access all these devices from any computer behind the tomato router !!! Great... Except that
The SAGEMCOM is still reported as down from OMR interface and in the SAGEMCOM interface the wan interface is reported as ip 0.0.0.0  with 255.0.0.0 as subnet masq
Very stange because I can use the phone plugged on the modem without issue, so it must some how be connected.
I suppose there are some Vlan behind it that mangle things up.

I made a test: I restarted SAGEMCOM modem : no difference
I turned off the 4G modem and restarted the SAGEMCOM modem -> no more issue with SAGEMCOM

I turned the 4G modem back on -> back to previous state no more ping no more access to SAGEMCOM

But now I looked at the routing table and I think that Something is definitively going wrong
[cid:image005.png@01D614C1.B1C8D160]
(The 198.168.1.1 is now the 4G router adress)

bernic, 2020-04-17:
So now I found a usb hub with a ethernet plug
I plugged it in  OMR and it was recognized straight away as eth1
So I set lan on eth0
And wan1 and wan2 on eth1

My network now looks like :
[cid:image003.png@01D614C7.6E1B5220]

But exact same issue
Wan1 unreachable… well if I restart it, it is reachable for a few second

Provenance : Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows 10

Envoyé le :jeudi 16 avril 2020 16:24
À : Ysurac/openmptcprouter<mailto:openmptcprouter@noreply.github.com>
Cc : bernic<mailto:bayarder@hotmail.com>; Author<mailto:author@noreply.github.com>

bernic, 2020-04-17:
Ok I think I found a solution
I separated both wan on different physical interfaces

Now my network is

Eth0 lan and wan2 plugged on the switch
Eth1 wan1 dirctly pluggedon SAGEMCOM

Both are now reported up and running

I just have a warning that vpn tunel is down on OMR what would that mean/ Imply ?

Thank you

Provenance : Courrier<https://go.microsoft.com/fwlink/?LinkId=550986> pour Windows 10

Ysurac, 2020-04-17:
Check that time on VPS and router part are the same.

closed 2020-09-14 by github-actions[bot]