Ysurac/openmptcprouter #967: Error: MPTCP subsocket connect failed,error -19

Franckrst, 2020-04-14:
## Expected Behavior
I want to have clean installation

## Current Behavior
I have this error in my log evrytime:
```
[ 2216.808773] __mptcp_init4_subsockets: token 0x9c0c2cbe bind() to 192.168.20.2 index 516 failed, error -99
```
Lot of:
```
[ 2216.900712] __mptcp_init4_subsockets: MPTCP subsocket connect() failed, error -19
```

## Context (Environment)

- disable all DHCP and set new IP. 
- Install debian on my OVH VPS
- Intall script "openmptcprouter-vps"
- Install from image OpenMPTCPRouter on RPI 3.
- Add configuration
- Add 2 * 4G connexion (operator: orange & bouyguetelecom)
- Add 2 * Adsl connexion (SFR & Orange)
- Install https://github.com/Ysurac/openmptcprouter/wiki/Pi-hole on my VPS
- Reboot all

[Current configuration](https://i.imgur.com/RhhhdUZ.png)

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: v0.1015
  - OpenMPTCProuter platform: try on RPI3 AND x86_64 -->

Ysurac, 2020-04-14:
It can't use the interface with IP 192.168.20.2. It's an USB dongle ? an ethernet router ? a phone ?

Franckrst, 2020-04-14:
It's an ethernet router ( it's same router for 192.168.10.1 ).
I set my computer ip static on 192.168.20.X and i ping with succes 192.168.20.1 & 192.168.20.2

My systéme log : 
Tue Apr 14 14:56:25 2020 user.notice firewall: Reloading firewall due to ifup of wan1 (wan1)
Tue Apr 14 14:56:26 2020 daemon.notice netifd: Interface 'wan2' is now down
Tue Apr 14 14:56:26 2020 kern.err kernel: [17506.363215] __mptcp_init4_subsockets: MPTCP subsocket connect() failed, error -19
Tue Apr 14 14:56:26 2020 kern.err kernel: [17506.364359] __mptcp_init4_subsockets: token 0xb266793 bind() to 192.168.20.2 index 3524 failed, error -99
Tue Apr 14 14:56:26 2020 user.notice ModemManager: hotplug: remove network interface wan2: event processed
Tue Apr 14 14:56:26 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=wan2, subsystem=net
Tue Apr 14 14:56:26 2020 daemon.notice netifd: Interface 'wan2' is disabled
Tue Apr 14 14:56:26 2020 daemon.notice netifd: macvlan 'wan2' link is down
Tue Apr 14 14:56:26 2020 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Tue Apr 14 14:56:26 2020 daemon.notice netifd: Interface 'wan2' is enabled
Tue Apr 14 14:56:26 2020 daemon.notice netifd: Interface 'wan2' is setting up now
Tue Apr 14 14:56:26 2020 daemon.notice netifd: Interface 'wan2' is now up
Tue Apr 14 14:56:26 2020 daemon.notice netifd: macvlan 'wan2' link is up
Tue Apr 14 14:56:26 2020 daemon.notice netifd: Interface 'wan2' has link connectivity
Tue Apr 14 14:56:26 2020 user.notice ModemManager: hotplug: add network interface wan2: event processed
Tue Apr 14 14:56:26 2020 user.notice ModemManager: hotplug: event reported: action=add, name=wan2, subsystem=net
Tue Apr 14 14:56:26 2020 kern.err kernel: [17506.458010] __mptcp_init4_subsockets: MPTCP subsocket connect() failed, error -19
Tue Apr 14 14:56:26 2020 kern.err kernel: [17506.458828] __mptcp_init4_subsockets: MPTCP subsocket connect() failed, error -19
Tue Apr 14 14:56:26 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Apr 14 14:56:26 2020 kern.err kernel: [17506.547278] __mptcp_init4_subsockets: MPTCP subsocket connect() failed, error -19
Tue Apr 14 14:56:26 2020 kern.err kernel: [17506.548463] __mptcp_init4_subsockets: MPTCP subsocket connect() failed, error -19
Tue Apr 14 14:56:26 2020 daemon.err omr-tracker[1351]: ping: sendmsg: No such device
Tue Apr 14 14:56:26 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan1 (wan1)
Tue Apr 14 14:56:26 2020 user.notice firewall: Reloading firewall due to ifup of wan2 (wan2)
Tue Apr 14 14:56:27 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (wan2)
Tue Apr 14 14:56:27 2020 user.notice post-tracking-post-tracking: Set server vps (51.38.37.112) default route via 192.168.10.1
Tue Apr 14 14:56:27 2020 user.notice post-tracking-post-tracking: Set server vps (51.38.37.112) route via 192.168.10.1 metric 3
Tue Apr 14 14:56:29 2020 user.notice omr-bypass: Starting OMR-ByPass...
Tue Apr 14 14:56:30 2020 daemon.info vnstatd[22507]: Info: Interface "wan1" enabled.
Tue Apr 14 14:56:31 2020 user.notice omr-bypass: Restart dnsmasq...
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[30552]: exiting on receipt of SIGTERM
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: started, version 2.81rc3 cachesize 150
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: DNS service limited to local subnets
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq-dhcp[937]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: using only locally-known addresses for domain test
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: using only locally-known addresses for domain onion
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: using only locally-known addresses for domain localhost
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: using only locally-known addresses for domain local
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: using only locally-known addresses for domain invalid
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: using only locally-known addresses for domain bind
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: using nameserver 10.255.255.1#53
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: using only locally-known addresses for domain lan
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: read /etc/hosts - 4 addresses
Tue Apr 14 14:56:31 2020 user.notice omr-bypass: OMR-ByPass is running
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq[937]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Tue Apr 14 14:56:31 2020 daemon.info dnsmasq-dhcp[937]: read /etc/ethers - 0 addresses
Tue Apr 14 14:56:32 2020 user.notice post-tracking-post-tracking: Set server vps (51.38.37.112) route via 192.168.20.1 metric 6
Tue Apr 14 14:56:34 2020 daemon.info dnsmasq-dhcp[937]: DHCPDISCOVER(eth0) f0:03:8c:b3:44:e6
Tue Apr 14 14:56:34 2020 daemon.info dnsmasq-dhcp[937]: DHCPOFFER(eth0) 192.168.100.163 f0:03:8c:b3:44:e6
Tue Apr 14 14:56:34 2020 daemon.info dnsmasq-dhcp[937]: DHCPREQUEST(eth0) 192.168.100.163 f0:03:8c:b3:44:e6
Tue Apr 14 14:56:34 2020 daemon.info dnsmasq-dhcp[937]: DHCPACK(eth0) 192.168.100.163 f0:03:8c:b3:44:e6 Roomba-3142841C00632710
[... cut ...]
Tue Apr 14 14:56:36 2020 daemon.notice netifd: macvlan 'wan2' link is down
Tue Apr 14 14:56:36 2020 daemon.notice netifd: Interface 'wan2' has link connectivity loss
Tue Apr 14 14:56:36 2020 daemon.notice netifd: Interface 'wan2' is enabled
Tue Apr 14 14:56:36 2020 daemon.notice netifd: Interface 'wan2' is setting up now
Tue Apr 14 14:56:36 2020 daemon.notice netifd: Interface 'wan2' is now up
Tue Apr 14 14:56:36 2020 daemon.notice netifd: macvlan 'wan2' link is up
Tue Apr 14 14:56:36 2020 daemon.notice netifd: Interface 'wan2' has link connectivity

Ysurac, 2020-04-14:
Do you use omr-quota ?
Else the wan2 interface is detected as down. I would need the full config (without keys) in System->OpenMPTCProuter and full parameters tab.

Franckrst, 2020-04-14:
Yes i have omr-quota : https://prnt.sc/rz7jen
But i have use 4Go and 3.8Go
I deactivate it to test and all the errors disappeared.
So my quote value is bad ?

my config : 
```
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].rebind_protection='0'
dhcp.@dnsmasq[0].server='10.255.255.1#53'
dhcp.@dnsmasq[0].ipset='/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
[... 1404 lines cut ...]
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2' 'wan3' 'wan4'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='11'
wireless.radio0.hwmode='11g'
wireless.radio0.path='platform/soc/3f300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='HT20'
wireless.radio0.disabled='1'
wireless.radio1=wifi-device
wireless.radio1.type='mac80211'
wireless.radio1.channel='36'
wireless.radio1.hwmode='11a'
wireless.radio1.path='pci0000:00/0000:00:14.0/0000:02:00.0'
wireless.radio1.htmode='VHT80'
wireless.radio1.disabled='1'
wireless.default_radio1=wifi-iface
wireless.default_radio1.device='radio1'
wireless.default_radio1.network='lan'
wireless.default_radio1.mode='ap'
wireless.default_radio1.ssid='OpenWrt'
wireless.default_radio1.encryption='none'
```

Ysurac, 2020-04-14:
Can you give me the result of ```vnstat -i wan2 --json``` via SSH on the router ?

Franckrst, 2020-04-14:
```{"vnstatversion":"2.4","jsonversion":"2","interfaces":[{"name":"wan2","alias":"","created":{"date":{"year":2020,"month":4,"day":14}},"updated":{"date":{"year":2020,"month":4,"day":14},"time":{"hour":16,"minute":0}},"traffic":{"total":{"rx":162661127,"tx":89825622},"fiveminute":[{"id":4,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":5},"rx":0,"tx":0},{"id":3,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":10},"rx":0,"tx":0},{"id":16,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":15},"rx":0,"tx":0},{"id":23,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":20},"rx":0,"tx":0},{"id":30,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":25},"rx":0,"tx":0},{"id":37,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":30},"rx":0,"tx":0},{"id":44,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":35},"rx":0,"tx":0},{"id":57,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":40},"rx":3380,"tx":6536},{"id":56,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":45},"rx":1440,"tx":8162},{"id":71,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":50},"rx":7518,"tx":27904},{"id":80,"date":{"year":2020,"month":4,"day":14},"time":{"hour":8,"minute":55},"rx":5088,"tx":27086},{"id":89,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":0},"rx":25722,"tx":13080},{"id":98,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":5},"rx":26295,"tx":13966},{"id":107,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":10},"rx":36512,"tx":11386},{"id":116,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":15},"rx":28806,"tx":16784},{"id":125,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":20},"rx":3724166,"tx":71628},{"id":138,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":25},"rx":4192,"tx":10820},{"id":137,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":30},"rx":78272,"tx":17280},{"id":157,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":35},"rx":10148,"tx":8794},{"id":161,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":40},"rx":8077049,"tx":6595092},{"id":175,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":45},"rx":350206,"tx":38034},{"id":179,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":50},"rx":13599550,"tx":3147193},{"id":188,"date":{"year":2020,"month":4,"day":14},"time":{"hour":9,"minute":55},"rx":16433,"tx":9751},{"id":197,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":0},"rx":55533,"tx":70924},{"id":206,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":5},"rx":28291909,"tx":707254},{"id":215,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":10},"rx":1552747,"tx":64254},{"id":224,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":15},"rx":6388,"tx":9190},{"id":233,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":20},"rx":6470,"tx":6448},{"id":242,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":25},"rx":114942,"tx":36256},{"id":251,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":30},"rx":85067,"tx":6246},{"id":260,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":35},"rx":666844,"tx":32458},{"id":269,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":40},"rx":1895514,"tx":59702},{"id":278,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":45},"rx":3043401,"tx":82794},{"id":287,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":50},"rx":1595313,"tx":12908676},{"id":296,"date":{"year":2020,"month":4,"day":14},"time":{"hour":10,"minute":55},"rx":13065,"tx":8732},{"id":305,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":0},"rx":47418,"tx":184},{"id":314,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":5},"rx":108659,"tx":8346},{"id":323,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":10},"rx":49262,"tx":9048},{"id":332,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":15},"rx":47203,"tx":25190},{"id":341,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":20},"rx":33671,"tx":808},{"id":350,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":25},"rx":35147,"tx":20243},{"id":359,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":30},"rx":24330,"tx":27082},{"id":368,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":35},"rx":18322,"tx":12202},{"id":377,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":40},"rx":16367919,"tx":263166},{"id":386,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":45},"rx":31006,"tx":40305},{"id":395,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":50},"rx":2085667,"tx":66910},{"id":404,"date":{"year":2020,"month":4,"day":14},"time":{"hour":11,"minute":55},"rx":71708,"tx":35107},{"id":413,"date":{"year":2020,"month":4,"day":14},"time":{"hour":12,"minute":0},"rx":108980,"tx":9356},{"id":422,"date":{"year":2020,"month":4,"day":14},"time":{"hour":12,"minute":5},"rx":24846,"tx":20534},{"id":431,"date":{"year":2020,"month":4,"day":14},"time":{"hour":12,"minute":10},"rx":5708,"tx":5422},{"id":440,"date":{"year":2020,"month":4,"day":14},"time":{"hour":12,"minute":15},"rx":18772,"tx":17350},{"id [... cut ...] "day":14},"time":{"hour":14,"minute":0},"rx":718030,"tx":391928},{"id":67,"date":{"year":2020,"month":4,"day":14},"time":{"hour":15,"minute":0},"rx":78093619,"tx":64437604}],"day":[{"id":2,"date":{"year":2020,"month":4,"day":14},"rx":162661127,"tx":89825622}],"month":[{"id":2,"date":{"year":2020,"month":4},"rx":162661127,"tx":89825622}],"year":[{"id":2,"date":{"year":2020},"rx":162661127,"tx":89825622}],"top":[{"id":2,"date":{"year":2020,"month":4,"day":14},"rx":162661127,"tx":89825622}]}}]}
```

Ysurac, 2020-04-14:
```"rx":162661127,"tx":89825622``` so quota seems to be reached (and there is a bug in omr-quota that make the interface up/down/up/down..., fixed for next release).

Franckrst, 2020-04-14:
Ok, so i think my router 4G display bad information. Thx !

closed 2020-04-14 by Franckrst