Ysurac/openmptcprouter #968: Memory Leak ?

Dragonox77, 2020-04-14:
## Expected Behavior
No memory leak

## Current Behavior
Hello,
I don't know if this behaviour is normal,
http://prntscr.com/rz3a9i
Only OMR is installed on it.

## Steps to Reproduce the Problem

  1. Let server run 1 month.

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: RPI3 x86_64

Ysurac, 2020-04-14:
I don't have this memory leak. Nothing special is installed on the the VPS ?
What is the VPN used ?

Dragonox77, 2020-04-14:
Thanks for your reply.
Only OMR is installed on this vps.
I'm using glorytun TCP.

Dragonox77, 2020-04-14:
If you need credentials, feel free to ask.

Ysurac, 2020-04-14:
I only need the result of a ```ps aux``` on the VPS.

Dragonox77, 2020-04-14:
https://pastebin.com/jkgc2Rcc

Dragonox77, 2020-04-14:
Sorted by MEM on htop:
http://prntscr.com/rz3w0u

Ysurac, 2020-04-15:
Looks like an issue with my API script. I'm looking on that.

Dragonox77, 2020-04-15:
oright.

closed 2020-09-14 by github-actions[bot]