Ysurac/openmptcprouter #969: VPN Down

LordBilbon, 2020-04-15:


LordBilbon, 2020-04-15:
[image]

VPN Down depuis la dernière maj vps et omr, ça a tenue quelque jours puis plus moyen de l’avoir.

J’ai bien sûr reboot le vps et omr, relancer le script, toujours pareil

Ysurac, 2020-04-15:
Quel est le VPN utilisé ? Il y a quoi dans Etat->Journal Système ?

LordBilbon, 2020-04-15:
Tout est par défaut, donc normalement Glorytun TCP

[system.txt](https://github.com/Ysurac/openmptcprouter/files/4482603/system.txt)

Ysurac, 2020-04-15:
Pas de fichiers joint, ce n'est pas pratique pour moi.

LordBilbon, 2020-04-15:
> Pas de fichiers joint, ce n'est pas pratique pour moi.

Ha bizarre, pourtant je vois bien le link du fichier 🤔

Ysurac, 2020-04-15:
Je veux dire que j'aime pas ça. Ce n'est pas pratique. Copie le contenu dans un message ou sur pastebin, en tout cas sur un truc où j'ai pas à télécharger.

LordBilbon, 2020-04-15:
> Je veux dire que j'aime pas ça. Ce n'est pas pratique. Copie le contenu dans un message ou sur pastebin, en tout cas sur un truc où j'ai pas à télécharger.

Ha pardon, je n’avais pas compris, je te fait un pastebin 😉

LordBilbon, 2020-04-15:
[https://pastebin.com/xYRMmEkr](url)

Voila :)

Ysurac, 2020-04-15:
Visiblement il y a pas mal de soucis de connexion...
Essaye de revalider le wizard.

kevin39, 2020-04-15:
FYI, a la vue du log je pense que tu as les mêmes soucis que moi, je suis chez le **même hébergeur**. 
Logs globalement similaires, connectivité instable et j'ai aussi connu ces derniers jours : erreur kernel dans les logs, erreur CPU et erreur I/O.

Leurs serveurs sont surchargés (cf leur page de statut). Ils auront de nouvelles machines prochainement pour faire baisser la charge.

Perso, j'ai pris un VPS ailleurs, j'attends la livraison pour migrer. En attente OpenMTCPRouter fait son job et me balance sur les connexions en direct quand le VPS est répond plus :)

LordBilbon, 2020-04-15:
> Visiblement il y a pas mal de soucis de connexion...
> 
> Essaye de revalider le wizard.

Pareil, toujours Down

Wed Apr 15 17:47:43 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Apr 15 17:47:46 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Wed Apr 15 17:47:47 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.100.253 dev wan1
Wed Apr 15 17:47:56 2020 user.notice omr-tracker: Launching...
Wed Apr 15 17:47:57 2020 user.notice omr-tracker: Launched
Wed Apr 15 17:47:57 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Wed Apr 15 17:47:58 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.1.1.1)
Wed Apr 15 17:48:30 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down
Wed Apr 15 17:48:30 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.100.253 dev wan1
Wed Apr 15 17:54:19 2020 daemon.err /usr/bin/ss-redir[8750]: remote recv: Operation timed out

LordBilbon, 2020-04-15:
Bon, je fait une reinstall complete du vps, on verra bien ;)

kevin39, 2020-04-15:
Tu peux faire un ticket si ma théorie est bonne. Ainsi ils vont déplacer ton vps !

Kevin LABECOT

LordBilbon, 2020-04-20:
Bon bah c'est stable depuis la reinstall du VPS, je clos on verra bien ;)

Merci a vous

closed 2020-04-20 by LordBilbon