Ysurac/openmptcprouter #970: Question about MTU

kevin39, 2020-04-15:
Hi,
By default the MTU seems detected by OpenMTCPRouter. Is it a good behavior or tuning it may be required?

I've two 4G connections.
First is currently having a MTU of 1282
Second is currently having a MTU of 1455

I'm wondering if a such difference can decrease multipath performance and if a low MTU of 1282 is a normal behavior ?

Ysurac, 2020-04-15:
1282 is really low. MTU test should be done every hours, so it's always the same or it's fixed after some times ?

kevin39, 2020-04-16:
Hi,
This morning, same MTU.

[image]

Ysurac, 2020-04-16:
MTU is calculated again only if it is not set in Network->Interfaces and for the interface.

kevin39, 2020-04-16:
I understand but the calculated MTU is 1282 it seems too low... Do you recommend to set it manually ?

Ysurac, 2020-04-16:
yes

kevin39, 2020-04-16:
Thank you. I'll try differents values to find the best settings.

closed 2020-09-14 by github-actions[bot]