Ysurac/openmptcprouter #971: DHCPV6 Persistence not effective after reboot

lolo31660, 2020-04-16:
## Expected Behavior

During setting of DHCPv6 shoudl remain after reboot of opepnmptcrouter

## Current Behavior

During setting of DHCPv6 on Lan Interface to "server mode". This setting is not stored and not effective after a reboot

## Possible Solution

## Steps to Reproduce the Problem

  1. Set lan dhcpV6-Service to server mode
  2.Save & Apply
  3.Reboot
  4.Setting is not effective (and DHCPV6 server is not vailable on LAN tested on client)

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.54.7
  - OpenMPTCProuter VPS version: 0.1015 4.19.104
  - OpenMPTCProuter platform: RPI4B

Ysurac, 2020-04-17:
IPv6 is enabled in the wizard ?

lolo31660, 2020-04-17:
Yes. An additional thing. After saving in Wizard I've got the same problem. DHCPV6 is deactivated.

Ysurac, 2020-04-17:
I need the config, without keys, available in System->OpenMPTCProuter and all settings tab.

lolo31660, 2020-04-17:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].rebind_protection='0'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/s3.ll.dash.row.aiv-cdn.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/s3-dub.cf.dash.row.aiv-cdn.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/dmqdd6hw24ucf.cloudfront.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/d25xi40x97liuc.cloudfront.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/aiv-delivery.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/netflix.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxext.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxvideo.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxso.net/omr_dst_bypass_all,omr6_dst_bypass_all'
dhcp.@dnsmasq[0].server='10.255.255.1#53'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.start='9'
dhcp.lan.limit='50'
dhcp.lan.ra='server'
dhcp.lan.ra_default='1'
dhcp.lan.ra_preference='high'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra_management='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].mac='00:11:32:11:A5:85'
dhcp.@host[0].name='DS-CAMP'
dhcp.@host[0].dns='1'
dhcp.@host[0].ip='192.168.1.13'
dhcp.@host[1]=host
dhcp.@host[1].mac='B8:27:EB:B3:69:32'
dhcp.@host[1].name='jeedom'
dhcp.@host[1].dns='1'
dhcp.@host[1].ip='192.168.1.12'
dhcp.@host[2]=host
dhcp.@host[2].mac='28:C6:8E:9F:AA:18'
dhcp.@host[2].name='netgearswitch'
dhcp.@host[2].dns='1'
dhcp.@host[2].ip='192.168.1.10'
dhcp.@host[3]=host
dhcp.@host[3].name='pihole'
dhcp.@host[3].dns='1'
dhcp.@host[3].ip='192.168.1.44'
dhcp.@host[3].mac='2A:7B:A3:13:9A:E1'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
[... 1405 lines cut ...]
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.channel='36'
wireless.radio0.hwmode='11a'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.htmode='VHT80'
wireless.radio0.disabled='1'

flimo44, 2020-04-20:
Hello,
I have same issue. Service DHCPv6 back to desabled after each reboot.

Ysurac, 2020-04-20:
I found the issue, this will be solved in next release.

flimo44, 2020-04-21:
Hello @Ysurac,
I have olso a other issue with activate or desable IPv6 ( paramètre avancé )
IP v6 is enable and it's impossible now to desable , I have "no ipv6 acces" message since 0.5x version.
No change in my configuration :/

DeeJayMX, 2020-05-04:
👍

exec, 2020-05-05:
@host, @dropbear, @classify, @samba, @zone, a few others and i are happy you've made progress, btw!

closed 2020-09-14 by github-actions[bot]