Ysurac/openmptcprouter #972: Suggested combination of MPTCP scheduler and congestion control?

jayanta525, 2020-04-17:
Not a bug report, but a inquiry about how different combination of scheduler and congestion control impacts the network performance.

I managed to compile a build with only shadowsocks and MPTCP primarily for bandwidth-aggregation, thanks to your repo openmptcprouter-feeds.

Could you comment a little about the inclusion of various scheduler such as BLEST and ECF in addition to MPTCP's default options. And nanqinlang for congestion control?

Thanks for this awesome work.

gnosoz, 2020-04-17:
which setup of wans do you have?

jayanta525, 2020-04-17:
@gnosoz College campus internet 1Gbps link, internet bandwidth capped at 15Mbps per MAC. I am creating multiple macvlan interfaces over two physical interface.

OrangePi R1 easily does 90Mbps with MPTCP and shadowsocks at just 20% CPU usage.

gnosoz, 2020-04-17:
As there aren't LTE connections in your setup I would try the following
tests:
  bbr/reno  vs  bbr/balia
 ecf/reno vs ecf/balia

You should wait 1/2 min between changes or check logs to see all
configuration applied.

Try omr-test-speed and then omr-test-speed interface1/2/etc...

Supposedly if connections are very stable the bbr/reno should give you best
speed but you never know.
BBR => best rtt
RENO => is best for interfaces with little interference / low ping / high
speed

as a reference you might want to look at this paper:
https://arxiv.org/pdf/1812.03210

Can you let us know final tests results and which one suits you best?

vinceducat, 2020-04-17:
and for setup with adsl and 4g usb tethering ? 

what do you advise ?

gnosoz, 2020-04-17:
depends on the 4g apparatus but generally ecf will try to maximize the
stable adsl/xdsl/fiber signal whilst balia will make sure the 4g is added
nicely and in a stable and constant way.

this is not valid in all setups and I always need to twitch / play around
with congestion and schedule.

I find fluctuations in the bandwidth when an unstable signal is added.
I believe (from researching the protocols) this is due to the fact he
doesn't know how much to send out or receive therefore tries to reach
congestion of the windows.

at the moment I am using:
     Gigabit fiber 250Mb reduced to 100mbps
      4g tplink mr6400
      4g tplink mr300
      4g Huawei r216
      4g Huawei k4203 ==》》baaaad!!!

Having said that, note that I had great satisfaction with (same sim card
/spot) huawei k4203 and very bad with Huawei r216 (might be faulty or just
the way it is can't tell).
Sometimes is the apparatus that creates instability rather than the network
operator or protocols used.

vinceducat, 2020-04-17:
thanks i will try ...

my adsl is low rate 1200ko/s down /  110k/s up maxi
and 4g is better for me ...2500-3500k/s

gnosoz, 2020-04-18:
I tried to install an adsl for testing purposes but I have been denied by
all operators due to FTTH coverage in my area :/

can you please show the results?
I am thinking of setting up also squid or varnish in conjunction with
pihole and test if they bring large improvements.

so far pihole cuts between 20 to 46% DNS requests. You have no idea what
crap some phones are trying to pull from port 443 every second. I can
believe it might be worse with others

vinceducat, 2020-04-18:
result of omr-test-speed?

jayanta525, 2020-04-18:
@vinceducat i didn't install the package omr-test-speed. 

after trying different combination of scheduler and congestion control, i didn't notice much differences. IMO i found good multi-thread download speeds with ECF/reno. The default setting ECF/nanqinlang was perfect for regular browsing and downloading.

using reno, iperf3 on router hovered around 150Mbps whereas in bbr,olia and nanqinlang, it was around 130~140Mbps.

I guess, scheduler and congestion algorithms doesn't make a difference in a good internet connection.

gnosoz, 2020-04-18:
i bet some ppl might love to see those extra 10/20Mbps showing up ;)

for testing if I change config I usually monitor what is going on with
iptraf-ng during a stress test as it gives me a perfect idea of what lines
he is using and how the algorithm is working

jayanta525, 2020-04-19:
compiled a build with method=none for shadowsocks.

[Netgear R6220](https://openwrt.org/toh/hwdata/netgear/netgear_r6220) with kernel 4.14 and MPTCPv0.94, was able to achieve 130Mbps easily and then gets bottle-necked (50%) by the CPU. 

This is an huge improvement over 25Mbps with chacha20-ietf-poly1305.

Although 4.14 doesn't have the additional scheduler and congestion algorithms, stuck with the defaults.

Overtime, 130Mbps degrades to ~100Mbps as ksoftirqd starts showing up with ~40% cpu usage. irqbalance doesnt seem to have any effect on this build.

talalmash, 2020-04-20:
> @gnosoz College campus internet 1Gbps link, internet bandwidth capped at 15Mbps per MAC. I am creating multiple macvlan interfaces over two physical interface.
> 
> OrangePi R1 easily does 90Mbps with MPTCP and shadowsocks at just 20% CPU usage.
> 
> [Edit] ~90Mbps is actually the 100Mbps LAN bottleneck. iperf3 on router shows ~130Mbps within the 20% cpu usage.

Can confirm with Orange Pi Zero (I flashed your R2 OMR image, working perfectly, except the new stripped image). I wish you could maintain it to the latest version, big thanks.

talalmash, 2020-04-20:
> compiled a build with method=none for shadowsocks.
>
> [Netgear R6220](https://openwrt.org/toh/hwdata/netgear/netgear_r6220) with kernel 4.14 and MPTCPv0.94, was able to achieve 130Mbps easily and then [...]

This would be perfect for the millions of routers running on ar71xx, I'm guessing that mediatek chip having similiar performance on single thread?

vinceducat, 2020-04-20:
@talalmash  hello i have wac124 https://www.netgear.com/business/products/wireless/essentials-wireless/WAC124.aspx think this is possible to transform this in omr-router ?

jayanta525, 2020-04-21:
Flashed shadowsocks=none to OrangePi R1 and was surprised with the perfomance.

LAN interface was bottle-necked (~92Mbps) with <5% CPU usage by ss-redir. In comparison to 13-15% with chacha20-ietf-poly1305.

Since, i'm not trying to bypass censorship, shadowsocks method=none shouldn't be an issue and i have my VPS within 50ms ping.

jayanta525, 2020-04-21:
> (I flashed your R2 OMR image, working perfectly except IPv6 and the new stripped non omr image). I wish you could maintain it to the latest version, Ill see if I can build one, big thanks.

Those are untested builds (might have bugs) as i usually upload them to GitHub to manage versions. If you want a properly tested build, i could provide you with one.

I am looking to get an SBC with gigabit ethernet for more throughput ([this setup](https://forum.openwrt.org/t/extending-ethernet-ports-on-sbc-with-a-managed-switch/28883)). Neatgear 6220 being able to do more then 100Mbps is great but bothers me as i cant use other OpenWrt services (adblock).

jayanta525, 2020-04-21:
> @talalmash hello i have wac124 https://www.netgear.com/business/products/wireless/essentials-wireless/WAC124.aspx think this is possible to transform this in omr-router ?
> 
> https://wikidevi.wi-cat.ru/Netgear_WAC124

Your device first needs to support OpenWrt. Head over to openwrt forums and create a topic over there. Most routers have the similar configuration, with minor differences (like in flash layout). Once the minor differences are noted, support for OpenWrt should be ready.

vinceducat, 2020-04-21:
> Your device first needs to support OpenWrt. Head over to openwrt forums and create a topic over there. Most routers have the similar configuration, with minor differences (like in flash layout). [...]

I @jayanta525 the wac124 seems to be support by openwrt like R6350
https://forum.openwrt.org/t/netgear-wac124-mediatek-mt7621at-support/37959/4

so just have to flash the wac124 to have openwrt ? this firmware ?
http://downloads.openwrt.org/releases/19.07.2/targets/ramips/mt7621/openwrt-19.07.2-ramips-mt7621-netgear_r6350-squashfs-factory.img right ?

and after what i have tu do to have a openmptcprouter working?

Ysurac, 2020-04-21:
To use on an unsupported platform: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform
I only support platform that I own, else it's difficult to test.

vinceducat, 2020-04-21:
@Ysurac  thanks i will try ;-) ....

did i test my WAC124 with un official openwrt image to make a first test ? or direct create and load omr image ?

Ysurac, 2020-04-21:
You can test it with an openwrt image, at least you will know if this work :)
There is only 128Mo Flash on this device, you may need to remove some default OpenMPTCProuter packages to make a small image.

vinceducat, 2020-04-21:
@jayanta525 ok i will try...

is it possible to came back to the OEM firmware in case of problem ?

Ysurac, 2020-04-21:
I don't have any mips platform to test, I'm open to donation ;)

jayanta525, 2020-04-21:
> I don't have any mips platform to test, I'm open to donation ;)

MIPS platforms are quite low on cpu, memory and flash. Glorytun UDP or Shadowsocks with encryption would suffer to even reach 30Mbps. The full OMR won't be any good.

jayanta525, 2020-04-21:
>  is it better? And wac is only uses like wifi ap...

@vinceducat it all boils down to your personal preferences. i used OMR for a year before moving to my own customized build. you can give it a try and check how it performs compared to your current setup.

OMR is a great piece of software,  OMR gives you an all in one solution and the ease of deployment and usage. but it felt heavily bloated to me, with those extra scripts, and primarily it failed for my purpose of using it. With my current setup, i can share my VPS with multiple routers and also use it as shadowsocks proxy and vpn(wireguard) gateway with multiple devices.

Ysurac, 2020-04-21:
VPN is used for anything that is not TCP. Shadowsocks can only aggregate TCP as it's a socks proxy. Wireguard can't aggregate connection alone because it's an UDP based VPN. So no really other choice for now.

Shadowsocks encryption can be disabled in OMR via the shadowsocks interface.

For multiple routers on a VPS, it's in test with partners and seems to work quite well now.

vinceducat, 2020-04-21:
> it all boils down to your personal preferences. i used OMR for a year before moving to my own customized build. you can give it a try and check how it performs compared to your current setup.
>
> OMR [...]

do you think raspi 3 have best native hardware perf than Wac124 ?

jayanta525, 2020-04-26:
ECF and nanqinlang seems to be the perfect combination for my setup.

closed 2020-04-26 by jayanta525