Ysurac/openmptcprouter #973: Question : How can I setup specific DNS Servuer on my DHCP ?

lolo31660, 2020-04-17:
## Expected Behavior

Is it possible to descibe how to setup specific DSN server n DHCP Server config ?
I've tried to put domain-name-servers xxx.xxx.xxx.xxx without success. 
Thanks

Ysurac, 2020-04-17:
For a specific IP ? look at this: https://openwrt.org/docs/guide-user/base-system/dhcp_configuration
Using tag and dhcp option this is possible.

vinceducat, 2020-04-17:
@Ysurac  is it possible to assign via LUCI specific DNS server to a client with statical lease define ?

bnhf, 2020-04-18:
> is it possible to assign via LUCI specific DNS server to a client with statical lease define ?

If you've assigned a static lease (aka DHCP reservation), through LuCI, then that LAN client will also receive any DNS servers or WINS servers defined under DHCP-Options.

This is done under Network – Interfaces – LAN – Edit – DHCP Server – Advanced Settings – DHCP Options, then enter the DNS servers you'd like used in the form 6,8.8.8.8,8.8.4.4 (or substitute whatever two DNS servers for Google's in this example). This also off-loads DNS duties from the router. WINS servers (to make Samba discovery fast) can also be defined this way using 44,192.168.100.1 (assuming the OMR router is setup as your WINS server).

Rebooting clients after making this change will insure a fresh request for that static LAN IP, along with DNS and WINS servers.

vinceducat, 2020-04-18:
thanks for this compelete explantions

bnhf, 2020-04-18:
> other question is it possible to define different "dns server" client by client in Luci ?

I don't think so. Unless you don't care if these clients are on different subnets, then you could perhaps put them in different VLANs, each with their own DNS servers. Each client would need to support VLAN assignment at the adapter level, or you'd need individual WiFi networks for each VLAN, or a managed switch to route clients to a specific VLAN.

> if i dont put the 6 options witch DNS is used in real? client receive omr ip adresse but who is the real dns server

Confirmation needed from @Ysurac here, but I believe if you don't specify otherwise in OMR, it looks like Unbound is used (which is your own caching DNS server). My x86 version of OMR has Unbound WAN DNS set to "Use Upstream".

Ysurac, 2020-04-18:
By default, DNS are redirected to Unbound and Unbound is set to use [Root name servers](https://en.wikipedia.org/wiki/Root_name_server).

gnosoz, 2020-04-18:
please correct me if I am wrong but it seems to me that If you want
different dns for different clients the best way is to set this up on the
client side.

Set static IP address and  specific dns configuration directly to the
client rather than try configure the router...

vinceducat, 2020-04-18:
Thanks for those information ;-)

and the job done @Ysurac :-)

closed 2020-09-14 by github-actions[bot]