Ysurac/openmptcprouter #974: Aggregated speed not restored on one WAN failure

talalmash, 2020-04-18:
Downloading an ubuntu image from the official website on 10mbps(LTE)+(4mbps WDSL) results in perfect 14mbps, manually disconnecting LTE and reconnecting results in speed of 4mbps and no load on the LTE connection from the realtime page. However stopping and restarting the download after this effect restores the full speed, and torrents aren't affected.

Am I expecting a different behavior?

The status page in System>OMR is all green, and the performance including VoIP is perfect when manually cutting off one connections.

Config: Wizard setup + Glorytun enabled + DSVPN disabled, using softVlan (e.g eth0.40).

I'm unaware of any possible fix.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.52.3-107-g62584c3 r0+11546-a74095c68c / LuCI Master git-19.323.39847-15816e6
  - OpenMPTCProuter VPS version: 0.1015 4.19.104-mptcp
  - OpenMPTCProuter platform: Orange Pi R2 build unofficial (also happens with a vmware machine)

Thanks!

Ysurac, 2020-04-19:
Do you have the same issue when you use ```omr-test-speed``` via SSH on the router ?

talalmash, 2020-04-19:
Yes, I just tried, and its the same behavior, however I tried disconnecting the other WAN (unplug) based on PPPoE and replug and the issue did not happen!
[image]

talalmash, 2020-04-19:
Oddly, running in this sequence:

1. Disconnect WAN1
2. Run omr-speed-test = 400kb/s
3. Connect WAN1, speed becomes = 1000+kb/s
4. Disconnect WAN1 = 400kb/s
5. Connect WAN1 still = 400kb/s
6. Restarting omr-speed-test = 1000+kb/s

WAN1 connection:  Openwrt router -> Wifi client conn. to Android AP -> relay bridge with vlan -> OMR

Ysurac, 2020-04-19:
I will check if I can reproduce the issue, and if changes I made solve it in develop branch.

Ysurac, 2020-04-19:
And for the bypass this seems strange... And I don't understand how you can get an empty response without that.

talalmash, 2020-04-19:
> And for the bypass this seems strange... And I don't understand how you can get an empty response without that.

Neither SSH or HTTP respond after the tunnel is up without this option. I'll try on VMware again. I've just ordered rpi4b seeing this potential, thank you.

talalmash, 2020-04-19:
> > And for the bypass this seems strange... And I don't understand how you can get an empty response without that.
> 
> Neither SSH or HTTP respond after the tunnel is up without this option. I'll try on VMware again. I've just ordered rpi4b seeing this potential, thank you.

Changing the IP address of the LAN back to stock 192.168.100.1 fixed it, no need for bypass, perhaps this should be opened for another issue. I'll retest the setup, however, do I have to disable multipath in config/network of the macvlan as it's showing "X" red, I'm using PPPoE. Thank you!

Ysurac, 2020-04-19:
I don't understand what you mean in your last sentence: "Using eth0 directly works, setting other in the wizard then changing the macvlan created wan1 to pppoe resets to static.."

talalmash, 2020-04-19:
> I don't understand what you mean in your last sentence: "Using eth0 directly works, setting other in the wizard then changing the macvlan created wan1 to pppoe resets to static.."

Sorry for the confusion, if I set a macvlan created interface in the Wizard -> Protocol to Other, then go to the Interface page, choose desired interface that was set to "Other" in the Wizard, then change the Protocol to PPPoE. Hit save and apply, it would reset to static. So I removed the macvlan and manually setup interface with eth0 instead of wan1. 

However, I just reset the firmware (re-import vmware image) and redone everything, but this time I've set the desired interface to PPPoE using CLI UCI, instead of LuCI and its working great.

Ysurac, 2020-04-19:
Macvlan are set in network->macvlan menu. And a macvlan interfaces can't be set as PPPoE, only DHCP or Static because it's a "virtual" interface based on a real interface.
Why are you using macvlan ?

talalmash, 2020-04-19:
> Macvlan are set in network->macvlan menu. And a macvlan interfaces can't be set as PPPoE, only DHCP or Static because it's a "virtual" interface based on a real interface.
> Why are you using macvlan ?

It was running without macvlan before, I thought that would be an issue as it was outside the wizard configuration, I'll set it up without macvlan as you said then. Should I try and compile the devel branch? Thank you.

Ysurac, 2020-04-19:
No, the devel branch is still in dev :)

Ysurac, 2020-04-22:
I found the issue and this will be solved in next release.

closed 2020-09-14 by github-actions[bot]