Ysurac/openmptcprouter #975: Banana Pi BPI-R64 support OpenMPTCPRouter?

matfil92, 2020-04-19:
Hi,
I want purchase the  Banana Pi BPI-R64 but in download section of openmptcprouter web page I see that banana pi is in beta and not supported. 

I can see that here seems that is openmptcp is supported http://wiki.banana-pi.org/5G_%2B_4G_LTE%2BWifi_AC%2BGigabit_Multiplex_aggregate_route 

Could it possible to know it if it will work or not?
If yes with what image?

Ysurac, 2020-04-19:
Duplicate of https://github.com/Ysurac/openmptcprouter/issues/674
No image support BPI-R64, I can't test it for now. Sinovoip, the manufacturer, sent one to me more than a month ago but I still don't have it.

closed 2020-09-14 by github-actions[bot]

cbenhagen, 2020-11-26:
@Ysurac did you receive the board by now?

Ysurac, 2020-11-26:
Yes I received it. There is an eMMC image support but not yet tested.

cbenhagen, 2020-11-26:
Thanks for the quick reply. Which version did you receive? The standard BPI-R64? Or the "5G + 4G LTE+Wifi AC+Gigabit Multiplex aggregate route" version?

Ysurac, 2020-11-26:
The standard BPI-R64

JORDANDJ41, 2021-03-01:
bonjour je viens de recevoir mon banana pi r64 je n'arrive pas a installer openmptcp

Kirball, 2022-04-29:
If anyone has advice on how to get the bp r64 to work with OpenMPTCP it would be greatly appreciated.  Kind of wish I just bought a standard router as I can't get the thing to work at all.