Ysurac/openmptcprouter #976: Raspberry Pi 4 Argon One Case

RapFan89, 2020-04-19:
Hi, thank you for your work,
i have a raspberry pi 4 with a argon1 case. It's pretty famous, because it is stylish and has good termals performance since it has a fan and an alluminum case. The problem is that the script to get the fan on is written for raspbian. Is it possibile to port in openmptcprouter? This software is very CPU intensive and a fan will help to keep the raspberry cool.

This is the script:

https://download.argon40.com/argon1.sh

Ysurac, 2020-04-19:
This would need  many changes and to create some packages.

RapFan89, 2020-04-29:
Ok, maybe it's better to ask to openwrt forum?

Ysurac, 2020-04-29:
You can try but I think they will also say it's lot of work for a fan ;)

closed 2020-09-14 by github-actions[bot]

jcouch1, 2021-04-02:
RapFan89, Did you ever have any luck getting an openmptcp or ddwrt script for the argon case fan?

Ysurac, it seems like this is a widely popular case for the rpi4 with a lot of people using the pi openmptcp.